SteveMacenski/slam_toolbox #295: Map stays silent

LiJing9876, 2020-09-23:
**Required Info:**
- Operating System: Ubuntu 18.04
- Installation type: from source
- ROS Version: melodic
- Version or commit hash: [8b8b7ef](https://github.com/SteveMacenski/slam_toolbox/commit/8b8b7ef3861390673d5bccb7c3196a9fbc69ea75)
  
#### Steps to reproduce issue
Described in [242](https://github.com/SteveMacenski/slam_toolbox/issues/242)
- Option A: 
Download the bag from [198](https://github.com/SteveMacenski/slam_toolbox/issues/198), run it
- Option B:
Run turtlebot and slam_toolbox in sync mode.

#### Actual behavior
Map is not being updated
#### Additional information
Hi,
I encountered the same problems described in issues [242](https://github.com/SteveMacenski/slam_toolbox/issues/242) and [294](https://github.com/SteveMacenski/slam_toolbox/issues/294): map stays silent when the algorithm is running.
I get the same behavior as in  [242](https://github.com/SteveMacenski/slam_toolbox/issues/242) with same messages:
[image]

Could you share the exact sequence of actions required to run the algorithm with rosbag or turtlebot simulation (including tf)?

Thanks

SteveMacenski, 2020-09-23:
The message buffer is full, that debug information doesn't really tell us anything. Does the data ever make it through the pipeline? Did you provide the full TF transformation tree required?

LiJing9876, 2020-09-24:
Here is tf tree [frames.pdf](https://github.com/SteveMacenski/slam_toolbox/files/5275175/frames.pdf)
**base_scan** is the frame of laser scanner in turtlebot simulator, so I used:
``
rosrun tf static_transform_publisher 0 0 0 0 0 0 base_footprint base_scan 10
``
to make it work with the default configuration.

How to determine if the data goes through?

SteveMacenski, 2020-09-24:
Print something in the callbacks to see if its available, and looking at the debug output from the message filter if its rejecting candidates due to not being able to transform.

SteveMacenski, 2020-09-24:
I just tested via

```
mkdir -p catkin_ws/src
cd catkin_ws/src
git clone git@github.com:SteveMacenski/slam_toolbox.git -b melodic-devel
cd ../../
catkin config --install 
catkin build
```

In a terminal each:
```
roscore
rosbag play --clock <rosbag/path>
roslaunch slam_toolbox online_sync.launch
rviz # to see map
```

and things are working fine. Closing ticket, something to do with your config, look through your setup or ask on ROS Answers. This doesn't represent a bug or feature request in the package so this shouldn't be a github ticket.

closed 2020-09-24 by SteveMacenski