SteveMacenski/slam_toolbox #303: Multi_laser for Foxy

MortenMDahl, 2020-10-21:
Hi,
I'm trying to implement slam_toolbox on a robot with two laser scanners (publishing to different topics), but as the current version of multi_laser doesn't build on Foxy, i'm having some issues. 
Do you have any plans for updating this feature in the near future?

Thanks!

SteveMacenski, 2020-10-21:
There is no support for multi-laser configs. Try a laser scan assembler.

closed 2020-10-21 by SteveMacenski

MortenMDahl, 2020-11-03:
Thank you for the tip, and for anyone else looking to solve this problem, I made my own assembler which converts LaserScan messages into PointCloud2, transforms them into the same coordinate system and then concatenates them. 

The code can be found [here](https://github.com/MortenMDahl/kmriiwa_ws_devel/blob/foxy/src/kmr_concatenator/nodes/concatenator_node.py). 
WARNING: I'm very much a coding and ROS beginner, and this code is specified to the robot i'm working on. 
The reasoning behind making my own is that many of the packages doing this is out of date.