SteveMacenski/slam_toolbox #307: failed to launch on ros2-dashing exit code -11

DecentBumpkin, 2020-11-09:
**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binary, apt-get
- ROS Version
  - Ros2-Dashing
- Version or commit hash:
  - 
- Laser unit:
  - 
  
#### Steps to reproduce issue

ros2 launch slam_toolbox online_async_launch.py
#### Expected behavior

#### Actual behavior
[ERROR] [slam_toolbox-1]: process has died [pid 17610, exit code -11, cmd '/opt/ros/dashing/lib/slam_toolbox/async_slam_toolbox_node __params:=/opt/ros/dashing/share/slam_toolbox/config/mapper_params_online_async.yaml'].

#### Additional information
I am following the tutorial for ros2 navigation2 with a turtlebot3 gazebo simulation, unfortunately when I launched the node it immediately failed

SteveMacenski, 2020-11-09:
That tells us nothing, please provide a backtrace and a full stdout.

DecentBumpkin, 2020-11-09:
ros2 launch slam_toolbox offline_launch.py 
[INFO] [launch]: All log files can be found below /home/wei/.ros/log/2020-11-08-16-26-38-805205-lenovo-17731
[INFO] [launch]: Default logging verbosity is set to INFO
RTI Data Distribution Service Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00 See www.rti.com for more information.
[INFO] [slam_toolbox-1]: process started with pid [17753]
[ERROR] [slam_toolbox-1]: process has died [pid 17753, exit code -11, cmd '/opt/ros/dashing/lib/slam_toolbox/sync_slam_toolbox_node __params:=/opt/ros/dashing/share/slam_toolbox/config/mapper_params_offline.yaml'].

I tried quickly on another laptop with same same ros distribution and your dashing binary and it worked, I totally don't understand. the only difference is that I followed ros2 tutorial and built the realtime linux kernel and selected connext as dds, so terrible. I don't know whether this will make a difference. For now I am going to switch to the other laptop. Thank you very much.

SteveMacenski, 2020-11-09:
> I tried quickly on another laptop with same same ros distribution and your dashing binary and it worked, I totally don't understand. 

This is then a setup problem for your application and not from the pkg, closing.

closed 2020-11-09 by SteveMacenski