SteveMacenski/slam_toolbox #308: [Question] Loading previously made maps

MortenMDahl, 2020-11-12:
Hi, I'm having some issues loading previously made maps. 

**TLDR; I seem to be referencing my map file incorrectly. Do I have to add its path in launch parameters?
Also, it seems your example config file 'mapper_params_localization.yaml' has `mode: mapping` enabled. Not sure if this is supposed to be changed to localization.**

I made a map using the normal mapping parameters and saved it using
`ros2 run nav2_map_server map_saver_cli -f 'GAZEBO'`

Which gave me the two files, 
GAZEBO.yaml
GAZEBO.pgm

I placed the files in a sub-folder with the name 'created_maps' and added GAZEBO to map_file_name in the config file.

    slam_toolbox:
      ros__parameters:

    # Plugin params
    solver_plugin: solver_plugins::CeresSolver
    ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
    ceres_preconditioner: SCHUR_JACOBI
    ceres_trust_strategy: LEVENBERG_MARQUARDT
    ceres_dogleg_type: TRADITIONAL_DOGLEG
    ceres_loss_function: None

    # ROS Parameters
    odom_frame: odom
    map_frame: map
    base_frame: base_footprint
    scan_topic: /scan_concatenated
    mode: localization #mapping

    map_file_name: created_maps/GAZEBO
    map_start_pose: [0.0, 0.0, 0.0]
    #map_start_at_dock: true

    debug_logging: false
    throttle_scans: 1
    transform_publish_period: 0.02 #if 0 never publishes odometry
    map_update_interval: 5.0
    resolution: 0.05
    max_laser_range: 10.0 #for rastering images
    minimum_time_interval: 0.1
    transform_timeout: 0.2
    tf_buffer_duration: 30.
    stack_size_to_use: 100000000 #// program needs a larger stack size to serialize large maps
    # Was 40000000
    enable_interactive_mode: false #was true

    # General Parameters
    use_scan_matching: true
    use_scan_barycenter: true
    minimum_travel_distance: 0.1 #was 0.5
    minimum_travel_heading: 0.1 #was 0.5
    scan_buffer_size: 50 #was 10
    scan_buffer_maximum_scan_distance: 20.0
    link_match_minimum_response_fine: 0.1  
    link_scan_maximum_distance: 1.5
    loop_search_maximum_distance: 3.0
    do_loop_closing: true 
    loop_match_minimum_chain_size: 10           
    loop_match_maximum_variance_coarse: 3.0  
    loop_match_minimum_response_coarse: 0.35    
    loop_match_minimum_response_fine: 0.45

    # Correlation Parameters - Correlation Parameters
    correlation_search_space_dimension: 0.5
    correlation_search_space_resolution: 0.015
    correlation_search_space_smear_deviation: 0.1 

    # Correlation Parameters - Loop Closure Parameters
    loop_search_space_dimension: 8.0
    loop_search_space_resolution: 0.015
    loop_search_space_smear_deviation: 0.03

    # Scan Matcher Parameters
    distance_variance_penalty: 0.5      
    angle_variance_penalty: 1.0    

    fine_search_angle_offset: 0.00349     
    coarse_search_angle_offset: 0.349   
    coarse_angle_resolution: 0.0349        
    minimum_angle_penalty: 0.9
    minimum_distance_penalty: 0.5
    use_response_expansion: true

When trying to start the localization, it gives me the following errors:

[localization_slam_toolbox_node-1] [ERROR] [1605190195.677149050] [slam_toolbox]: serialization::Read: Failed to open requested file: created_maps/GAZEBO.
[localization_slam_toolbox_node-1] [ERROR] [1605190195.678616123] [slam_toolbox]: DeserializePoseGraph: Failed to read file: created_maps/GAZEBO.

Changing map_file_name to something that doesn't exist gives me the exact same error, which leads me to believe that the cause is that I don't reference the file correctly. Any tip would be appreciated.

Thank you very much.

MortenMDahl, 2020-11-12:
I now see that the `map_file_name` parameter takes in a 'pose-graph file' which I guess is a (de?)serialized map. 
Is there any built-in way to convert the .pgm and .yaml, except for the method mentioned in the rviz plugin?

SteveMacenski, 2020-11-12:
Consult the readme. If you had any additional questions, see the navigation2 tutorials and ask on ROS Answers. The issue tracker is not asking for assistance

closed 2020-11-12 by SteveMacenski

MigVega, 2022-12-04:
Hi all,

Now you can use the [Ogm2Pgbm package](https://github.com/MigVega/Ogm2Pgbm/tree/main/ogm2pgbm) to do the conversion from `.pgm` + `.ymal` to Pose graph based map in ` .posegraph` format !

If haven't compiled cartographer already, you can follow the provided instructions in the [readme](https://github.com/MigVega/Ogm2Pgbm). It will allow you to compile everything in one isolated container.

Otherwise you can also use the package as a normal separated catkin package.

Have fun, and enjoy the power of graph-based localization! :D

santoshbalaji, 2023-09-12:
You can use the SlamToolboxPlugin in rviz for saving the map in serialized or deserialized format. On doing this you will get two files with the format .posegraph and .data

[image]