SteveMacenski/slam_toolbox #311: Moved graph node went back to the previous position after pushing `Save Changes` button

buster84, 2020-11-20:
**Required Info:**

- Operating System:
  - Ubuntu 18.04 
- Installation type:
  - ros-melodic-slam-toolbox (binaries) 
- ROS Version
  - ROS1 melodic 
- Version or commit hash:
  -  1.1.6 (in package.xml (/opt/ros/melodic/share/slam_toolbox/package.xml)) 
- Laser unit:
  - libgazebo_ros_laser.so ( we are using gazebo( simulator))
  
#### Steps to reproduce issue

1. Run gazebo with world and spawn robot model
2. Launch ` roslaunch slam_toolbox online_sync.launc` 
3. Run rviz
4. Doing slam without hitting wall
5. Hit wall to make odometry mess up
6. Check Interactive mode
7. Move one of graph node to the position which the node is supposed to be
8. Click `Save Changes` button

#### Expected behavior
I expected the node stayed close to the position I moved. 

#### Actual behavior
It seemed the node went back to the previous position. 

#### Additional information
The black circle in below picture is the node I moved to the point arrow pointing. 
[image]
After clicking the `Save Changes` button, it seems the node went back to the previous position. 
[image]

I am not so much familiar with slam algorithm so actually I am not sure whether this is expected behaviour or bug.

SteveMacenski, 2020-11-20:
Its not always enough to just move a single node. The single node has constraints against local neighbors so its necessary to massage more of the neighborhood. 

https://github.com/SteveMacenski/slam_toolbox/issues/70 covers a ticket who's aim is to reduce that requirement significantly by updating the constraints with the moved nodes so they are "more incentivized" mathematically to deflect the other nodes to match this one / not move it back. However, its not in my immediate roadmap but I'd be happy to review PRs or design discussions around it.

buster84, 2020-11-24:
Thank you for the quick response and the nice explanation. I think I understood. 

I would like to make a PR. Before that, I would like to hear your opinion. Would you like to keep the current `Save Changes` feature as is and add the feature which enable the user to update the constraints with the moved nodes or replace the `Save Changes` feature with the updating the constraints feature? @SteveMacenski

SteveMacenski, 2020-11-24:
The button just calls a service that performs the action, so if we changed the code to adjust constraints, then the button would just call the new changes.

buster84, 2020-11-24:
> The button just calls a service that performs the action, so if we changed the code to adjust constraints, then the button would just call the new changes.

We can create a new service and a new button but I think we don't need to keep the current behaviour. Okay, I will change the service behaviour to adjust constraints.

SteveMacenski, 2021-02-24:
@buster84 any progress on this?

buster84, 2021-02-25:
@SteveMacenski Sorry, I was busy in other things and still I don't have time to work on this.

SteveMacenski, 2021-02-25:
Ok, then I'm going to close this as a duplicate of #70 (and link them so people can see this discussion) since they cover the same topics. Happy to have this work done and help however I can, but just trying to keep the issue tracker clean to direct actionable topics and that feature request has already been documented.

closed 2021-02-25 by SteveMacenski