SteveMacenski/slam_toolbox #312: Slam_toolbox creates 2 different maps

joe28965, 2020-11-27:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source
- ROS Version
  - ROS Foxy
- Version or commit hash:
  - git rev-parse HEAD: 0cf69f2435d101470d9e81ff93d06e7c7f59b1bb
- Laser unit:
  - Gazebo simulated, with gazebo_ros ray plugin
  
#### Steps to reproduce issue
I start my world, with my robot in it. I start the various slam_toolbox launch files, and use teleop to drive my robot around.

#### Expected behavior
I would expect the map to be one map, instead of the two that it produces for me.

#### Actual behavior
After driving for a little while, the map somehow becomes 2 maps, where the second map will rotate (always roughly the same amount) and the maps will no longer really update.

[image]

#### Additional information

SteveMacenski, 2020-11-30:
Please provide a bag file to reproduce

joe28965, 2020-12-01:
I made a bag, but it's too big to upload to github, so I put it on my google drive [link](https://drive.google.com/file/d/17vu2WTB__XHyKvwZprxkY8Dh5EivZEnv/view?usp=sharing)

I recorded all the topics, I'm not sure if that's what you meant, if not, please let me know and I'll make another one

SteveMacenski, 2021-02-03:
Where is your lidar mounted (forward, backward, angled, upside down, etc) and what is it configured at (so I can reproduce)?

SteveMacenski, 2021-02-03:
We may want to move this discussion to #281 covering the same topic

closed 2021-02-03 by SteveMacenski