SteveMacenski/slam_toolbox #314: Editing Maps and Posegraph Files

Rojekk, 2020-11-30:
Hi, first of all thanks for this great slam tool.

Here is my issue, i sometimes need to edit my maps' .pgm files with a photo editor when i want to erase a spot or add obstacles to the map. I want to be able to do that when i use continue mapping of slam_toolbox. 

When i serialize a map, slam_toolbox saves it as .posegraph format.  But if i edit the pgm file, posegraph file doesn't change, obviously. So the question is, is there a way to implement my changes to posegraph files. If not, would you consider an implementation which will allow users to edit maps which will also edit posegraphs ? And any kind of advise to accomplish this would be appreciated.

SteveMacenski, 2020-11-30:
Sorry, but I don't have any suggestions.

SteveMacenski, 2021-02-16:
#329 covers this intent

closed 2021-02-16 by SteveMacenski