SteveMacenski/slam_toolbox #318: [JOSS review] Example usage

mosteo, 2020-12-18:
Hello, Steve.

I'm one of the reviewers at https://github.com/openjournals/joss-reviews/issues/2783.

I'm opening this issue in regard to the point:
- [ ] **Example usage**: Do the authors include examples of how to use the software?

Although there are a few videos and captures, the only hands-on info I see in the README is the line:

> You can run via ros2 launch slam_toolbox online_sync_launch.py

which, by itself, doesn't bring up a complete example.

The page https://github.com/SteveMacenski/slam_toolbox/wiki does not contain one either.

Is https://navigation.ros.org/tutorials/docs/navigation2_with_slam.html the intended complete example?

SteveMacenski, 2020-12-18:
Hi, yes we have example usage tutorials
- https://navigation.ros.org/tutorials/docs/navigation2_with_slam.html (how to do slam while navigating)
- http://wiki.ros.org/Robots/ARI/Tutorials/Navigation/Mapping (an unrelated company created a tutorial for their customers based on this work)

The Nav2 website tutorial is assuming you have functional knowledge of navigation leading up to it, so it skips over topics covered in other getting started tutorials. Since it leverages a bunch of ROS standards, its a drop in replacement for others in the ecosystem. 

While this is a stand alone piece of software, you have to have inputs coming in from sensors (real or simulated) to have it do anything of interest. The required inputs are described https://github.com/SteveMacenski/slam_toolbox#subscribed-topics and these are well-established standards for the ROS open-source community that all SLAM and localization implementations comply to.

mosteo, 2020-12-30:
To fulfill the JOSS guideline, these tutorials should be linked from the README.

Also, as I saw in https://github.com/SteveMacenski/slam_toolbox/issues/320, and in particular about

> Perhaps an aside, but I disagree that its "too much of a burden"

I don't consider installing ROS2+Gazebo excessive, but I think that at least a pointer to a complete, self-contained example should be provided in the README. One that when followed to the letter from start to finish, having only cursory familiarity with a similar framework (e.g., Orocos) would get you a working demo in simulation. I see most of the pieces are here and there, but for some glue.

https://navigation.ros.org/tutorials/docs/navigation2_with_slam.html is almost there, provided that installation instructions were complete (some necessary packages are not mentioned) and it could be entirely done in simulation.

SteveMacenski, 2021-01-04:
> To fulfill the JOSS guideline, these tutorials should be linked from the README.

Fair enough, has been added: https://github.com/SteveMacenski/slam_toolbox/blob/foxy-devel/README.md

> provided that installation instructions were complete (some necessary packages are not mentioned)

I'm not sure what you mean by that. The installation instructions should be complete, many users use that to get started and I haven't heard an issue in quite awhile on that.

I'm not sure I totally grasp your other comment. Is there an action item I need to take? I added it to the readme which I think fulfills your first comment but I don't get the Orocos comment.

mosteo, 2021-01-25:
Sorry for the absence. 

Yes, you're correct, I had missed some context with my previous message. I meant that the simple tutorial you link to is not a starting point but another one in a series for the ARI robot, which may be confusing. Also, if my reading is correct, that's for ROS1. 

Ideally, a self-contained tutorial for ROS2 would be optimal, with a simpler simulated robot. But I guess that's icing on the cake.

closed 2021-01-25 by mosteo