SteveMacenski/slam_toolbox #319: map_start_pose is always set

eliasdc, 2020-12-18:
From the code map_start_pose is always set but it can be an empty vector. The if statement in
https://github.com/SteveMacenski/slam_toolbox/blob/0cf69f2435d101470d9e81ff93d06e7c7f59b1bb/src/slam_toolbox_common.cpp#L308 will always result in true so we cannot use map_start_at_dock. 

Shouldn't this be a check if this list is empty or the default should be null?

SteveMacenski, 2020-12-18:
Indeed, there's a reason they're commented out in the config files, these 2 parameters are mutually exclusive. You cannot both start at a dock and start at a pose, its one or the other. You should not set the variables in your config file you're not using for that situation.

eliasdc, 2020-12-19:
I know that they are meant to be mutually exclusive. But now dock pose can't be used as the start pose has an empty vector as default value so the if statement is always true.

SteveMacenski, 2020-12-19:
Are you sure that it doesn't work if you comment out the start pose variable? Maybe I broke this in a refactor, but I swore that I tested this to make sure that if that variable wasn't set at all it would fail. 

if you find that yes, in fact it does not work even with `map_start_pose` not set at all, then I think the easiest solution is to move the `get_parameter` for `map_start_pose` one line up and then `if (start_pose.empty())` to be the check, so that if its empty, we ignore it.

eliasdc, 2020-12-22:
Just tested it again and yes I'm sure, I always get this error:
```
[slam_toolbox]: LocalizationSlamToolbox: Incorrect number of arguments for map starting pose. Must be in format: [x, y, theta]. Starting at the origin
```
Even when `map_start_pose` is not set. I'll create a PR with that empty check and an else condition if nothing is set but `map_file_name`.

eliasdc, 2020-12-22:
So the idea of `get_parameter` was correct but if you declare it with a default parameter you'll always get true returned. I removed the defaults in the declare call and checked the return value of `get_parameter` to check if a config has been set. If none are set I use the origin and set `map_start_at_dock` to false. To my knowledge this should create the desired behavior.

SteveMacenski, 2020-12-22:
Removing the defaults was fine - though setting the origin is not a logical default so I've asked you to remove it in the file.

On the if set, it might be worth capturing the return type on the `declare` and then checking if the `if/if else` statements if the return type is not `rclcpp::ParameterType::PARAMETER_NOT_SET` rather than `get` as the checks to then set the value. Only then `get` after we know its been set.

eliasdc, 2021-01-04:
Implemented your comments as requested.

SteveMacenski, 2021-01-05:
Merged, closing ticket

closed 2021-01-05 by SteveMacenski