SteveMacenski/slam_toolbox #320: JOSS Review

carlosjoserg, 2020-12-21:
Hi @SteveMacenski

I'm reviewing your submission to JOSS _SLAM Toolbox: SLAM for the dynamic world_

This is a first round of issues on the checklists proposed in https://github.com/openjournals/joss-reviews/issues/2783:

1. > **Automated tests**: Are there automated tests or manual steps described so that the functionality of the software can be verified?

By quick inspection, I see only [this file](https://github.com/SteveMacenski/slam_toolbox/blob/foxy-devel/test/lifelong_metrics_test.cpp) containing automated tests.

But also see that it is comented in the [CMakeLists.txt](https://github.com/SteveMacenski/slam_toolbox/blob/foxy-devel/CMakeLists.txt#L169-L177), at least for the `foxy-devel` and the [`melodic-devel`](https://github.com/SteveMacenski/slam_toolbox/blob/melodic-devel/slam_toolbox/CMakeLists.txt#L154-L160) branches.

Is there any reason for this non-homogenous testing across the branches?

2. > **Example usage**: Do the authors include examples of how to use the software (ideally to solve real-world analysis problems).

I saw the thread https://github.com/SteveMacenski/slam_toolbox/issues/318

Having to set and run a gazebo world is too much of a burden to being able to run an example. Consider having just the required topics (for instance, as a rosbag) to emulate and run an example.

3. > **Community guidelines**: Are there clear guidelines for third parties wishing to 1) Contribute to the software 2) Report issues or problems with the software 3) Seek support

(1) I could not find any contribution guidelines (see for example [MoveIt!](https://moveit.ros.org/documentation/contributing/) or [MongoCPPDriver](https://github.com/mongodb/mongo-cxx-driver/blob/master/CONTRIBUTING.md)).

Looking at the [branches](https://github.com/SteveMacenski/slam_toolbox/branches) does not help either. I'd find helpful to state the branching strategy (see again [MoveIt](https://github.com/ros-planning/moveit#branches-policy) or [MongoCPPDriver](https://github.com/mongodb/mongo-cxx-driver#branches---releasesstable-versus-master))

(2) Github templates.. OK

(3) I could not find an explicit way to do that. (e.g. Github Issue, ROS Answers, Discourse, Slack, Email, etc.)

SteveMacenski, 2020-12-21:
> Automated tests: Are there automated tests or manual steps described so that the functionality of the software can be verified?

We have automated testing suites as part of the build farm as well (linting, code styling, static analysis, etc) that are done behind the scenes and pass for every PR, merge, and distribution release. These are _way_ more complete in ROS2. We meet that requirement as a requirement of the first and second: we have strong static analysis and linting automated test suites and manual steps as part of the tutorials you linked to as manual steps for verification on live data streams.

I definitely know we need more automated test coverage, but I think we squeak by on the actual requirement. More tests are planned.

> Having to set and run a gazebo world is too much of a burden to being able to run an example. Consider having just the required topics (for instance, as a rosbag) to emulate and run an example

Not a bad idea, however I'm not able to share any of my datasets publicly. I disagree that its "too much of a burden", that's subjective to your target audience. As this is a production-grade SLAM solution, its the expectation of a minimum viable robot system in order to even generate the data that any SLAM system requires. If those can't be provided, we do offer our tutorials which documents the steps to do so in gazebo so that anyone and everyone can have an example use. I'd also argue that installing ROS is quite a burden to test this, but that's totally unavoidable. Niche things like like naturally have a higher barrier to entry than a basic importable python library. 

We meet the requirement of having example usages (even more than 1), though a rosbag of REP-103 compliant data would work fine, I have plenty that I run it over to verify major PR changes. The average new user doesn't come loaded up with datasets so being able to run a real simulation with real-live simulated data using the canonical demo of ROS is a far more practical and meeting the "Ideally..." note in the requirement list.

> Contribute to the software

You are right, I overlooked that, thank you. I edited the default branch readme file to give that context. File tickets or email me; then filling in templates and we'll get back to users. https://github.com/SteveMacenski/slam_toolbox#support-and-contribution

> Report issues or problems with the software

I think we both agree this was met?

> Seek support

You are right, I overlooked that, thank you. I edited the default branch readme file to give that context. Config / help -> ROS Answers; issues/features -> issues. Second part also makes more explicit point 2.  https://github.com/SteveMacenski/slam_toolbox#support-and-contribution

carlosjoserg, 2021-03-24:
Hi @SteveMacenski , apologies for the big delay.

I (re-)read the paper some days ago, reviewed your replies to my issues, and ticked the related check-boxes in the main thread.

The remaining part is about the actual funcionality (and its documentation), which I'd like to go through in the next few days. I'll keep the issue open  just in case.

And thanks for your patience!

SteveMacenski, 2021-03-24:
Thanks! By the way, for the still-unmarked box API documentation: http://docs.ros.org/en/noetic/api/slam_toolbox/html/namespaces.html (and https://vimeo.com/378682207 video where I explain all the features as well)

If you have any questions on testing the features, don't hesitate to ask!

SteveMacenski, 2021-04-27:
@carlosjoserg anything I can do to help unblock? The only 3 remaining items on your review list are 
- Functionality
- Performance
- Functionality documentation

carlosjoserg, 2021-05-11:
Hi @SteveMacenski 

Thanks for your patience (and for sharing).

Finally got to test part of the functionality on `melodic` and `noetic`. It all looks good as described, rviz plugin loaded fine too.

I'm marking all items resolved, closing the issue, and wish for a reduction in the amount of parameters (not a particular request for slam_toolbox though, but for anything in general ;)).

Best,