SteveMacenski/slam_toolbox #328: Map topic silent and LaserRangeScan warning when using a bag

imstevenpmwyca, 2021-02-10:
**Required Info:**

- Operating System:
  - Ubuntu 20.04.2 LTS
- Installation type:
  - Binaries
- ROS Version
  - ROS Noetic 1.14.3
- Version or commit hash:
  - noetic-devel
- Laser unit:
  - Capteur LiDAR 2D OBD10M-R2000-4EP-V1V17
  
#### Steps to reproduce issue
1. Filter the bag with: `rosbag filter mappingData_input.bag mappingData_output.bag 'topic != "/tf" or topic == "/tf" and m.transforms[0].header.frame_id != "map" and m.transforms[0].child_frame_id != "odom"'`
2. For some reason step 1 also deletes some parts of the tf robot description so: rosrun the robot description to recover them
3. Play the bag and roslaunch slam_toolbox with its default configurations

#### Expected behavior
1. Map is published and can be visualized/saved from rviz
2. No warnings regarding the LaserScan messages

#### Actual behavior
1. No msg is published in the /map topic
2. `LaserRangeScan contains 2706 range readings, expected 2707` is shown multiple times along the test

#### Additional information
Dear @SteveMacenski, thanks for your time and help. Let me elaborate the issue further:

We are currently evaluating the possibility of using the slam_toolbox package for our industrial mobile robot. I wanted to evaluate its slam capabilities by testing several rosbags that we have collected from real industrial environments.
The first step is to filter the bag file, since we want to get rid of the map->odom tf that was recorded by following the command above. As shown in the images below, after the filter some other tf are deleted so I launch our robot description node to recover the tfs lost. Finally I get a tf tree without the map->odom. 

[image] 
[image] 
[image] 

I also proceed to rosbag play the filtered bag, the slam_toolbox sync node with the offline default config (All of this is actually done in a single launch file) and using sim time. Slam_toolbox node is successfully initialized and no error is shown. However, the /map topic never gets a msg and after running for a couple of seconds the LaserRangeScan messages is constantly printed. The tf tree now includes the tf map->odom from the slam_toolbox node and the rqt_graph seems fine, as shown in the images below (I made slam_toolbox node listen to the /scan_filtered node instead of /scan from the config file).

[image]
[image]

This issue happens to me with different bags and with the other slam_toolbox nodes. However, it only seems to happen when using a rosbag since everything works fine if getting the sensor data from a simulation.

[image]

I suspect that, at least for the quiet /map issue, either I'm not using the rosbags as it should be or I haven't configured the system (including slam_toolbox) right or there is a real issue with the package. I also found [this](https://answers.ros.org/question/368741/slam-toolbox-unable-to-get-map-messages-when-running-the-toolbox-in-ros-noetic/), #242 and #295 which all discuss similar behaviors. And for the LaserRangeScan I am clueless,  I found #63 and #278 but both of them seem to be resolved/merged.

Please let me know if something isn't clear or if you need further information, I can also attach a portion of the rosbag if needed.
Thanks!

SteveMacenski, 2021-02-10:
> LaserRangeScan contains 2706 range readings, expected 2707

This clearly seems like the issue - you shouldn't ignore warnings. Is your ROS driver properly implementing the message?

imstevenpmwyca, 2021-02-10:
> Is your ROS driver/ simulation data generator properly implementing the message?
What do you mean? Is there any proper way to check that?

> I'm not going to try to dissect all of the rosbag issues - those aren't related to this package.
I know the filtering and robot description aren’t but if the procedure is right, then it may highlight  an underlying issue with the package. The three referenced tickets makes me think so. Or maybe just by solving the LaserRangeScan issue the /map topic issue will be sorted out.

Thanks for the quick reply!

SteveMacenski, 2021-02-10:
I don't think there's an issue in the package, it sounds like the inputs are incorrect and there's a clear warning being thrown you should investigate before anything else.

imstevenpmwyca, 2021-02-10:
Ok, I will continue investigating these following days. I will keep the thread updated.
Do you have any rosbag that you know for sure has worked with the package (or link)? Maybe I can do an input comparison between that bag file and the ones I'm trying to use.

imstevenpmwyca, 2021-02-11:
I now understand where the issues are coming from and how to solve them.

1. The rosfilter and robot description procedures -and therefore, the /tf- were correct.
2. No msg is published in the /map topic because of the LaserRangeScan warning. In the [Validate](https://github.com/SteveMacenski/slam_toolbox/blob/54d735ffae6b2a7a255ec4f9c94675917d0a7108/slam_toolbox/lib/karto_sdk/src/Karto.cpp#L212) function, false is returned if the NumberOfRangeReadings don't match.
3. The reason why the NumberOfRangeReadings don't match is because the [Update](https://github.com/SteveMacenski/slam_toolbox/blob/54d735ffae6b2a7a255ec4f9c94675917d0a7108/slam_toolbox/lib/karto_sdk/include/karto_sdk/Karto.h#L4363) function doesn't identify our LiDAR as a 360 one. I am not sure why; it might be either because our driver or because of a further issue with the function GetIs360Laser(). Anyway, I solved the issue temporarily by setting the residual variable's value to 0 anyway.

Will update if I find the real reason.

SteveMacenski, 2021-02-11:
You're right on the money, so what I'd look at is that check, print a laser message out of simulation and pull out a calculator and see what's up. We've had issues with that 360 detection before, but I'm 90% sure we made it right so I'd make sure the message seems reasonable from player (I don't know I've met anyone before using Player in recent history so maybe their driver didn't implement something we're picky about?) 

https://github.com/SteveMacenski/slam_toolbox/blob/5364ba143c138da6126702c1f8568ba4d94f0c0b/src/laser_utils.cpp#L120 this is where the is 360 is set. You can check the criteria and let us know if you spot an issue.

imstevenpmwyca, 2021-02-12:
I noticed that two other users suffered this exact same issue in #140 and solved it in a 'hackish' manner, it is important to highlight that all three of us were using the same r2000 LiDAR. And also I realized that we are using our own driver for performance improvements.

> (I don't know I've met anyone before using Player in recent history so maybe their driver didn't implement something we're picky about?)

What do you mean by Player?

SteveMacenski, 2021-02-12:
Your TF trees have a player node, is that not Player, the simulator?

doisyg, 2021-02-15:
Not a SLAM toolbox issue, @imstevenpmwyca you can close the issue.
For reference, it was indeed a bug in our lidar driver not filling properly the `angle_max` field (it was exactly one `angle_increment` too large). Turns out SLAM toolbox is the only piece of software really making good use of this field (and many lidar drivers are buggy).

> Your TF trees have a player node, is that not Player, the simulator?

I love Player/Stage but that was just the rosbag player

closed 2021-02-15 by imstevenpmwyca