SteveMacenski/slam_toolbox #330: [Foxy] slam_toolbox crashes after loading posegraph

wienans, 2021-02-14:
**Required Info:**

- Operating System:
  - 20.04
- Installation type:
  - from source slam_toolbox foxy-devel but same issue with slam_toolbox foxy release 2.3.0 from source
- ROS Version
  - ROS2, foxy
- Version or commit hash:
  - 875568697689774af3f9c65dae91c0028b98d798 (i think this is right, commit from 5 Jan)
- Laser unit:
  - Gazebo rayscan
  
#### Steps to reproduce issue
I serialized this map / posegraph with the slam_toolbox using Gazebo
[map.zip](https://github.com/SteveMacenski/slam_toolbox/files/5977879/map.zip)
After restart of slam_toolbox i tried to load the posegraph with
`ros2 service call /deserialize_map slam_toolbox/srv/DeserializePoseGraph "{filename: 'map', match_type: 2, initial_pose: {x: 0.0, y: 20.0, theta: 0}}"`

#### Expected behavior
Map and posegraph is loaded and we can do slam on the prerecorded map
#### Actual behavior
After calling the service, the Map is loaded (i can see it in rviz) but /grap_visualization does not show. And then the slam_toolbox crashes instantly
with this messages:
`[async_slam_toolbox_node-1] Load From File map.posegraph 
[async_slam_toolbox_node-1] Mapper <- Module
[async_slam_toolbox_node-1] Mapper <- m_pSequentialScanMatcher
[async_slam_toolbox_node-1] Mapper <- m_pGraph
[async_slam_toolbox_node-1] MapperGraph <- Graph; Graph <- m_Edges; Graph <- m_Vertices
[async_slam_toolbox_node-1] MapperGraph <- m_pMapper; MapperGraph <- m_pLoopScanMatcher; MapperGraph <- m_pTraversal
[async_slam_toolbox_node-1] Mapper <- m_pMapperSensorManager
[async_slam_toolbox_node-1] MapperSensorManager <- m_ScanManagers; MapperSensorManager <- m_Scans
[async_slam_toolbox_node-1] Mapper <- m_Listeners
[async_slam_toolbox_node-1] **Finished serializing Mapper**
[async_slam_toolbox_node-1] Load From File
[async_slam_toolbox_node-1] **Serializing Dataset**
[async_slam_toolbox_node-1] Dataset <- m_SensorNameLookup
[async_slam_toolbox_node-1] Dataset <- m_Data
[async_slam_toolbox_node-1] Dataset <- m_Lasers
[async_slam_toolbox_node-1] Dataset <- m_pDatasetInfo
[async_slam_toolbox_node-1] **Finished serializing Dataset**
[async_slam_toolbox_node-1] Unregistering sensor: Custom Described Lidar
[async_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
[async_slam_toolbox_node-1] terminate called after throwing an instance of 'std::logic_error'
[async_slam_toolbox_node-1]   what():  basic_string::_M_construct null not valid`
#### Additional information
There should be a posegraph vertex at the position the service is called as i started the mapping at this position
I tried to dig a bit in the code but can't find where these messages are produced. I only found the function deserializePoseGraphCallback which should be called from the service and because the map is updated ant the function updateMap() is called at 746 i would assume that everything is working until that point but there is not mutch left after the updateMap() funciton.

Thanks in advance

SteveMacenski, 2021-02-16:
Without GDB tracebacks, I can't say anything in particular about this. Was the file properly serialized?

wienans, 2021-02-20:
Hey Steve, thanks for your answer. I never worked with gdb directly. That why i hope this is what you asked for. (Followed the Get Backtrace in ROS2 Tutorial from NAV2.) After i saw the output i think i should look into gdb more for ROS debugin it looks quite helpfull :+1:  
``(gdb) backtrace                                                                                                                                                                                                                           
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50                                                                                                                                                                 
#1  0x00007ffff7496859 in __GI_abort () at abort.c:79                                                                                                                                                                                     
#2  0x00007ffff771c951 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6                                                                                                                                                                 
#3  0x00007ffff772847c in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6                                                                                                                                                                 
#4  0x00007ffff77284e7 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6                                                                                                                                                   
#5  0x00007ffff7728799 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6                                                                                                                                                        
#6  0x00007ffff771f25e in std::__throw_logic_error(char const*) () from /lib/x86_64-linux-gnu/libstdc++.so.6                                                                                                                              
#7  0x00007ffff7c5f51b in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*> (this=this@entry=0x7fffd80077e8, __beg=0x0,                                                                
    __end=0x7ffff71395b0 "\363\017\036\372H\211\367\351\264l\376\377\017\037@") at /usr/include/c++/9/bits/char_traits.h:365                                                                                                              
#8  0x00007ffff7c60495 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct_aux<char*> (__end=<optimized out>, __beg=<optimized out>, this=0x7fffd80077e8)                                    
    at /usr/include/c++/9/bits/basic_string.h:936                                                                                                                                                                                         
#9  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*> (__end=<optimized out>, __beg=<optimized out>, this=0x7fffd80077e8)                                                              
    at /usr/include/c++/9/bits/basic_string.h:266                                                                                                                                                                                         
#10 std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string (__str=<Fehler beim Lesen der Variable: Cannot create a lazy string with address 0x0, and a non-zero length.>,                          
    this=0x7fffd80077e8) at /usr/include/c++/9/bits/basic_string.h:451                                                                                                                                                                    
#11 karto::Name::Name (rOther=..., this=0x7fffd80077e0) at /home/sven/robot_ws/src/slam_toolbox/lib/karto_sdk/include/karto_sdk/Karto.h:407                                                                                               
#12 karto::SensorData::SensorData (this=0x7fffd8007780, rSensorName=...) at /home/sven/robot_ws/src/slam_toolbox/lib/karto_sdk/src/Karto.cpp:122                                                                                          
#13 0x00007ffff7e3fc58 in karto::LaserRangeScan::LaserRangeScan (rRangeReadings=std::vector of length 720, capacity 1024 = {...}, rSensorName=..., this=0x7fffd8007780)                                                                   
    at /home/sven/robot_ws/src/slam_toolbox/include/../lib/karto_sdk/include/karto_sdk/Karto.h:5420                                                                                                                                       
#14 karto::LocalizedRangeScan::LocalizedRangeScan (rReadings=std::vector of length 720, capacity 1024 = {...}, rSensorName=..., this=0x7fffd8007780)                                                                                      
    at /home/sven/robot_ws/src/slam_toolbox/include/../lib/karto_sdk/include/karto_sdk/Karto.h:5422                                                                                                                                       
[... cut ...]
For your Question if the map was properly serialized. I only can assume yes, because there were no crashes and i didn't get an error. This was the response from the service call. So it looks like it was correct.
`ros2 service call /serialize_map slam_toolbox/srv/SerializePoseGraph "{filename: 'map_test'}"
waiting for service to become available...
requester: making request: slam_toolbox.srv.SerializePoseGraph_Request(filename='map_test')

response:
slam_toolbox.srv.SerializePoseGraph_Response()

`

SteveMacenski, 2021-03-04:
This doesn't really tell me a whole lot to work with unfortunately. There's nothing clear in that that's problematic to me

wienans, 2021-03-04:
Okay, so I assume if you use the map provided above https://github.com/SteveMacenski/slam_toolbox/files/5977879/map.zip you can’t reproduce the issue and it is loading without a problem?

SteveMacenski, 2021-03-04:
>   8755686 (i think this is right, commit from 5 Jan)

For me to know from source, I'd have to know the exact git hash to test with in case some of the recent changes to the core library has made them incompatible. I'd really need a rosbag of data used to generate that serialized file I can test serializing / deserializing. Unfortunately, if the issue is the serialization, only looking at those files won't help me look into the issue. If the issue is deserialization and versioning related, I could spin in circles for weeks. This isn't enough information about your issue to have me start looking into it

SteveMacenski, 2021-03-04:
I've tested with another dataset and I also see this issue, it doesn't look to relate to the actual deserialization process (e.g. I see the map update after calling this and before the crash). I will begin to investigate later today or tomorrow. Hopefully its something simple that I can resolve the same day.

SteveMacenski, 2021-03-04:
Unsurprisingly, its in the same place `getLocalizedRangeScan` when creating the `LocalizedRangeScan * range_scan = new LocalizedRangeScan(...)`. This is interesting because this is populating a new laser range scan, not when trying to access a new one. That's good in terms of serialization issues that its not when trying to access the graph (that would be much more complex). 

Instead, it looks like `laser->getName()` is what's crashing, which while also concerning, is much _less_ concerning. The graph seems to be loading OK from file since the actual map does appear for me and only crashes on the next iteration when trying to access the laser info. 

Its not immediately clear to me why the laser object seems to have issues in ROS2 but not ROS1. Maybe serialization issue with that object? But we `getLaser` far before it gets here so it should augment the current laser if there's a bogus entry. I print the laser in `getLaser` and  also crashes there -- so that database object may be bad from serialization. I also tried to print just some metadata like the min range and that also crashes so it doesn't look like the laser object is valid at all.

At the end of `loadSerializedPoseGraph` I printed the `dataset_->GetLasers()` names before we re-register the sensor, so this is the freshest you can get next to deserialization and  get a legit print out `Custom Described Lidar`, so I think that means the actual serialized file is A-OK (which is great great news). Its even valid after the `dynamic_cast`s (damn, I was expecting that to be the issue place).

Ooooooooooooh I see, I removed [this block](https://github.com/SteveMacenski/slam_toolbox/blob/melodic-devel/slam_toolbox/src/slam_toolbox_common.cpp#L665-L687) in the ROS2 port which as part of it populated the [`lasers_` entry](https://github.com/SteveMacenski/slam_toolbox/blob/melodic-devel/slam_toolbox/src/slam_toolbox_common.cpp#L676-L677) which right now isn't touched at all in this process. But `getLaser` should still trigger an update because its not in `lasers_`... mhm. I did a print and its **not** triggering for a new laser, so that seems like our issue.

That would make sense since the `lasers_[]` would return a [null-ish](https://en.cppreference.com/w/cpp/container/map/operator_at) laser object that isn't truly null for a quick test, but in accessing any field it fails (which is what we see).

-----

Anyhow, that's my 90 minute first look at this. I think I know where the issue is, we need to populate `lasers_` with the serialized laser or figure out why `getLaser()` isn't calling an update after the serialization case.

SteveMacenski, 2021-03-05:
@wienans see 2 PRs above for ROS2 and Foxy. Please test and verify for me it fixes what you're seeing as well and seems reasonable. I think this should resolve it. 

I also today finished the rviz panel port, so it should make your life even easier testing!

wienans, 2021-03-05:
Hey @SteveMacenski thank you very much for your fast work.  As i am now awaking and need to go to work I can start to test only in around 12 hours after I am back home. I will update you.

SteveMacenski, 2021-03-05:
No worries, sorry it took me so long to get to this

SteveMacenski, 2021-03-06:
I'm closing, tested again and fixed the issue - but please do follow back here to make sure this resolved it or we can reopen

closed 2021-03-06 by SteveMacenski

wienans, 2021-03-06:
Verified. Fixed the issue for me too.
Thanks appreciate your work.