SteveMacenski/slam_toolbox #331: [ROS2] SLAM mapper parameters not handled propperly

PGotzmann, 2021-02-24:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - from source: slam_toolbox foxy-devel
- ROS Version
  - ROS2 foxy
- Version or commit hash:
  - 875568697689774af3f9c65dae91c0028b98d798
- Laser unit:
  - gazebo plugin

#### Steps to reproduce issue
- add an additional log call for one of the parameter setters in `SMapper::configure()` 
e.g. add  ` RCLCPP_INFO(node->get_logger(), "Setting minimum_travel_distance to %f", minimum_travel_distance);` before https://github.com/SteveMacenski/slam_toolbox/blob/875568697689774af3f9c65dae91c0028b98d798/src/slam_mapper.cpp#L110
- run slam_toolbox in sync/async/localization mode/... with a value other than the default value for the selected parameter (e.g. minimum_travel_distance = 1.5).

#### Expected behavior
Provided parameters in launch/yaml file should be passed to Graph SLAM mapper.

#### Actual behavior
Parameters are declared correctly and show their correct values when inspected (e.g. with `ros2 param get`), but only the default values are passed in `SMapper::configure()` https://github.com/SteveMacenski/slam_toolbox/blob/875568697689774af3f9c65dae91c0028b98d798/src/slam_mapper.cpp#L88

#### Additional information
E.g. see https://github.com/SteveMacenski/slam_toolbox/blob/875568697689774af3f9c65dae91c0028b98d798/src/slam_mapper.cpp#L91:L96 or one of the following parameter declarations in this file.
```
  bool use_scan_matching = true;
  if (!node->has_parameter("use_scan_matching") &&
    node->declare_parameter("use_scan_matching", use_scan_matching))
  {
    mapper_->setParamUseScanMatching(use_scan_matching);
  }
```
According to [https://github.com/ros2/rclcpp/blob/a2004f8369b4407975cd227718047107527a41b0/rclcpp/include/rclcpp/node.hpp#L308](https://github.com/ros2/rclcpp/blob/a2004f8369b4407975cd227718047107527a41b0/rclcpp/include/rclcpp/node.hpp#L308) there is no implementation of `declare_parameter` taking a reference to the  internal param variable and returning a bool and like it is assumed here.
The return value of `declare_parameter()` used here in a binary condition actually matches either the provided or default value.
Thus, it depends on the (implicitly casted) param value if the respective setter function is called.

To actually have the desired effect the code needs to be changed to something like this:

```
  bool use_scan_matching = true;
  if (!node->has_parameter("use_scan_matching")) {
    node->declare_parameter("use_scan_matching", use_scan_matching);
  }
  node->get_parameter("use_scan_matching", use_scan_matching);
  mapper_->setParamUseScanMatching(use_scan_matching);
```

I am not fully aware of all parameter related API changes of previous ROS2 versions, but I think this at least should also effect `eloquent` an `dashing` branch.

SteveMacenski, 2021-02-25:
I believe you may be correct, I did this around Crystal so perhaps the meaning of a failed param `get` has changed. it always had the `ParameterValue` return type but I thought for some reason I tested this and the empty type evaluated to false at one point. 

A PR to update these would be appreciated! Thanks for catching this

PGotzmann, 2021-02-25:
I have opened a PR with the required changes for the  foxy-devel branch.

I have not been actively contributing to a ROS project before, so I am not sure I got everything right. So please just let me know if there is anything that needs changing.

SteveMacenski, 2021-03-02:
Merged :smile:

SteveMacenski, 2021-03-02:
@PGotzmann if you felt compelled to do dashing as well, that would be nice, but not required

closed 2021-03-02 by SteveMacenski

PGotzmann, 2021-03-02:
@SteveMacenski I just had a look at `dashing-devel`. Looks like the branch is already a little behind. Mostly formatting and some small things like a added `<#include <memory>`.

If it is of any use I can still create a PR (If you think it is no just say so :wink: ) Otherwise, would you like me to just apply the patched part or the file include the minor changes from `foxy-devel`?

SteveMacenski, 2021-03-02:
It would be good to have it fixed, but I wouldnt worry about formatting