SteveMacenski/slam_toolbox #333: Initial scan is not added to localization buffer/ not removed from graph

PGotzmann, 2021-02-26:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - from source: slam_toolbox foxy-devel
- ROS Version
  - ROS2 foxy
- Version or commit hash:
  - 875568697689774af3f9c65dae91c0028b98d798
- Laser unit:
  - Gazebo plugin
  
#### Steps to reproduce issue
- run `slam_toolbox_localization_node` with a prerecorded map and a given initial pose
- re-localize the robot by publishing to `\initialpose`
- introspect the graph size before and after calls to  `\initialpose` (e.g. subscribe to `graph_visualization` topic)

#### Expected behavior
In localization mode all nodes added at runtime (including initial poses) are expected to be removed from the graph when the expire from the localization buffer.

#### Actual behavior

If an initial position is specified at startup or a new pose estimate is published to `\initialpose` , the `processor_type_` is set to `PROCESS_NEAR_REGION` in 
`LocalizationSlamToolbox::addScan()` ([slam_toolbox_localization.cpp#L125](https://github.com/SteveMacenski/slam_toolbox/blob/875568697689774af3f9c65dae91c0028b98d798/src/slam_toolbox_localization.cpp#L125)) only for the current scan. In this case, however, the scan is only added to the graph, not to the localization buffer, as would have been the case with `PROCESS_LOCALIZATION` ([Mapper.cpp#L2723](https://github.com/SteveMacenski/slam_toolbox/blob/875568697689774af3f9c65dae91c0028b98d798/lib/karto_sdk/src/Mapper.cpp#L2723)).
As a result this node will not expire from the graph and the size of the graph increases with each call to `\initialpose` thereby polluting the graph with (potentially mislocalized) scans.

#### Additional information

This also effects #116 where the introduced `ClearLocalizationBuffer()` ([initial_pose_reset_eloquent Mapper.cpp#L2910](https://github.com/SteveMacenski/slam_toolbox/blob/378870f0b078e2c9bb5d5b1ce26b09412a75269b/lib/karto_sdk/src/Mapper.cpp#L2910)) function also fails to remove this initial nodes from the graph.

**Here is a short log file with some additional log messages to show this behavior more clearly:**

_Note that this log was created already including `initial_pose_reset_eloquent` branch which introduces the clear buffer feature, but this issue exists independent of this feature._

The initial number of 116 nodes here in this log is the sum of `map_graph_size` (111) + `initial_scan_node` (1) + `localization_buffer_nodes` (4)

When a new pose is received on `\intitialpose` the nodes in the localization buffer are cleared from the graph, a new scan is processed in `PROCESS_NEAR_REGION` mode and added to the graph. The localization buffer, thereby, remaining empty.
One can see that the actual graph size increases with each of the two calls to `\initialpose` to its final count of 114.

```
1614161336.976479916: [slam_toolbox] [INFO]    Graph size: 116						
Clear localization buffer called: initial size = 4
Clear localization buffer finished: final size = 0
1614161338.766982707: [slam_toolbox] [INFO]    LocalizePoseCallback: Localizing to: (-2.66 -7.28), theta=0.74
1614161338.799768880: [slam_toolbox] [INFO]    Processing new scan...
1614161338.799879957: [slam_toolbox] [INFO]    Switching to processor type PROCESS_NEAR_REGION
ProcessAgainstNodesNearBy...
MapperSensorManager::AddScan called
MapperSensorManager inserted to ScanManager
MapperSensorManager inserted to MapperSensorManager
1614161338.898467484: [slam_toolbox] [INFO]    Added node to solver: new size 113
1614161341.864104330: [slam_toolbox] [INFO]    Graph size: 113
1614161346.879470434: [slam_toolbox] [INFO]    Graph size: 113
1614161351.971816252: [slam_toolbox] [INFO]    Graph size: 113
Clear localization buffer called: initial size = 0
Clear localization buffer finished: final size = 0
1614161355.102478391: [slam_toolbox] [INFO]    LocalizePoseCallback: Localizing to: (-7.97 2.99), theta=0.49
1614161355.113616633: [slam_toolbox] [INFO]    Processing new scan...
1614161355.113751172: [slam_toolbox] [INFO]    Switching to processor type PROCESS_NEAR_REGION
ProcessAgainstNodesNearBy...
MapperSensorManager::AddScan called
MapperSensorManager inserted to ScanManager
MapperSensorManager inserted to MapperSensorManager
1614161355.201269890: [slam_toolbox] [INFO]    Added node to solver: new size 114
1614161356.905400306: [slam_toolbox] [INFO]    Graph size: 114
1614161361.964081601: [slam_toolbox] [INFO]    Graph size: 114
```

_Originally posted by @PGotzmann in https://github.com/SteveMacenski/slam_toolbox/issues/116#issuecomment-785136361_

SteveMacenski, 2021-03-02:
100% an issue, that was an oversight on my part. So when you process near region in localization mode, the number of nodes in your graph increases by 1 for the session because `PROCESS_NEAR_REGION` adds to the  graph and not the localization buffer to remove later. It does **not** modify the underlying graph data for other sessions (just to be clear) but only the current one in RAM. 

What seems to me to be the lowest-friction way of dealing with this is to add a `bool` argument to `ProcessAgainstNodesNearBy` which defines whether the node should be inserted into localization scan vector buffer or not for clearing. I think for localization the answer is "yes" to insert it, but for `ProcessAgainstNodesNearBy` when restarting a SLAM session, there's no meaningful interpretation of the localization scan buffer to clear to be adding to. 

Do you agree? Does that seem reasonable?

PGotzmann, 2021-03-02:
Sounds reasonable to me and should require only minimum changes.
There is no real benefit in storing the "localization" mode internally only for this one case, so I think an additional parameter should be fine. I can prepare a PR with that if you like.

Just to be sure, there is no other valid processor type besides `PROCESS_LOCALIZATION` and `PROCESS_NEAR_REGION` that should be able to add something to the localization buffer, right?
Can we also assume that the buffer has just been cleared when we are in `PROCESS_NEAR_REGION` mode or should we include a check on the buffer size too for consistency?

I would thereby also fix the off-by-one error of the buffer size we talked about [here](https://github.com/SteveMacenski/slam_toolbox/issues/116#issuecomment-785136361). For me the most natural solution would be to just move the check _after_ adding the new scan vertex in [Mapper.cpp#L2841-L2844](https://github.com/SteveMacenski/slam_toolbox/blob/foxy-devel/lib/karto_sdk/src/Mapper.cpp#L2841-L2844). This should also be robust in case someone really sets the buffer size to zero.
Does this seem reasonable to you?

PGotzmann, 2021-03-02:
I created a draft PR #343 for this, but I will only have time tomorrow to test this in detail.
If you already have any comments on that, just let me know.

SteveMacenski, 2021-03-02:
> Just to be sure, there is no other valid processor type besides PROCESS_LOCALIZATION and PROCESS_NEAR_REGION that should be able to add something to the localization buffer, right?

closed 2021-03-09 by SteveMacenski