SteveMacenski/slam_toolbox #334: Localization performances get worst overtime

kenza-ifollow, 2021-03-01:
**Required Info:**

- Operating System:
  - Ubuntu 18.04.5 LTS
- Installation type:
  - from source : slam_toolbox melodic-devel
- ROS Version
  - ROS1
- Version or commit hash:
  - 5a8c76f (release 2.3.0)
- Laser unit:
  - Sick nanoScan 3
  
#### Steps to reproduce issue

 - `roslaunch slam_toolbox localization.launch` with a given map and localize the robot using /initialpose 
 
 - run slam_toolbox's localization on the same path, back and forth. The path is three long corridors in a "Z" shape. The total length of these corridors is about 130 m. The speed of the robot used is under 1 m/s and the environment is dynamic (a 
type of warehouse with moving objects, people, etc...)
[image]

 - run the localization in the environment, which is pretty dynamic 

#### Expected behavior

Good localization with uniform performances over time

#### Actual behavior

Localization is really good during ~ 20 minutes and then it becomes impossible to re-localize the robot using /initialpose. If the slam_toolbox's node is killed and then relaunch at the same place with no other changes, re-localization is possible 
[image]

#### Additional information

Several leads have been investigated to try to solve this problem : 

- **Memory consumption** : slam_toolbox's memory usage has been monitored along the way but the usage seems coherent and stable (around 1% of the total memory capacity). The size of the posegraph is also correct (around 800 points) and does not looks like it is increasing 

- **CPU consumption** : slam_toolbox's cpu usage has been monitored along the way. Noticeable peaks have been noticed in the most changed areas (with respect to the .data .posegraph map), around 4 cores are used during 10/15 seconds. 

- **Mapping process** : mapping has been done by passing several times in the same corridors and the passages are overlapping (a lot of points are really close to each others). Could it be possible that this has an impact on the localization's performances ? 

 The configuration used is : 
```
Plugin params
solver_plugin: solver_plugins::CeresSolver
ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
ceres_preconditioner: SCHUR_JACOBI
ceres_trust_strategy: LEVENBERG_MARQUARDT
ceres_dogleg_type: TRADITIONAL_DOGLEG
ceres_loss_function: None

ROS Parameters
 odom_frame: odom
 map_frame: map
 base_frame: base_footprint
scan_topic: /scan
mode: localization 

debug_logging: false
throttle_scans: 1
transform_publish_period: 0.02 #if 0 never publishes odometry
map_update_interval: 5.0
resolution: 0.05
max_laser_range: 40.0 #for rastering images
minimum_time_interval: 0.5
transform_timeout: 0.2
tf_buffer_duration: 30.
stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps

General Parameters
use_scan_matching: true
use_scan_barycenter: true
minimum_travel_distance: 0.2
minimum_travel_heading: 0.1
scan_buffer_size: 10
scan_buffer_maximum_scan_distance: 10
link_match_minimum_response_fine: 0.1
link_scan_maximum_distance: 1.5
do_loop_closing: true
loop_match_minimum_chain_size: 3
loop_match_maximum_variance_coarse: 3.0
loop_match_minimum_response_coarse: 0.35
loop_match_minimum_response_fine: 0.45

Correlation Parameters - Correlation Parameters
correlation_search_space_dimension: 0.8
correlation_search_space_resolution: 0.05
correlation_search_space_smear_deviation: 0.3

Correlation Parameters - Loop Closure Parameters
loop_search_space_dimension: 7.0
loop_search_space_resolution: 0.05
loop_search_space_smear_deviation: 0.3
loop_search_maximum_distance: 7.0

Scan Matcher Parameters
distance_variance_penalty: 0.5
angle_variance_penalty: 1.0

fine_search_angle_offset: 0.00349
coarse_search_angle_offset: 0.43
coarse_angle_resolution: 0.0349
minimum_angle_penalty: 0.9
minimum_distance_penalty: 0.5
use_response_expansion: true
```

SteveMacenski, 2021-03-01:
Have you printed to see if the initial pose message is being received and processed? You say after 20 minutes it stops localizing as well, but does it recover if you keep moving around? If it doesn't recover, we should look into it more. If it does, then that's potentially part of the technique if you ran into a patch of significant odometry drift and the local window was too large so it took a few updates to clear out that bad data chain. I won't say that this is an end-all-be-all solution suited for every person. This really does work best with good, regular odometry where you don't expect substantial slippage and you're just dealing with typical drift.

kenza-ifollow, 2021-03-03:
@SteveMacenski , first of all, thank you very much for your answer and sharing this repository. Here are some observations we have made on this issue : 

- In the situation that I have described above, /initialpose was being processed (apparition of the LocalizePoseCallback logs) and moving around would not help with relocalization.
- Based on your suggestion, we have tried to localize the robot on the same path with a better odometry (without remapping though) and have witnessed the same issue.
- We have also that by **disabling loop closure**,  the robot relies on odometry only to localize itself and in this case, we do not have this issue anymore.
- In addition, we have seen that you push a new feature on melodic to clear the localization buffer while giving a /initialpose and we have tried it, without any improvement on this issue.

We have investigated our code and realized that the problem might come from the fact that the posegraph has been shifted along the path.
Indeed, during the past months, we had the [same issue as the one described here](https://github.com/SteveMacenski/slam_toolbox/issues/285) : **because Karto through loop closures slightly modify the posegraph which might not be aligned with the one which is uploaded when we launch the localization node.** Therefore , the map would get distorded and would not overlap anymore with our static navigation path (with respect to the world/map frame). Because of that, the robot would follow a wrong path in reality while localized in rviz.
To temporary solve this, we disabled the map's update (because it was indicated that the update was only for visualization purposes) but did not take into account that the graph was moving as well as the map.

--> The issue that we see on the image above is therefore the result of the map being shifted with respect to the posegraph : on the non-updated map, the robot is not localized but in reality it is well localized with respect to the updated posegraph.

You said that this could be a result of a bad odometry. Our odometry is acceptable but our path is more than 100 m long with no loop closure possible which affect the quality of the map. Do you have any empirical observations to improve the mapping ?
Do you confirm that our problem could come from that ?
If so, do you see any leads to prevent the modification of the posegraph overtime during localization ? (we would like the posegraph to stay just as it is when we launch slam_toolbox's node)
Thank you again for your time !

SteveMacenski, 2021-03-03:
> We have also that by disabling loop closure, the robot relies on odometry only to localize itself and in this case, we do not have this issue anymore.

Huh, that is interesting. Does this result in any worse localization performance? That could be a very real solution to this issue since localization is just a rolling chain of local scans - if you do not see downsides. If you're only keeping some very local set of localization scans, there's no reason to loop close within the localization rolling chain. Loop closure against the localization chain and the base map... I do not believe... has any significant value to the localization mode. You're still scan matching against the map to correct drift and I could potentially see how updating the pose-graph many times and then removing the constraints could push the optimization problem into another local minima, showing "drift". I'm not sure I see any upside of enabling loop closure in localization mode. 

> We have investigated our code and realized that the problem might come from the fact that the posegraph has been shifted along the path.

That's odd, this is supposed to lock the pose / orientation of the graph in place https://github.com/SteveMacenski/slam_toolbox/blob/melodic-devel/slam_toolbox/solvers/ceres_solver.cpp#L183 since that first node cannot move. That should be preventing drift. I'm now wondering if for some reason that isn't called on deserialization in localization mode, but I'm almost certain that I validated that. So perhaps that's not doing "enough"?

>  To temporary solve this, we disabled the map's update (because it was indicated that the update was only for visualization purposes) but did not take into account that the graph was moving as well as the map.

Can you explain how this solved your problem? How does not publishing `/map` mess with loop closure drift in the pose-graph?  I understand how loop-closures would resolve it. In localization mode, removing loop closures actually seems like a very reasonable action to take. 

> You said that this could be a result of a bad odometry. 

I think you've done enough debugging and the visualizations you provided make me think that's not the case. Disregard those comments. 

> do you see any leads to prevent the modification of the posegraph overtime during localization ?

I think you should try checking that that Ceres first node doesn't move. If it does, we definitely have a problem to look into. I'm also trying to think about ways we can verify what's drifting (is it the pose-graph or the occupancy grid, or what). Have you found some evidence pointing you specifically to the pose-graph? I think that's also likely the case, but I want to verify before  going down that rabbit hole.  If you had a map, I'd try publishing the pose of some nodes far away from the robot (such that it wouldn't be effected by local updates) and publish the XY pose of it and see how / if it drifts with the map to show that the entire pose-graph is drifting. Though from what I talked about above, I could see it happening from a local minima issue.

SteveMacenski, 2021-03-10:
>  changing to Huberloss function helped with this issue as you can this on this image, representing the end of the same path (from rosbag) using Huberloss function

This tells me that this is probably a local minima issue and that this loss function is better at removing the optimization problem's instability around some solution. A change in loss function can do wonders for removing the symptoms of a problem (until it grows to a point it can't since it looks like a legit source of pose error in the system) but its not a solution in it of itself for anything.

Did you try setting the loop closures to false like we mentioned above? Any issues that this came up with when that was set to false? If the optimizer never triggers, then this would never be a problem. This is obviously not a real solution for VSLAM, but for pure localization, I don't actually foresee any downside. I think testing this would be valuable. If you don't see any downside, I can document this feature of localization mode with loop closure and change the default localization parameters and we're done with this issue. 

> I had this problem when I've set size_buffer_scan above 10 in localization but it is not impossible that I've set size_buffer_scan > 10 while mapping, I will try which would explain the crashes, I will try to remap and retest slam_toolbox in this environment to see if I still notice crashes.

Yeah sorry about that, if you leave it at `10`, you'll see the same behavior you were getting before (since that parameter wasn't being read in). We're working on it in the other ticket, but its unrelated to this issue.

PGotzmann, 2021-03-11:
Just a short side note to avoid confusion here:

>Yeah sorry about that, if you leave it at 10, you'll see the same behavior you were getting before (since that parameter wasn't being read in). We're working on it in the other ticket, but its unrelated to this issue.

If I understand this correctly @kenza-ifollow opened this issue for `eloquent` so he should not have been effected by the ROS2 read parameter issue (#331) you mentioned here. Never the less using a `scan_buffer_size` in localization mode greater or equal than the one used for mapping seems to avoid the "mutex lock failed" issue (#355), but let's keep the discussion separated :wink:

kenza-ifollow, 2021-03-11:
@PGotzmann yes indeed I am not using ROS2 so the read parameter issue should not affect me and is not related to this issue :) 

@SteveMacenski 
> Did you try setting the loop closures to false like we talked about above? Did that fully solve the issue? Any side effects? 

I did try without loop closing in localization mode and I could not notice any shift in the posegraph but the localization was really bad, as if it was completely relying on the odometry and never trying to correct the pose of the robot. 

Normally I would be able to record a bag of this behavior next week

SteveMacenski, 2021-03-11:
> @PGotzmann yes indeed I am not using ROS2 so the read parameter issue should not affect me and is not related to this issue :)

But if you used 10 for mapping, you'll still have the exact same issue as #355 in ROS1. So its really "dont use anything for `scan_buffer_size` less than that of which you mapped" for all versions until we figure it out and backport a solution. It was just exposed in ROS2 by that parameter update ticket. 

Without loop closure was bad? I would have expected that without scan matching enabled, but I don't know why without loop closure that would happen. Didn't you mention above what you tested that briefly before and got reasonable results?

SteveMacenski, 2021-03-23:
Any word?

kenza-ifollow, 2021-03-25:
Sorry for the long delay @SteveMacenski , we had a lot of other tests to do on the field. 

To show you what happen when loop closure is disabled in localization, here are some captions of a tests we have made . 

[image]
At the beginning of the test, the robot is perfectly well localized (red arrow) as we can see on the picture above : the lidar'points cloud (purple points) are overlapping with the map. 

The robot navigate at a pretty low speed (less than 1 m/s) for about 30 seconds and everything is still good : I've monitored the tfs between map and odom and it is still published 
[image]

After a while, we can see that there is a small orientation problem, which is not getting corrected : lidars do not match the map anymore on the right 
[image]

This is never getting corrected and less than a minute after the navigation has begun, we have a delocalization 
[image]

I did not get time to dig deeper in it but some leads : 

- In a first iteration, I have changed some mapping's parameters such as scan_buffer_size. Regarding our conversation I thought that it might be the problem so I've rerunned a mapping from a rosbag with new params to prevent pthread issues. Then, I've played the same rosbag with the old and the new map and I've noticed that the delocalization happen quicker in the first case 

- In some places, slam_toolbox's localization without loop closure works well (it seems like the position gets corrected) and it is areas in which I think that odometry is better : could it be that without loop closure, slam_toolbox can only correct position's drift in some extend ? 

Regarding what I have tested that got reasonable results : it was changing ceres_loss_function loss function to Huberloss

SteveMacenski, 2021-03-25:
Got it - so perhaps a combination of the loop closure off + HuberLoss could help? I fully recognize whenever the loss function "fixes" an issue, that means there's another real issue to be fixed and that its just doing a good job at masking it. However, if the optimizer isn't running at all, there should be no reason any of the original graph nodes should move (you can inspect the code yourself to verify that claim) I do not believe. As such, I'm a little at a loss as to where that issue could be showing up, unless the `do_loop_closure` parameter wasn't properly disabling the optimizer. I do not believe there is anywhere else in the code where any object has the ability to change the positions of the nodes to cause the "pose-graph-drift" you've reported.

navidmulla60, 2022-01-05:
we also facing the same issues... so (inside slam_toolbox_localization yaml)  I reduced the map_update_interval to the size of the resolution of the map.... in my case resolution was 0.05 so i modified map_update_interval to 0.05 and i got some better results (although i m facing issues but after setting it to 0.05 ,i noticed better performance)

tugbakara, 2023-07-21:
> To temporary solve this, we disabled the map's update (because it was indicated that the update was only for visualization purposes) but did not take into account that the graph was moving as well as the map.

which parameter did you change to achieve this?

NhatTanXT3, 2023-12-22:
I just want to add more info about the effect of disabling /map update on the localization mode. For my use case (ROS1 noetic branch):

When enabled, the updateMap() function uses 100% 1 core CPU at about 1 sec for my map size ~630 vertices.
From [the publishVisualizations loop](https://github.com/techtilesg/slam_toolbox/blob/ee00cf7cab882b4484ccd9fa18413e237b1bccfa/slam_toolbox/src/slam_toolbox_common.cpp#L320C18-L320C18), it will lock the [main scan processing loop](https://github.com/techtilesg/slam_toolbox/blob/ee00cf7cab882b4484ccd9fa18413e237b1bccfa/slam_toolbox/src/slam_toolbox_localization.cpp#L150C43-L150C43)

So I implemented a dirty change of first_map_only feature that publishes /map one time only. Something like below
[image]

Kaetera, 2024-04-30:
While investigating some TF timing issues on my system, I noticed that the process time took by [addScan()](https://github.com/SteveMacenski/slam_toolbox/blob/94cec982a7f850818187c81295d1212f145efe37/src/slam_toolbox_localization.cpp#L140) increases progressively with time:
Few sec after navigation start, process time is about 150ms
After 30s, it is around 400ms
After 1min, around 600ms
After 1m30, around 1s (and then everything breaks on my system due to obsolete TF timestamps)

Just dropping this here, may be a clue.
Like if something was accumulating somewhere. 
(nb: my system has changed a lot since my last post: more nodes which consume more CPU, higher resolution lidar)