SteveMacenski/slam_toolbox #355: Irregular terminations with boost: mutex lock failed

PGotzmann, 2021-03-08:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - from source: slam_toolbox foxy-devel
- ROS Version
  - ROS2 foxy
- Version or commit hash:
  - 437d64cccddc7d036cb8024028e3861f2e0e8877
- Laser unit:
  - Gazebo plugin
  
#### Steps to reproduce issue
_still working on a better reproducible minimal example, but for now:_

- ensure parameter `scan_buffer_size` is set to a small value (e.g. 3 link in the `mapper_params_localization.yaml`)
- start `localization.launch.py` with a prerecorded map and initial position.
- move around in map randomly

#### Expected behavior
Slam_toolbox keeps publishing localization result as tf and does not throw an exception.

#### Actual behavior

Slam toolbox irregularly terminates with the following exception or just freezes and stops to publish `map`->`odom` tf.

```
terminate called after throwing an instance of 'boost::wrapexcept<boost::lock_error>'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
  ```
This is unfortunately not quite reliably reproducible. 
Exception is thrown approximately 7 out of 10 runs with a short recorded bag file (90sec) of the used test setup.

GDB backtraces:
[backtrace_slam_toolbox_debug.log](https://github.com/SteveMacenski/slam_toolbox/files/6101377/backtrace_slam_toolbox_debug.log)
[backtrace_slam_toolbox_debug_full.log](https://github.com/SteveMacenski/slam_toolbox/files/6101379/backtrace_slam_toolbox_debug_full.log)

#### Additional information

Looks like this may have been indirectly introduced by #332 and might be somehow related to the parameter `scan_buffer_size`. Setting scan buffer size to e.g. 10 (as it always was before #332) does not lead to such exceptions.

SteveMacenski, 2021-03-09:
I'm wondering if you used `10` in your initial mapping. If you used, lets say, 11, does the localization mode change to `11` and above being the magic number? 

Just trying to suss out exactly where this magic number comes from. The 2 places I can think of are the serialized file settings and the other is the processors, but I think the first is the most likely to think about first.

PGotzmann, 2021-03-10:
I had  some time to test this on our robot in the lab today, which showed the exact same behavior as the simulation (values >= 10 work as expected, values smaller than that leading to a `<boost::lock_error>`).

> I'm wondering if you used 10 in your initial mapping. If you used, lets say, 11, does the localization mode change to 11 and above being the magic number?

The maps for both test setups (simulation, real robot) have actually been recorded with `scan_buffer_size=10` as I used the default parameters from https://github.com/SteveMacenski/slam_toolbox/blob/ros2/config/mapper_params_online_sync.yaml.

As you recommended I also tried changing the butter size to another value for recording. Seems you could be actually right about that correlation. At least I have seen the magic number changing accordingly for `scan_buffer_size` 3 and 11.

SteveMacenski, 2021-03-10:
I'll need to look more into it but I'm guessing `m_RunningScans` has something to do with it. Maybe we should check on that size being set correctly? If the running scan buffer size is somehow larger than the localization buffer scan size, then the localization buffer would delete some scans while the running scan buffer might still want it for scan matching. The the mutex issue is really just the first member of the object crashing because it doesn't exist anymore (I did see that a few times before when developing the localization feature; sometimes deleted object appears as mutex issues)

PGotzmann, 2021-03-11:
Your right about `m_RunningScans` having the wrong size. The reason for that is that `m_RunningBufferMaximumSize` is only set if `Mapper::Initialize()` is called, which is not the case if we load a serialized graph. So we end up with the running buffer size of the previous mapping session.
The new parameters are read in again in  `loadSerializedPoseGraph()`  but never make it to the `ScanManager` https://github.com/SteveMacenski/slam_toolbox/blob/9af02a710c7a23e4d3fa527e2f9b6274ed48337e/src/slam_toolbox_common.cpp#L680

We need to update `m_RunningBufferMaximumSize` and `m_RunningBufferMaximumDistance` but I am unsure about the best way to do this. Any thoughts on that? Is there anything else that might need to be reset (e.g. ensure running buffer is empty)?

I am not 100% sure, but i think we might also have a similar issue with the scan matcher parameters.

SteveMacenski, 2021-03-11:
So it sounds like `configure` isn't making it to the scan manager - is that the only object that its not getting to? That would be good to know for debugging. I think the best solution here is to figure out where these parameters are going if they're not making it to where they should be. 

We could hot-patch for just those 2 parameters, but I suspect that there's more than just those 2 parameters having issues if `configure()` isn't getting `scan manager` updated at all

PGotzmann, 2021-03-11:
I now see that I did not make my point clear enough, sorry for that.
`configure` does call all the setters and updates the respective parameters in the `Mapper` object. 
If i understand correctly, the main issue here is that the scan manager parameters are only once passed down in the constructors, but we are using deserialized objects where these values are already set and there just is no other update routine for these parameters. (At least none I can see)

**To be a little more specific here:**
In the standard slam use case a new `MapperSensorManager` object is created here in `Mapper::Initialize()`
https://github.com/SteveMacenski/slam_toolbox/blob/9af02a710c7a23e4d3fa527e2f9b6274ed48337e/lib/karto_sdk/src/Mapper.cpp#L2560
As far as I can see this is the only time the member variables `m_runningBufferMaximumSize` and `m_runningBufferMaximumDistance` are assigned a new value. 

When we load a previously recorded pose graph `Mapper::Initialize()` (naturally) is not called at all, as this would basically delete the map. This however means that these values remain unchanged even if we choose a different `scan_buffer_size`.

So when `RegisterSensor()`is called we are using the old value for our new `ScanManager`
https://github.com/SteveMacenski/slam_toolbox/blob/9af02a710c7a23e4d3fa527e2f9b6274ed48337e/lib/karto_sdk/src/Mapper.cpp#L250

(I am not sure if it is expected to get the same sensor_id if I run slam toolbox on the same robot again. In this case we would probably only use the old `ScanManager` and the parameters there would be unaffected of all changes in `MapperSensorManager` too.)

---

>We could hot-patch for just those 2 parameters, but I suspect that there's more than just those 2 parameters having issues if configure() isn't getting scan manager updated at all

From what I see `karto::ScanManager` is not updated at all but it basically only has these two real parameters.
I think we should have a look on all Objects that would usually be initialized in  `Mapper::Initialize()` https://github.com/SteveMacenski/slam_toolbox/blob/9af02a710c7a23e4d3fa527e2f9b6274ed48337e/lib/karto_sdk/src/Mapper.cpp#L2549:L2567, as this is the step we are kind of "missing" here.

`MapperGraph` should be fine but as I said, I think we might have a similar issue with the `ScanMatcher` parameters that are set here too.

---
I am sure you know the code much better than I do - I barley scratched the surface, so please tell me if I am wrong or missing something.  :wink:

SteveMacenski, 2021-03-11:
Ah, I see, so it is unique to `ScanManager`. I see there are `GetRunningScanBufferSize` functions, perhaps we could add a `Set` version for these as well, and we `Set` as part of our `configure()` function or deserialization process? Seems like a relatively straight forward solution

PGotzmann, 2021-03-12:
I think it is not entirely unique to `ScanManager` but also effects at least the two `ScanMatcher`, that are also only initialized with their parameters at creation.

So from what I have seen so far I would say that the following parameters are probably not fully handled when we start from a previously serialized map: (sorted by the object that finally uses these parameters with their calling structure)
- **ScanManager** (`Mapper::Initialize()` -> `MapperSensorManager::RegisterSensor()` -> `ScanManager()`)
  - scan_buffer_size
  - scan_buffer_maximum_scan_distance
- **Mapper::m_pSequentialScanMatcher** (`Mapper::Initialize()` -> `ScanMatcher::Create()`)
  - correlation_search_space_dimension
  - correlation_search_space_resolution
  - correlation_search_space_smear_deviation
- **MapperGraph::m_pLoopScanMatcher** (`Mapper::Initialize()` -> `MapperGraph()` -> `ScanMatcher::Create()`)
  - loop_search_space_dimension
  - loop_search_space_resolution
  - loop_search_space_smear_deviation

SteveMacenski, 2021-03-12:
Looking over it, I agree, I think those are the problem parameters. `Initialize()` is called on the first `ProcessXYZ` function but we also store `m_Initialized` in the serialized file. 

I think re-setting `m_Initialized` would do the trick to update those parameters, but also I believe create new objects destroying some of the serialized data we require in the `m_pMapperSensorManager` and `m_pGraph`, but its not immediately clear to me that that's 100% the case. 

What we could do instead is add a `InitializeFromFile()` function, which takes an existing `m_pMapperSensorManager`, `m_pSequentialScanMatcher`, `m_pGraph` objects and calls a few new functions we could make `SetScanBufferSize()`, etc. 

Then in the `SMapper::configure`, we simply call `mapper_->InitializeFromFile();`

SteveMacenski, 2021-03-12:
@PGotzmann I know this is unrelated, but I'd appreciate it if you cloned the branch that makes sense for you [in this comment](https://github.com/SteveMacenski/slam_toolbox/issues/281#issuecomment-797739118) and just quickly localized in it to see if you see any errors popping up. This ticket is to fix some issues allowing for multiple laser scanner support and enabling users with 360 lidars to fix some large outstanding issues - it would be a huge help. 

We know this PR fixes issues for mapping, but a user reported that with serialized files generated previous to this patch, while using the patch, there were Ceres warnings in localization mode. I'd like to know if you see this as well since I couldn't reproduce it with my little Nav2 sandbox environment which is not really a production space. If you were also so kind as to re-generate some serialized file from mapping _using_ this patch and test localization on that as well, that would fully close the loop on if you still see those warnings (if you saw them at all) when the serialized and deserialized software versions were the same. 

If you're open to it, tl;dr
- [ ] Run localization mode using this branch on an existing serialized file, note any warnings out of the ordinary
- [ ] Run SLAM mode using this branch and save the serialized file
- [ ] Run localization mode using this branch on this new serialized file, note any warnings out of the ordinary 

I did this for my Nav2 environment and saw no warnings which would indicate to me I can merge this work. But I need a second opinion using more "real" data than my little sandbox world. Thanks if you're able to swing it!

SteveMacenski, 2021-03-23:
@PGotzmann back on topic here, any progress with this?

PGotzmann, 2021-03-24:
@SteveMacenski sorry for keeping you waiting. I've been busy with other issues lately, but I should find time to take a closer look at this in the coming days.

I also think the best way to go forward is to add a `InitializeFromFile()` function to `Mapper`. Updating the parameters should not be too hard. However, I am still unsure if there are any other thing that might need to be updated/reset  in `m_pMapperSensorManager`, `m_pSequentialScanMatcher`, `m_pGraph` apart from the mentioned parameters. It is a little hard to tell just from looking at the code here, what we really want to keep or delete from the previous session.

e.g. I am not sure if `ClearRunningScans` need so be executed for all ScanManagers here. It looks like we only reset it for the current ScanManager in `ProcessAgainstNodesNearBy`od `ProcessAgainstNode`, wheras ClearLocalizationBuffer resets all of them (but is not called when we start from a deserialized map).
I'll have to look into it a little more to be sure, but if you happen to have any ideas on what we should really keep or remove here, I'd appreciate it.

SteveMacenski, 2021-03-24:
I think just changing the params for those particular ones should be good enough, but  I think it'll become clear if there's an issue while doing it. 

(right now, its only a single laser, so there's only 1 scan manager)

PGotzmann, 2021-03-29:
@SteveMacenski just wanted to give you a short update on this:
I have created a first draft to solve this problem, but unfortunately there still seem to be some issues with this solution that prevent me from opening a MR so far. If you still want to have a look at it, here is the related commit: https://github.com/PGotzmann/slam_toolbox/commit/d44f801812ac6afdf7d8b038e8a30e60de052581

The updates for the buffer related parameters are quiet straight forward, however, it turns out it is a little more difficult for the `ScanMatcher`. These parameters are only used in the constructor to calculate the actual size of the correlation grid and can not just be updated. I first thought that it should be save to delete the old scan matchers and create new ones with the correct size, but since I see some unexpected behavior I am not so sure anymore.

Is there something inside the ScanMatchers that needs to be persisted? I thought of it more like temporary data for each iteration of the correlation. (sorry if that should be obvious from the algorithm)

The actual error I am getting is a
```
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
```
(gdb bactrace: [bt_slam_tb_bad_alloc.log](https://github.com/SteveMacenski/slam_toolbox/files/6224153/bt_slam_tb_bad_alloc.log))

If that happens to means anything and you or if you have a specific piece of code in mind to look at pleas let me know, otherwise i will keep digging :wink: 

There is also a small good thing to report: At very least the original `mutex lock failed` error seems to be gone and the mentioned parameters no have the desired values. (but what good is that if it breaks the rest :see_no_evil: )

PGotzmann, 2021-03-30:
> terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
  
 I think I have found the real reason for that by now. Looks like this error is independent after all.

From what I can tell, the root cause here is that `SlamToolbox::updateMap()` accesses scans while the might be removed from the buffer by `SMapper::clearLocalizationBuffer()` in another thread.
 Adding a lock on  `smapper_mutex_` before each call on `SMapper::clearLocalizationBuffer()` does the trick for me.

If that sounds reasonable to you, I can include this fix in this PR.

SteveMacenski, 2021-03-30:
Sounds reasonable, though I think it should be properly locked already, if it fixes the issue I'll review it

PGotzmann, 2021-03-31:
I had some time to test this in the field today. The results were basically identical with the simulation.
I do not see any of the parameter related issues anymore, but as with the simulation, I eventually get a `std::bad_alloc`  if I repeatedly localize the robot (without the additional mutex locks). With the added locks (https://github.com/SteveMacenski/slam_toolbox/pull/371/commits/aab910110252d732dbcfd6d70f7ad3cc2b570b90) I am unable to reproduce this error anymore within a reasonable number of `/initialpose` calls.

>Sounds reasonable, though I think it should be properly locked already, if it fixes the issue I'll review it

Took another look at the code here. There is proper locking for all other cases I can think of, but I still do not see this happening for `clearLocalizationBuffer()`. If you are sure this is done somewhere else, please let me know where to look.

SteveMacenski, 2021-03-31:
OK, sounds good to me! Of course a service is async from the mapping processing :facepalm: I don't know why I didn't see that yesterday