SteveMacenski/slam_toolbox #366: karto-scan-visualization frame does not take mounting position into account

NikolasE, 2021-03-16:
**Required Info:**

- ROS Version: ROS1/Noetic

#### Steps to reproduce issue
Use scanner that looks to the side, activate the interactive mode and select an interactive marker

#### Expected behavior
Visualized laserscan fits quite well to the map

#### Actual behavior
Visualized laserscan has wrong orientation (and maybe also position)

The laser-scanner is mounted so that it looks 90deg to the right (along robots negative y-axis). But it looks as if the frame 
[image]

If the frame is manually rotated by 90deg to the right, the laser fits 
[image]

The frame is computed in void LoopClosureAssistant::processInteractiveFeedback and I think only the robots pose is used here (via solver_->GetNodeOrientation)

SteveMacenski, 2021-03-16:
I believe the updates in PRs #325 - #362 might fix this for you if we use the robot pose in the edges / nodes. Please try with a serialized file generated from one of those PR branches? 

If that doesn't fix, a PR to resolve would be nice!

SteveMacenski, 2021-03-23:
@NikolasE did it solve your issue? They've also been merged

NikolasE, 2021-03-24:
I can check next week, my mapping setup is in the office am I'm not ;)

SteveMacenski, 2021-04-13:
How'd it go?

Wimll, 2021-12-03:
@NikolasE I am facing the same issue on my setup w.r.t. the position of the laser scan, did you solve this in the end?

closed 2025-06-03 by SteveMacenski