SteveMacenski/slam_toolbox #369: Listen to a namespaced scan topic ?

guilyx, 2021-03-19:
Hi there, I'm using ROS2 Foxy (20.04, Xavier, Differential Robot, Sick Scanner - but has nothing to do with the robot configuration I guess) and got to make the slam_toolbox up and running smoothly without namespaces.
I'm now introducing a second robot and started to namespace everything. I changed the scan_topic parameter from /scan to /myrobot/scan but if I launch my slam launch file, I still see /scan as the topic slam_toolbox listens to.

Any clue ?

I can send the launch file if needed, maybe I got something wrong there. Namespaces did confuse me all day. Maybe I missed something.

Thanks.

SteveMacenski, 2021-03-19:
Is your launch file / config file set matching properly with ROS2 param namespaces? It might not be being read in. Please consult ROS2 documentation on this. The scan topic is [clearly parameterized](https://github.com/SteveMacenski/slam_toolbox/blob/9af02a710c7a23e4d3fa527e2f9b6274ed48337e/src/slam_toolbox_common.cpp#L143) and would receive proper parameter updates.

guilyx, 2021-03-20:
Hey there, thanks for the response @SteveMacenski. Wasted your time though, for some reason the slam_toolbox wasn't built from where it was supposed to and wasn't getting the param file I was changing the topics from, my bad.

All good now, closing.

closed 2021-03-20 by guilyx