SteveMacenski/slam_toolbox #376: Sync input & output timestamps  (for e.g. /scan and /map)

ohsirawat, 2021-04-08:
#### Feature description
Sync input & output timestamps  (/scan and /map)

#### Implementation considerations
Offline slam tuning by `ros2 bag play` is publishing /tf and all topics data in old timestamps when SlamToolbox is going to publish present timestamps by `this->now()`, this caused TF tree discontinuous and rviz2 can't visualize `/map --> /odom`

SteveMacenski, 2021-04-08:
You shouldn't have an issue if you have `use_sim_time` set to true for the slam toolbox node, then `now()` would use the ROS clock rather than the system clock. Can you try that first?

While your suggestion seems very rational, I think it could cause some serious issues downstream for users (hence why I'd like you to try simulation time first). The SLAM system in ROS needs to "lead" the system clock by a little bit in order for no TF transform into future issues to come up. By post dating this, depending on the update rate of the SLAM system and the lidar itself, this could introduce transient issues where a valid TF tree couldn't be found in a real system at run-time.

ohsirawat, 2021-04-09:
I have tried set `use_sim_time:=true` it still doesn't work, I think because there is no ROS clock publishing. And `rosbag2` hasn't had `clock/simtime` feature yet according to https://github.com/ros2/rosbag2/issues/99 and 
 https://github.com/ros2/rosbag2/pull/675

In my opinion, that is the purpose of `transform_tolerance` existence. It shouldn't be any issue if config parameters is reasonably correct (lidar update rate match transform_tolerance). But I agree with you, it might be issues downstream for users.

SteveMacenski, 2021-04-09:
first off, try in the params file, I dont recall off hand if the launch files have cli launch configurations for simtime. 

This seems like a rosbag issue then that should be resolved there. 

I need to do a little looking into other localization systems in ros to see what best practices are.

ohsirawat, 2021-04-12:
As I had a quick look at `navigaion2 AMCL` they also use `laser_scan->header.stamp` [link1](https://github.com/ros-planning/navigation2/blob/0ade9225278b89e9bf1fe5b56b970518b670fd8d/nav2_amcl/src/amcl_node.cpp#L730-L732) [link2](https://github.com/ros-planning/navigation2/blob/0ade9225278b89e9bf1fe5b56b970518b670fd8d/nav2_amcl/src/amcl_node.cpp#L1015-L1020)

If robot stops but lidar is still continue publishing, there should be no issue because TF timestamp is equal latest scan timestamp. (If use processesd scan timestamp, it might be)

SteveMacenski, 2021-04-13:
OK - I reviewed the PR and just need some small updates

closed 2021-04-16 by SteveMacenski