SteveMacenski/slam_toolbox #382: GPS Integration

nyxaria, 2021-04-15:
Are there any plans of having GPS integration into the SLAM algorithm? Something like what Cartographer has, where you can simply publish gps messages onto a topic and the algorithm uses it as another data source.

Thank you for all your work!

SteveMacenski, 2021-04-15:
No, there are no plans for that. Look at robot_localization for loosely coupled localization fusion.

closed 2021-04-15 by SteveMacenski

Pallav1299, 2021-04-19:
> No, there are no plans for that. Look at robot_localization for loosely coupled localization fusion.

@SteveMacenski, does that imply that we cannot use another source that gives us the tf between map->odom frames (e.g. two robot_localization instances)?

MrOCW, 2022-06-10:
@SteveMacenski @Pallav1299 Since slam_toolbox doesnt publish the pose/odom, we aren't able to fuse it into global ekf which publishes map->odom->base_link and use that odom generated?