SteveMacenski/slam_toolbox #384: Mirrored map as compared to real environment

Pallav1299, 2021-04-19:
**Required Info:**

- Operating System:
  - Ubuntu 20.04.02
- Installation type:
  - source
- ROS Version
  - ROS1
- Version or commit hash:
  -  https://github.com/SteveMacenski/slam_toolbox/tree/noetic-devel
- Laser unit:
  - 3D RS-16 LIDAR using pointcloud_to_laserscan
  
#### Steps to reproduce issue

I am using 3D LIDAR as the LaserScan source.
One robot_localization instance for fusing wheel_odometry and IMU data for the odom->base_link tf

Robot_localization EKF odometry visialization in RVIZ:
https://user-images.githubusercontent.com/32015105/115235780-30914780-a138-11eb-8e9e-da8acd889632.mp4

Slam_toolbox with robot_localization EKF odometry visualization in RVIZ:
https://user-images.githubusercontent.com/32015105/115235887-50287000-a138-11eb-8e7f-04d1650a1ca9.mp4

#### Expected behavior
A 2D-occupancy gridmap similar to real environment.

#### Actual behavior
I am getting a mirrored map.
#### Additional information
angle_min = -1.5708
angle_max = 1.56731
range_max = 10
I am using the clock from rosbag using use_sim_time set to true.

Rosbag: https://drive.google.com/file/d/1zxbVXn83B71oqNanvIaeGVSOcTHBN7yx/view?usp=sharing

Pallav1299, 2021-04-20:
Upon changing the base_frame to the lidar frame in SLAM_toolbox and robot_localization, everything works fine. Seems to be some tf related issue to me. Any guesses?

SteveMacenski, 2021-04-20:
Your issue is probably a rotated lidar you didn't properly insert into your URDF on your robot or your laser is spinning the wrong direction -- closing since not a SLAM Toolbox issue. You may want to ask / search on ROS Answers

closed 2021-04-20 by SteveMacenski