SteveMacenski/slam_toolbox #385: Map merging issue

Pallav1299, 2021-04-26:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source
- ROS Version
  - ROS1
- Version or commit hash:
  - https://github.com/SteveMacenski/slam_toolbox/tree/noetic-devel
- Laser unit:
  - Hokuyo UTM-30LX laser scanner
  
#### Steps to reproduce issue

I am trying to use the map_merging feature by creating two maps from a single rosbag. Afterwards I add these submaps upon launching the map_merge_kinematic node.
The manual alignment isn't affecting the merged map.

#### Expected behavior
Map with aligned submaps.
#### Actual behavior
Map with misaligned submaps.
#### Additional information
Seems like the alignment data is being lost and the default pose is being used for merging submaps.

Screenshots:
submap_1:
[image]

submap_2:
[image]

manually_aligned submaps:
[image]

merged submaps:
[image]

Pallav1299, 2021-04-28:
Seems to be as if some transformations accumulate over manual alignments. Aligning for the first time works well. The second time, there is some offset from the given transformation(rotation or translation). In my case, I had to apply both rotation and translation. I will try to look into the code to find if it's actually the case.

SteveMacenski, 2021-04-28:
Yeah, that is correct. This was built as a single-time application (assumed you'd be merging `N` maps 1 time) so I could totally believe that this doesn't reset state completely. To do so, you'd have to reset all the variables and probably also unload all of the maps since their associated markers can't really be easily zeroed out from this application

Pallav1299, 2021-04-29:
> To do so, you'd have to reset all the variables and probably also unload all of the maps since their associated markers can't really be easily zeroed out from this application

Can you give more clarity about the same? 

Just to be clear, multiple actions(rotation and translation) are not supported currently in map_merging, right?
Is there a scope for improvement?

SteveMacenski, 2021-04-29:
If you look at the code, I think it should be pretty clear - once merged nothing is reset, it just maintains state. So this was really only built to be executed 1 time per run, not multiple times. There's obviously room for improvement to make it work N times

Pallav1299, 2021-05-04:
> If you look at the code, I think it should be pretty clear - once merged nothing is reset, it just maintains state. So this was really only built to be executed 1 time per run, not multiple times. There's obviously room for improvement to make it work N times

Even for 1 execution per run, I witnessed that merging after a single rotation leads to misalignments in the merged map. 
@SteveMacenski, Did you face this behavior?

closed 2024-06-03 by SteveMacenski