SteveMacenski/slam_toolbox #393: Serialize map function call crashes on ROS Noetic

jelavice, 2021-05-18:
Hi Steve,

thanks for sharing your slam_toolbox. I've been running into some issues with saving the map which causes the node to crash.

I'd highly appreciate if you could give any hints on what might be causing the problem.

- Operating System:
  - Ubuntu 20.04 LTS
- Installation type:
  - from source
- ROS Version
  - Noetic
- Version or commit hash:
  - branch: noetic-devel
  - commit hash: 07190554c987c9e349e5b6a880b2210b19848675
- Laser unit:
  - 3D lidar converted to 2d scan using pointcloud to laserscan node  
#### Steps to reproduce issue
set /use_sim_time to true. Then do:
roslaunch slam_toolbox offline.launch (with the parameters file provided). Play the provided rosbag with --clock option. In the rviz gui, press serialize map button for any time t > 90s in the rosbag.

#### Expected behavior
map gets serialized and saved
#### Actual behavior
the node crashes
#### Additional information
It only crashes for times greater than 90 seconds which roughly when the robot starts revisiting a place. Before that, the serialize map service exits normally. 

Rosbag file, config file and the backtrace can be found here:

https://drive.google.com/drive/folders/10HAM2nrRPAMfd3T7ZDhwODgaU_4mXTlL?usp=sharing

SteveMacenski, 2021-05-18:
It is crashing while you try to save, this has nothing to do with the timestamp or bag. `stack_size_to_use` is the variable to set the stack size memory to use in your system to make sure there's enough stack memory to serialize your session, try increasing this (alot) to see if it solves the issue. If so, then you're just not storing enough stack memory in this process to successfully serialize. 

This bug is pretty clearly a serialization issue - the entire backtrace is internal to boost serialization. What's your CPU?

---

I downloaded your bag, ran offline.launch with the default parameters with your bag and was able to successfully serialize after 90s. I was able to serialize up until the end of the SLAM session without an issue. Using 20.04 / noetic devel source build / offline launch file and the same bag.

jelavice, 2021-05-19:
Hi Steve,

thank you for your help. 

> It is crashing while you try to save, this has nothing to do with the timestamp or bag. `stack_size_to_use` is the variable to set the stack size memory to use in your system to make sure there's [...]

I tried with the default parameters as well and it doesn't seem to crash anymore. Maybe one of the parameters is causing the system to run out of memory during the serialization (e.g. resolution being too fine). I'll try to play with the parameters and the stack size to find the exact reason.

> This bug is pretty clearly a serialization issue - the entire backtrace is internal to boost serialization. What's your CPU?
> 

Intel(R) Xeon(R) CPU E3-1535M v5 @ 2.90GHz, why is this important?

SteveMacenski, 2021-05-19:
> Intel(R) Xeon(R) CPU E3-1535M v5 @ 2.90GHz, why is this important?

Just making sure it wasn't some small platform / embedded system where the clear issue would be an insufficient amount of possible available memory.

jelavice, 2021-05-21:
Hi Steve,

I played a bit with the parameters and it seems that the following combination of parameters below causes the issue:

`loop_search_maximum_distance: 10.0`
`loop_search_space_dimension: 20.0`  
`loop_search_space_resolution: 0.04`

Is this expected behavior? If I understand correctly, these parameters should not influence anything in the serialization so it seems rather weird.

If I don't increase the `loop_search_maximum_distance` I don't get a loop closure and the map doesn't look correct.

SteveMacenski, 2021-05-21:
That’s bizarre. The best I could guess is that size is large (?) enough to have stored some additional data requiring more stack space to serialize. If you increase the stack size to something bonkers, does it work then?

jelavice, 2021-05-27:
Sorry for the late reply. Yes, I tried with the larger stack, but the problem persists.

jaredohmni, 2021-11-18:
Hi Steve! We just ran into this too on melodic, and tracked it down with valgrind. It looks like there's a use-after-free on serialization because you are calling delete [] m_pPoseReponse in ScanMatcher::CorrelateScan but not setting m_pPoseResponse = NULL and subsequently not checking for NULL in the ScanMatcher::serialize() call:

```  /**
   * Serialization: class ScanMatcher
   */
  friend class boost::serialization::access;
  template<class Archive>
  void serialize(Archive & ar, const unsigned int version)
  {
    ar & BOOST_SERIALIZATION_NVP(m_pMapper);
    ar & BOOST_SERIALIZATION_NVP(m_pCorrelationGrid);
    ar & BOOST_SERIALIZATION_NVP(m_pSearchSpaceProbs);
    ar & BOOST_SERIALIZATION_NVP(m_pGridLookup);
    ar & BOOST_SERIALIZATION_NVP(m_xPoses);
    ar & BOOST_SERIALIZATION_NVP(m_yPoses);
    ar & BOOST_SERIALIZATION_NVP(m_rSearchCenter);
    ar & BOOST_SERIALIZATION_NVP(m_searchAngleResolution);
    ar & BOOST_SERIALIZATION_NVP(m_nAngles);
    ar & BOOST_SERIALIZATION_NVP(m_searchAngleResolution);
    ar & BOOST_SERIALIZATION_NVP(m_doPenalize);
    kt_int32u poseResponseSize =
      static_cast<kt_int32u>(m_xPoses.size() * m_yPoses.size() * m_nAngles);
    if (Archive::is_loading::value) {
      m_pPoseResponse = new std::pair<kt_double, Pose2>[poseResponseSize];
    }
    ar & boost::serialization::make_array<std::pair<kt_double, Pose2>>(m_pPoseResponse,
      poseResponseSize);
  }
};    // ScanMatcher```

When running without valgrind, whether it crashes or not depends on whether that chunk of delete[]'d memory has been overwritten/reused, which explains why it's quite sensitive to varying random parameters. In any case, is there a need to actually serialize m_pPoseResponse if it's just used as a temporary holding area to accumulate the threaded results?

SteveMacenski, 2021-11-18:
>  In any case, is there a need to actually serialize m_pPoseResponse if it's just used as a temporary holding area to accumulate the threaded results?

I think not. But also removing it from serialization would be bad since it would invalidate all previously serialized files in a new executable. So we should keep it serialized, but add the appropriate protections around this so that it doesn't try to do something that is deleted. It would be nice to add the `nullptr` too. The original library didn't do serialization, that's something we added, so its not surprising that Karto didn't properly set to `nullptr` because there was no reason for that memory to be accessed afterwards.

I'd love a PR to set it to nullptr and then use that check in the serialize function to only do when needed!

jaredohmni, 2021-11-18:
Will do! We'll test the changes across our fleet for a couple days and submit the PR once we've completed validation.  Thanks for the hard work on slam-toolbox, and for encouraging community development!

SteveMacenski, 2021-11-30:
merged and backported to all active distributions

closed 2021-11-30 by SteveMacenski