SteveMacenski/slam_toolbox #394: Mapping in dynamic environment

AndreaEirale, 2021-05-18:
Hi everyone!
I'm trying to use Slam-Toolbox online async configuration in order to map a dynamic environment, but I really cannot figure out how to do it.
My main problem is that when I put an obstacle in the field of view of the lidar sensor, it is marked on the gridmap, but if I remove it, it is still marked within the gridmap and is not removed. I tried with different configurations, I also set the update of the map at every second, regardless of the distance traveled, but the obstacle is still there. 
Am I missing something or

SteveMacenski, 2021-05-18:
You have to collect more data to have the past measurements be probabilistically removed because sufficient number of new measurements say that space is free. Because you only update scans after moving sufficient distance over sufficient time, you need to move the robot and collect new measurements to the slam problem

AndreaEirale, 2021-05-19:
Thank you for your quick reply!
I understand this, but it does not seem to work! I tried to navigate around the position where the obstacle was for a little while, but it is still not removed.
We are trying to implement a person following algorithm, we retrieve the position of the person using a depth camera and a neural network, then we pass the goal of the person to Nav2 (using the follow-point behaviour tree). It seems to work at first, but then sometimes the person followed is marked as an obstacle by Slam-Toolbox, and this affects also the global costmap of Nav2, and the robot is stucked.
We are trying to make Slam-Toolbox remove the obstacle (marked at the person's position) when the person moves from there, but we cannot figure out how to do this.
Do you know any kind of workaround to solve this problem?
Thank you very much,
Andrea Eirale

SteveMacenski, 2021-05-19:
I tested quickly on an existing dataset that I have and it seems to work OK. The proportional of misses to hits has to reach a threshold for those to be "raytraced" out. I can't comment / help with your specific application (not fully on topic for this project). You could try posting on ROS Answers and see if someone who's done a similar thing can help.

SteveMacenski, 2021-08-25:
closing since no comments since May and it appears to be working as expected.

closed 2021-08-25 by SteveMacenski