SteveMacenski/slam_toolbox #397: How would one configure slam_toolbox to only perform mapping (i.e, publish to /map) and no localization?

safdark, 2021-05-30:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - apt-get
- ROS Version
  - Foxy
- Version or commit hash:
  - 2.4.0
- Laser unit:
  - Hokuyo 30lx
  
#### Steps to reproduce issue
N/A

#### Expected behavior
N/A

#### Actual behavior
N/A

#### Additional information

SteveMacenski, 2021-05-30:
Do you use TF? If not, then just this way would work. The only outputs are TF and the map. If you do use TF, the best way would be to add a new parameter, “publish_tf” that you could set to false and have it skip publishing the TF transformations.

safdark, 2021-06-01:
Thanks for the prompt response @SteveMacenski.
Yes, I use TF.
If time permits, I will try to submit a PR for that addition shortly.

Btw, what's the difference between "localization" and "mapping" for the **mode** parameter that's presently exposed? To someone who is new to SLAM, that ('mode') parameter available in slam_toolbox might seem like the answer to this issue, but the actual behavior is different. Perhaps I didn't look enough, but I couldn't find documentation for that parameter on wiki.ros.org (or elsewhere).

SteveMacenski, 2021-06-01:
See the readme section about it: https://github.com/SteveMacenski/slam_toolbox#localization

closed 2021-06-01 by safdark