SteveMacenski/slam_toolbox #398: Irregular "Ceres could not find a usable solution" warnings in localization mode

PGotzmann, 2021-05-31:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source
- ROS Version
  - ROS2 foxy
- Version or commit hash:
  - bd77076fbd147434f6f0937a8f2853edcc2ce8b8
- Laser unit:
  - 360deg Scanner
  
#### Steps to reproduce issue
- run `slam_toolbox_localization` with a prerecorded map
- make sure`scan_buffer_size` is set to a small value (e.g. the default 3 for localization) and `debug_logging` is enabled.
- drive around and wait for graph optimizations to happen (e.g. when a loop-closure occurs)

#### Expected behavior
Ceres solver finds a suitable solution and terminates with a debug log message like this:
``` 
Termination:  CONVERGENCE (Function tolerance reached. |cost_change|/cost: 9.408368e-04 <= 1.000000e-03)
```

#### Actual behavior
From time to time, Ceres solver is unable to produce a valid solution:
```[WARN]	CeresSolver: Ceres could not find a usable solution to optimize.``` 
with the corresponding debug log message:
```
Termination:  FAILURE (ParameterBlock: 0x7fc0eb94a7d0 with size 1 has at least one invalid value.
First invalid value is at index: 0.
Parameter block values:          nan )
```
#### Additional information
This issue was previously reported e.g. here https://github.com/SteveMacenski/slam_toolbox/pull/326#issuecomment-774596333  or here https://github.com/SteveMacenski/slam_toolbox/issues/281#issuecomment-800424766 but could not be linked to a cause until now.

After some testing it looks like this issue does **only happen in localization mode and if some scans have previously been removed from the graph** i.e. by expiring from the localization buffer or reciving a `initialpose` message. The warnings are gone if `localization buffer` is set to a value higher than the max number of nodes added to the graph.

Actually, the memory address `0x7fc0eb94a7d0` of the `ParemeterBlock` from the log above belongs to the parameter block from a node that has previously been removed from the graph and from what I see the object it points to (2D pose of the node) has already been deleted here https://github.com/SteveMacenski/slam_toolbox/blob/bd77076fbd147434f6f0937a8f2853edcc2ce8b8/solvers/ceres_solver.cpp#L344

##### Node removal from solver:
Generally, if a node is removed in `Mapper::RemoveNodeFromGraph()` the corresponding constraints (Ceres Residual block) and node is also removed from the solver. However, according to Ceres documentation **`RemoveResidualBlock` does not remove the dependent parameters.** (https://github.com/ceres-solver/ceres-solver/blob/ec4f2995bbde911d6861fb5c9bb7353ad796e02b/include/ceres/problem.h#L294:297). This leads to an ever increasing number of solver `Parameter blocks` visible in the Ceres solver summary:
```
                    Original                  Reduced
Parameter blocks        966 <-increases          882<-const
Parameters              966                      882
Residual blocks         344 <-const              344
Residuals              1032                     1032
``` 
The solver correctly reduces the parameter blocks to the actually used ones but it looks like the old parameter blocks are still accessed in some way, leading to the failure described in this issue.

#### Possible Solution
A simple solution to that would be to **also remove the parameter blocks when removing the node from the solver** in https://github.com/SteveMacenski/slam_toolbox/blob/bd77076fbd147434f6f0937a8f2853edcc2ce8b8/solvers/ceres_solver.cpp#L338 i.e add the following lines before `nodes_->erase(nodeit)`
```
    problem_->RemoveParameterBlock(&nodeit->second(0));
    problem_->RemoveParameterBlock(&nodeit->second(1));
    problem_->RemoveParameterBlock(&nodeit->second(2));
    nodes_->erase(nodeit);
```
With that fix, I can not reproduce this issue anymore.
 
If that sounds reasonable to you, I can create a PR with this fix.

SteveMacenski, 2021-06-01:
A PR would be good, that seems reasonable

SteveMacenski, 2021-06-01:
Merged

NhatTanXT3, 2023-12-28:
In my case, after the localization buffer is full and starts to remove nodes, the optimizer fails and produces the same message as above: `Ceres could not find a usable solution to optimize` because of `ParameterBlock: 0xsomething with size 1 has at least one invalid value`

my quick solution is to check `problem_->HasParameterBlock` before removing it. patch something like this:
[image]

After the patch, the optimizer and the initial pose feature work normally (only tested on noetic)

SteveMacenski, 2024-01-02:
@NhatTanXT3 that seems like a good solution! Can you submit a PR once you have done a bit of testing?

Noetic vs ROS 2 distros for this won't make a difference. If it works for you, it'll work for those too.

NhatTanXT3, 2024-01-07:
I created several PRs for this issue. I only tested on Noetic, the Ceres solver working well on improving localization repeatability for my use case. For ROS 2, I only copy, paste, and compile it, and haven't tested it.

SteveMacenski, 2024-01-08:
Thanks! Tiny change to all 3 I put in the ROS 2 PR, then I can merge the set

NhatTanXT3, 2024-01-09:
I updated the log for all 3 PRs.

I noticed this issue with a map where the optimizer runs very frequently and for long-term operations (like one day more). I also tried to run the localization mode on the map built with the rosbag of issue #198, but it only happens near the end of this rosbag. So it's not easy to notice.

My map looks like this:
[image]

I am also looking for any chance to improve the precision/repeatability of the localization mode (i.e. same scan data, same output, with no or small jumping). So do you have any issue related to localization mode I could help with? I checked #334 but there is no rosbag available to continue.

SteveMacenski, 2024-01-09:
The error you see there is the error this ticket is open regarding :-) It doesn't appear that your patch actually resolves the issue?

> So do you have any issue related to localization mode I could help with? I checked https://github.com/SteveMacenski/slam_toolbox/issues/334 but there is no rosbag available to continue.

I think that's a good entry point and discussion to start off with. I unfortunately don't have a dataset I can provide for you, the ones I have are on "loan" from companies for my personal development of this package and Nav2 and aren't for sharing :(

NhatTanXT3, 2024-01-09:
> The error you see there is the error this ticket is open regarding :-) It doesn't appear that your patch actually resolves the issue?

Those warning messages from my picture are before applying the patch, I see them every day, and annoying me. Then I enabled the Ceres summary log, which surprised me the optimizer always fails and produces the message. 

After this patch, I no longer see this message.

SteveMacenski, 2024-01-09:
Oh ok, that was unclear from context. So the PRs are good to go and the localization question is another topic?

NhatTanXT3, 2024-01-10:
> Oh ok, that was unclear from context. To be clear, the PRs are good to go and the localization question is another topic?

Yeah, it's good to go.
The localization question is a different topic.

SteveMacenski, 2024-01-10:
Merged! We can continue on localization in one of those other tickets - or if different - a newer one?

closed 2024-01-10 by SteveMacenski

SteveMacenski, 2024-01-10:
Those sound like good contributions, not niche at all. Though, they might need to be parameterized options so users can select if they want that behavior or not. Yay! More parameters :-)