SteveMacenski/slam_toolbox #406: (De)Serialize services could return a boolean

nickvaras, 2021-06-04:
The current srv's for the SerializeGraph and DeSerializeGraph calls don't return a value.
It would be nice that in a future version they returned a Bool with success true/false.
I've had a few cases of the file failing to open and be read (the cause is a different topic of discussion), but with the current implementation, there's no assurance that the call succeeded and without an indication that there was an error (other than the ROS_ERROR that gets logged), handling the exception is difficult.

SteveMacenski, 2021-06-04:
PRs would be appreciated.

nickvaras, 2021-06-04:
:wink:

SteveMacenski, 2021-11-19:
Closing as a feature request without solution (and not that big, if someone actually wanted this, I'd merge a PR, but not important enough to keep tracking if there's no implementer or user interested)

closed 2021-11-19 by SteveMacenski