SteveMacenski/slam_toolbox #408: slam_toolbox mapping with small resolution

Kaiser-Wang, 2021-06-16:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- ROS Version
  - ROS2 Foxy
- Version or commit hash:
  - 2.4.0 Release for Foxy
- Laser unit:
  - hls_lfcd_lds from Gazebo
  
#### Steps to reproduce issue
Simulate Turtlebot3 waffle pi in Turtlebot3 world and use sync_slam_toolbox_node to generate a map with a resolution of 0.01

#### Expected behavior
slam_toolbox generates occupancy grid map for the area where laser beam hits an object, like:
[image]

#### Actual behavior
slam_toolbox only generates a small area of the map scanned by laser.
[image]

#### Additional information
Gazebo laser information:

> 
      <sensor name="hls_lfcd_lds" type="ray">
        <always_on>true</always_on>
        <visualize>true</visualize>
        <pose>-0.032 0 0.171 0 0 0</pose>
        <update_rate>5</update_rate>
        <ray>
          <scan>
            <horizontal>
              <samples>360</samples>
              <resolution>1.000000</resolution>
              <min_angle>0.000000</min_angle>
              <max_angle>6.280000</max_angle>
            </horizontal>
          </scan>
          <range>
            <min>0.120000</min>
            <max>3.5</max>
            <resolution>0.015000</resolution>
          </range>
          <noise>
            <type>gaussian</type>
            <mean>0.0</mean>
            <stddev>0.01</stddev>
          </noise>
        </ray>
        <plugin name="turtlebot3_laserscan" filename="libgazebo_ros_ray_sensor.so">
          <ros>
            
            <argument>~/out:=scan</argument>
          </ros>
          <output_type>sensor_msgs/LaserScan</output_type>
          <frame_name>base_scan</frame_name>
        </plugin>
      </sensor>

slam_toolbox sync_slam_toolbox_node parameter:

> 
    slam_toolbox:
      ros__parameters:

        # Plugin params
        solver_plugin: solver_plugins::CeresSolver
        ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
        ceres_preconditioner: SCHUR_JACOBI
        ceres_trust_strategy: LEVENBERG_MARQUARDT
        ceres_dogleg_type: TRADITIONAL_DOGLEG
        ceres_loss_function: None

        # ROS Parameters
        odom_frame: odom
        map_frame: map
        base_frame: base_footprint
        scan_topic: /scan
        mode: mapping

        debug_logging: false
        throttle_scans: 1
        transform_publish_period: 0.02 #if 0 never publishes odometry
        map_update_interval: 1.0
        resolution: 0.01
        max_laser_range: 20.0 #for rastering images
        minimum_time_interval: 0.5
        transform_timeout: 0.2
        tf_buffer_duration: 30.0
        stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps
        enable_interactive_mode: true

        # General Parameters
        use_scan_matching: true
        use_scan_barycenter: true
        minimum_travel_distance: 0.1
        minimum_travel_heading: 0.1
        scan_buffer_size: 10
        scan_buffer_maximum_scan_distance: 10.0
        link_match_minimum_response_fine: 0.1  
        link_scan_maximum_distance: 1.5
        loop_search_maximum_distance: 3.0
        do_loop_closing: true 
        loop_match_minimum_chain_size: 10           
        loop_match_maximum_variance_coarse: 3.0  
        loop_match_minimum_response_coarse: 0.35    
        loop_match_minimum_response_fine: 0.45

        # Correlation Parameters - Correlation Parameters
        correlation_search_space_dimension: 0.5
        correlation_search_space_resolution: 0.01
        correlation_search_space_smear_deviation: 0.1 

        # Correlation Parameters - Loop Closure Parameters
        loop_search_space_dimension: 8.0
        loop_search_space_resolution: 0.01
        loop_search_space_smear_deviation: 0.03

        # Scan Matcher Parameters
        distance_variance_penalty: 0.5      
        angle_variance_penalty: 1.0    

        fine_search_angle_offset: 0.00349     
        coarse_search_angle_offset: 0.349   
        coarse_angle_resolution: 0.0349        
        minimum_angle_penalty: 0.9
        minimum_distance_penalty: 0.5
        use_response_expansion: true

When sync_slam_toolbox_node is running, I constantly receive the following message from terminal:

> 
    [sync_slam_toolbox_node-1] [INFO] [1623883619.420966109] [slam_toolbox]: Message Filter dropping message: frame 'base_scan' at time 3208.001 for reason 'Unknown'

I tried to increase the number of samples of laser ray in Gazebo, and slam_toolbox gave me a larger scan area of the map.

I also tried to increase the resolution back to 0.05 in slam_toolbox parameter file, and it also generates a larger area of the map.

I noticed that slam_toolbox generates less area of the map for the same scan and resolution in cartographer. I'm not an expert of SLAM but I assume this is because of different algorithms.

SteveMacenski, 2021-06-16:
Why do you need a resolution of 1cm? That is very excessive and likely isn't going to work well even if the issue you're seeing is resolved (memory, update times, localization quality, etc). It could very well have to do with the laser scan density relative to the resolution / range. If you have only 1 beam per degree, at 1 cm resolution you can't really say a whole lot about areas far away because the density is so low of values to consider that area "seen" so it would remain as unknown.

Kaiser-Wang, 2021-06-17:
Thanks for your reply. The reason why I need a such small resolution is because I want to get a more accurate map and use it in Navigation2 to navigate my robot to a specific goal pose more accurately. Yes, I totally agree with that it has to do with laser scan density and it seems it's unlikely to increase the density to a level that can get 1 cm resolution at a larger range due to memory issues. Also, I realized that there are many other factors affecting nav2 accuracy, such as noise in environment and amcl, nav2 goal pose checker, etc. Anyways, I'll try to use aruco marker to get better accuracy. Thanks for your explanation again as it really helped me to understand the issue.

SteveMacenski, 2021-06-17:
There are diminishing returns on resolution of the map for localization accuracy. Just lowering the map resolution to 1cm won't make it 1cm accurate if you don't account for alot of things in your system timing / improved odometry / etc. In general, anything below 2.5cm is probably not going to yield you any better localization quality without improvements in your localization algorithms / system accuracy. 

Closing since I think we've come to the conclusion that this is too small + laser density improvements to make it work better .

closed 2021-06-17 by SteveMacenski