SteveMacenski/slam_toolbox #412: CMake policy

quentingllmt, 2021-06-18:
**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - From source
- ROS Version
  - ROS2, foxy (from source too)
- Version or commit hash:
  - latest foxy-devel
  
#### Steps to reproduce issue
Try to build the repo from source with ros2 foxy on ubuntu 18.04 (bionic)

#### Expected behavior
Working compilation ;)

#### Actual behavior
The CMakeLists.txt is inconsistant as the minium required cmake version is 3.5 but the `cmake_policy(SET CMP0077 NEW)` seems to be only available in cmake > 3.13 ? Not sure.

Then, during the compilation (I have cmake 3.10), an error is thrown : 
`policy cmp0077 is not known to this version of cmake`

Removing `cmake_policy(SET CMP0077 NEW)`  from the CMakeLists.txt allow us to build the project without errors. Then why is is there ? Is this line actually needed ?

#### Additional information
Solution 1:
Remove  `cmake_policy(SET CMP0077 NEW)` 

Solution 2:
Replacing the actual  `cmake_policy(SET CMP0077 NEW)`  by  `set(CMAKE_POLICY_DEFAULT_CMP0077 NEW)` seems to work.

SteveMacenski, 2021-06-18:
I don't see this issue at all on 20.04, are you sure it doesn't have to do with that (and your cmake version)? Did you try installing the Eloquent or Dashing branches, which should support 18.04 correctly? 

> cmake_minimum_required(VERSION 3.5)

It clearly states the minimum is 3.5. If you're using 3.1, then that's not supported. 

You should be using the right branch for your OS / ROS2 versions. This is not an issue with the project but selectively mismatching required dependencies. Closing

closed 2021-06-18 by SteveMacenski

quentingllmt, 2021-06-18:
1- I am using 3.10. It is not 3.1 but 3.10.0.
2- My ROS2 distro is foxy. So I am using the foxy branch of your repo.
3- Problem: ament_cmake version update is limited on ubuntu 18.04

Anyway, I did fork your repo and fix it.