SteveMacenski/slam_toolbox #415: Initial localization fails unless robot is exactly

xperroni, 2021-06-23:
**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- ROS Version
  - ROS1 Melodic
- Version or commit hash:
  - 1.1.6-1 bionic.20210505.045310
- Laser unit:
  - Simulated
  
#### Steps to reproduce issue

1. Start Gazebo simulation, instantiate simulated environment and robot;
2. Run `async_slam_toolbox_node`, map environment, save map, terminate simulation;
3. Restart simulation with robot shifted 1 m from initial pose across X and Y axes;
4. Run `localizaton_slam_toolbox_node` in localization mode.

#### Expected behavior

SLAM should detect robot is away from the origin and shift the `odom` frame relative to `map` in order to account for that.

#### Actual behavior

After a few seconds the `map <-> odom` transform is updated, but it does not correctly account for the robot's pose.

Initial transform:

[image]

Updated transform:

[image]

#### Additional information

Localization settings:

    # Plugin params
    solver_plugin: solver_plugins::CeresSolver
    ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
    ceres_preconditioner: SCHUR_JACOBI
    ceres_trust_strategy: LEVENBERG_MARQUARDT
    ceres_dogleg_type: TRADITIONAL_DOGLEG
    ceres_loss_function: None

    # ROS Parameters
    mode: localization
    debug_logging: false
    throttle_scans: 1
    transform_publish_period: 0.02 #if 0 never publishes odometry
    map_update_interval: 5.0
    resolution: 0.05
    max_laser_range: 20.0 #for rastering images
    minimum_time_interval: 0.5
    transform_timeout: 0.2
    tf_buffer_duration: 30.
    stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps

    # General Parameters
    use_scan_matching: true
    use_scan_barycenter: true
    minimum_travel_distance: 0.5
    minimum_travel_heading: 0.5
    scan_buffer_size: 3
    scan_buffer_maximum_scan_distance: 10
    link_match_minimum_response_fine: 0.1
    link_scan_maximum_distance: 1.5
    do_loop_closing: true
    loop_match_minimum_chain_size: 3
    loop_match_maximum_variance_coarse: 3.0
    loop_match_minimum_response_coarse: 0.35
    loop_match_minimum_response_fine: 0.45

    # Correlation Parameters - Correlation Parameters
    correlation_search_space_dimension: 0.5
    correlation_search_space_resolution: 0.01
    correlation_search_space_smear_deviation: 0.1

    # Correlation Parameters - Loop Closure Parameters
    loop_search_space_dimension: 8.0
    loop_search_space_resolution: 0.05
    loop_search_space_smear_deviation: 0.03
    loop_search_maximum_distance: 5.0

    # Scan Matcher Parameters
    distance_variance_penalty: 0.5
    angle_variance_penalty: 1.0

    fine_search_angle_offset: 0.00349
    coarse_search_angle_offset: 0.349
    coarse_angle_resolution: 0.0349
    minimum_angle_penalty: 0.9
    minimum_distance_penalty: 0.5
    use_response_expansion: true

SteveMacenski, 2021-06-23:
Did you set the initial pose? There's a specific parameter set for this, as well as it responds to the 2D pose estimate tool. Please read the readme.

xperroni, 2021-06-23:
> Did you set the initial pose? There's a specific parameter set for this, as well as it responds to the 2D pose estimate tool. Please read the readme.

Hi Steve,

Thank you for your reply.

Yes, I am aware the initial pose can be set using either a parameter or by sending a message to the `initialpose` topic through the RViz _2D Pose Estimate_ tool.

However I was under the impression that the localization node would still be able to automatically correct a small discrepancy between the robot's initial pose and the map's origin, even without being given its value.

Is that not the case? Or is it simply that a shift of 1 m is too much? If the later, is there any way to broaden the initial pose search?

SteveMacenski, 2021-06-23:
The initial pose is taken as word of god, just as it is in AMCL. You have to move around a bit before it will  correct, but the localization mode of slam toolbox assumes you have good industrial grade odometry and relatively know starting positions. If that's not the case, then the localization mode of this package is not right for you and you should use AMCL.

paucarre, 2021-07-06:
We're having the same problem. I was going to start work on #414 to address mainly this issue. I'd like to know if @SteveMacenski could provide his point of view on whether the approach makes sense.

The idea having a mode without initial pose where the initial pose is guessed by scan matching points with the same label. Labels would reduce the space of potential points that can be scan matched and thus hopefully have higher chance to find automatically a good initial pose estimate.

closed 2021-09-14 by SteveMacenski