SteveMacenski/slam_toolbox #417: Offline - No map published

THLL245, 2021-06-28:
**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- ROS Version
  - 
- Version or commit hash:
  - 
- Laser unit:
  - 
  
#### Steps to reproduce issue
roslaunch slam_toolbox offline.launch
Topics: /scan, odom
rosrun rosbag file publishing those topics
gmapping works fine with these topics and successfully maps

#### Expected behavior
Publish /map

#### Actual behavior
No /map being published
No errors
Logs when debug = true

[DEBUG] [1624902528.723416491]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1624581533.376, count now 1
[DEBUG] [1624902528.724588866]: MessageFilter [target=odom ]: Message ready in frame velodyne at time 1624581533.376, count now 0
 
Repeats over and over again (green text)

#### Additional information

SteveMacenski, 2021-06-28:
Please ask on ROS Answers and provide **alot** more information, that is woefully insufficient description to even give you general advice about where to potentially look

closed 2021-06-28 by SteveMacenski

SteveMacenski, 2021-06-28:
These are the kinds of questions that should be asked on ROS answers so that there's a searchable solution found for the next person with a similar issue. Actually, if you haven't already, you should search to see if you can find a solution to your own problem before opening a new ROS Answers question. 

There's a near zero this is an actual bug in the software, so this is not the appropriate place to ask questions. This is for feature development and bug declarations, not configuration / first time set up help.

aidinism, 2021-09-08:
I ran into a simillar issue, I can get (& visualize) raw lidar data in /scan / point_cloud topics but no /map topic always says "**no map received**".

here is more info about the ROS and slam_toolbox:
running on **ROS1 noetic,** **ubuntu 20.04** and there are only 2 packages (slam_toolbox & Velodyne)  in the src directory. (also I'm using the same Lidar **VLP16**). please let me know if more information is needed.

`SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.11
 * /slam_toolbox/angle_variance_penalty: 1.0
 * /slam_toolbox/base_frame: base_footprint
 * /slam_toolbox/ceres_dogleg_type: TRADITIONAL_DOGLEG
 * /slam_toolbox/ceres_linear_solver: SPARSE_NORMAL_CHO...
 * /slam_toolbox/ceres_loss_function: None
 * /slam_toolbox/ceres_preconditioner: SCHUR_JACOBI
 * /slam_toolbox/ceres_trust_strategy: LEVENBERG_MARQUARDT
 * /slam_toolbox/coarse_angle_resolution: 0.0349
 * /slam_toolbox/coarse_search_angle_offset: 0.349
 * /slam_toolbox/correlation_search_space_dimension: 0.5
 * /slam_toolbox/correlation_search_space_resolution: 0.01
 * /slam_toolbox/correlation_search_space_smear_deviation: 0.1
 * /slam_toolbox/debug_logging: True
 * /slam_toolbox/distance_variance_penalty: 0.5
 * /slam_toolbox/do_loop_closing: True
 * /slam_toolbox/enable_interactive_mode: False
 * /slam_toolbox/fine_search_angle_offset: 0.00349
 * /slam_toolbox/link_match_minimum_response_fine: 0.1
 * /slam_toolbox/link_scan_maximum_distance: 1.5
 * /slam_toolbox/loop_match_maximum_variance_coarse: 3.0
 * /slam_toolbox/loop_match_minimum_chain_size: 10
 * /slam_toolbox/loop_match_minimum_response_coarse: 0.35
 * /slam_toolbox/loop_match_minimum_response_fine: 0.45
 * /slam_toolbox/loop_search_maximum_distance: 3.0
 * /slam_toolbox/loop_search_space_dimension: 8.0
 * /slam_toolbox/loop_search_space_resolution: 0.05
 * /slam_toolbox/loop_search_space_smear_deviation: 0.03
 * /slam_toolbox/map_frame: map
 * /slam_toolbox/map_update_interval: 5.0
 * /slam_toolbox/max_laser_range: 20.0
 * /slam_toolbox/minimum_angle_penalty: 0.9
 * /slam_toolbox/minimum_distance_penalty: 0.5
 * /slam_toolbox/minimum_time_interval: 0.5
 * /slam_toolbox/minimum_travel_distance: 0.5
 * /slam_toolbox/minimum_travel_heading: 0.5
 * /slam_toolbox/mode: mapping
 * /slam_toolbox/odom_frame: odom
 * /slam_toolbox/resolution: 0.05
 * /slam_toolbox/scan_buffer_maximum_scan_distance: 10
 * /slam_toolbox/scan_buffer_size: 10
 * /slam_toolbox/scan_topic: /scan
 * /slam_toolbox/solver_plugin: solver_plugins::C...
 * /slam_toolbox/stack_size_to_use: 40000000
 * /slam_toolbox/tf_buffer_duration: 30.0
 * /slam_toolbox/throttle_scans: 1
 * /slam_toolbox/transform_publish_period: 0.02
 * /slam_toolbox/transform_timeout: 0.2
 * /slam_toolbox/use_response_expansion: True
 * /slam_toolbox/use_scan_barycenter: True
 * /slam_toolbox/use_scan_matching: True

NODES
  /
    slam_toolbox (slam_toolbox/sync_slam_toolbox_node)

ROS_MASTER_URI=http://localhost:11311

process[slam_toolbox-1]: started with pid [19124]
[ INFO] [1631145155.518885933]: Node using stack size 40000000
[DEBUG] [1631145155.576978749]: Attempting to create managed instance for class solver_plugins::CeresSolver.
[DEBUG] [1631145155.577011787]: Class solver_plugins::CeresSolver maps to library libceres_solver_plugin in classes_available_.
[DEBUG] [1631145155.577097150]: Iterating through all possible paths where libceres_solver_plugin could be located...
[DEBUG] [1631145155.577120616]: Checking path /home/aidin/ws_slam_vlp16/devel/lib/libceres_solver_plugin.so 
[DEBUG] [1631145155.577140120]: Library libceres_solver_plugin found at explicit path /home/aidin/ws_slam_vlp16/devel/lib/libceres_solver_plugin.so.
[DEBUG] [1631145155.580028012]: solver_plugins::CeresSolver maps to real class type solver_plugins::CeresSolver
[ INFO] [1631145155.582125601]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[DEBUG] [1631145155.582172012]: boost::shared_ptr to object of real type solver_plugins::CeresSolver created.
[ INFO] [1631145155.582183411]: Using plugin solver_plugins::CeresSolver
[DEBUG] [1631145155.780345233]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631145155.775, count now 1
[DEBUG] [1631145155.830386141]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631145155.826, count now 2
[DEBUG] [1631145155.881281315]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631145155.876, count now 3
[DEBUG] [1631145155.930937970]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631145155.927, count now 4
^C[DEBUG] [1631145155.981973565]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631145155.977, count now 5
`

... same msg over and over

xwavex, 2022-03-23:
I have the exact same problem. Any updates on this? Odometry is published for odom -> base_link, and laser data is available at /scan. What could be the issue?

NamTruongTran, 2023-10-16:
I have the same issue. Any updates?