SteveMacenski/slam_toolbox #419: Set initpose multiple times causes program segmentation

randoms, 2021-07-08:
**Required Info:**

- Operating System:
 Ubuntu 20.04
- Installation type:
 source
- ROS Version
  ROS1 noetic
- Version or commit hash:
  noetic-devel
- Laser unit:
  sc_mini
  
#### Steps to reproduce issue
launch localization.launch and load a saved map, pub init pose to /initpose topic for multiple times. It will crash when publish init pose at second time.

#### Expected behavior
use new init pose
#### Actual behavior
crash
#### Additional information

When /initpose set at the second time, previous node was removed. So the edge of new node cannot be added([ref](https://github.com/SteveMacenski/slam_toolbox/blob/noetic-devel/slam_toolbox/lib/karto_sdk/src/Mapper.cpp#L1484)).
when remove the new node, it tries to remove the none exist edge which cause program crash.

Crash message
```
F0708 03:21:45.766316 36237 problem_impl.cc:597] Parameter block not found: 0x55f84d9fec70. You must add the parameter block to the problem before it can be removed.
```

SteveMacenski, 2021-07-08:
Would you try with removing these lines (https://github.com/SteveMacenski/slam_toolbox/pull/402/files https://github.com/SteveMacenski/slam_toolbox/pull/405/files)?  that was the most recent change in this area and this was working before. How are you testing, just setting pose, and setting pose again -- or moving in between. If you do the opposite, does the result change?

randoms, 2021-07-09:
Remove these lines fix the issue. 

I just setting pose, and setting pose again. No moving in between. If I moved between setting pose, it also not crash. Because new edge was added between current and previous node.
My suggestion is to use more previous scan if previous scan is none. 

Original
https://github.com/SteveMacenski/slam_toolbox/blob/a5e1ad33b07bf9acddcdfc6953f348cb6b5b15ba/lib/karto_sdk/src/Mapper.cpp#L1442-L1446

Edited
```c++
while(previousScanNum >= 0){
        LocalizedRangeScan* pPrevScan = pSensorManager->GetScan(rSensorName, previousScanNum);
        if (!pPrevScan)
        {
          std::cout << "pPrevScan is NULL, use more previsous scan: " << previousScanNum - 1 << std::endl;
          previousScanNum -= 1;
          continue;
        }
        LinkScans(pPrevScan, pScan, pScan->GetSensorPose(), rCovariance);
        break;
}
```

SteveMacenski, 2021-07-09:
Better option: when we clear a frame from localization updates, we can just set the last scan to the last non-localization scan https://github.com/SteveMacenski/slam_toolbox/blob/a5e1ad33b07bf9acddcdfc6953f348cb6b5b15ba/lib/karto_sdk/src/Mapper.cpp#L128

SteveMacenski, 2021-11-19:
We reverted the changes that caused this issue -- closing

closed 2021-11-19 by SteveMacenski