SteveMacenski/slam_toolbox #420: [ Feature Request ] Service Server Response

Tarzoozoo, 2021-07-09:
#### Feature description
As far as I concern. **slam_toolbox** have a lot services for call, But almost services have not response. It's significant to work with Web-Application via service, because it inform user for response of services. Example services are _/slam_toolbox/save_map_ and _/slam_toolbox/serialize_map_ services.

#### Implementation considerations
The simple approach would be add response structure in .srv file like bool _status_ and string _message_ for information. For instance, I'm modified /slam_toolbox/save_map service below. 
[image]
Or not Cloud you explain why you have not add response structure in your services?

SteveMacenski, 2021-07-09:
I just didnt find it necessary at the time, logging was sufficient. I’d be happy to field a PR to add those, but instead of string messages, it should only be a status code that maps to a specific result (success, fail, fail a specific way) that can be set in the srv files as constants (see visualization_msgs/Marker for example)

Tarzoozoo, 2021-07-12:
Thank you for replying. Now I'm currently improve this issue. So if I'm working on Ubuntu 20.04 (ROS Foxy), Which branch should I merge and use for SLAM (ros2 or foxy-devel)?

SteveMacenski, 2021-07-12:
`ros2` would be the most appropriate since Foxy is also released. This would only be available for future distributions

SteveMacenski, 2021-07-16:
merging

closed 2021-07-16 by SteveMacenski