SteveMacenski/slam_toolbox #422: SLAM problem when using LIDAR with a high scan rate

Tarzoozoo, 2021-07-14:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - from source
- ROS Version
  - ROS2 Foxy
- Version or commit hash:
  - Branch `ros2` 
  - Lasted commit hash `a5e1ad33b07bf9acddcdfc6953f348cb6b5b15ba`
- Laser unit:
  - Hokuyo UTM-30LX
  
#### Steps to reproduce issue
I've recorded bag files (_/tf, /tf_static_ one file and another is _/scan_). The default scan_rate of LIDAR is 40 Hz. Then I launch `offline_launch.py` and I play those bag files simultaneously.
The result of map is not properly and lag. So I change scan_rate of LIDAR to 10 Hz. The result map and posegraph node are properly and completely.

#### Expected behavior
Map and Posegraph node are properly and completely.

#### Actual behavior
Map and Posegraph node are is not properly and lag.
Error: `[sync_slam_toolbox_node-2] [INFO] [1626257152.370122716] [slam_toolbox]: Message Filter dropping message: frame 'hokuyo_scan_link' at time 1625041098.730 for reason 'Unknown'`
[image]

#### Additional information

I'm investigating this problem. Summary is When use scan_rate 40 Hz, Timestamped of _map_frame_ is produced by slam_toolbox **not synchronous** with  _/scan_. So I decrease scan_rate to 10 Hz Timestamped of _map_frame_ is synchronous with _/scan_. The result of Map and Posegraph node are properly and completely.

I founded the cause of this problem is `SlamToolbox::laserCallback` is not called when a laser message is coming. So I've increse _queue_size_   `scan_filter_` in `slam_toolbox_common.cpp` from 1 to 50. The result Map and Posegraph node are properly and completely. Localization look fine. Timestamped of _map_frame_  is **synchronous** with _/scan_.Even if I use scan_rate of LIDAR is 40 Hz.

**Bag file**
10Hz: [Bag_10HZ.zip](https://github.com/SteveMacenski/slam_toolbox/files/6815394/Bag_10HZ.zip)
40Hz: [Bag_40HZ.zip](https://github.com/SteveMacenski/slam_toolbox/files/6815395/Bag_40HZ.zip)

**TF_Frame**:
[frames.pdf](https://github.com/SteveMacenski/slam_toolbox/files/6814886/frames.pdf)

**Parameter**:
```
slam_toolbox:
  ros__parameters:

    # Plugin params
    solver_plugin: solver_plugins::CeresSolver
    ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
    ceres_preconditioner: SCHUR_JACOBI
    ceres_trust_strategy: LEVENBERG_MARQUARDT
    ceres_dogleg_type: TRADITIONAL_DOGLEG
    ceres_loss_function: None

    # ROS Parameters
    odom_frame: t265_odom # odom
    map_frame: map
    base_frame: base_footprint
    scan_topic: /scan   # /scan
    mode: mapping #localization
    debug_logging: false
    throttle_scans: 1 # 1 
    transform_publish_period: 0.02 #if 0 never publishes odometry
    map_update_interval: 10.0 # 10.0 **save .pgm
    resolution: 0.05
    max_laser_range: 30.0 #for rastering images
    minimum_time_interval: 0.5
    transform_timeout: 0.2 # 0.2
    tf_buffer_duration: 14400.
    stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps
    enable_interactive_mode: true

    # General Parameters
    use_scan_matching: true
    use_scan_barycenter: true
    minimum_travel_distance: 0.4  # 0.5 node as robot's pose
    minimum_travel_heading: 0.4   # 0.5 node as robot's pose
    scan_buffer_size: 10  # 10
    scan_buffer_maximum_scan_distance: 10.0 # 10.0
    link_match_minimum_response_fine: 0.1 # 0.1   
    link_scan_maximum_distance: 1.5 # 1.5
    loop_search_maximum_distance: 3.0
    do_loop_closing: true 
    loop_match_minimum_chain_size: 10           
    loop_match_maximum_variance_coarse: 3.0 # 3.0 
    loop_match_minimum_response_coarse: 0.35    
    loop_match_minimum_response_fine: 0.45

    # Correlation Parameters - Correlation Parameters
    correlation_search_space_dimension: 0.5 # 0.5 (.0)**Higher -> CPU
    correlation_search_space_resolution: 0.01
    correlation_search_space_smear_deviation: 0.1 # 0.1 (0.005-0.1) 

    # Correlation Parameters - Loop Closure Parameters
    loop_search_space_dimension: 8.0 # 8.0 (No any change)
    loop_search_space_resolution: 0.05 # 0.05 (0.05-1)
    loop_search_space_smear_deviation: 0.03 # 0.03

    # Scan Matcher Parameters
    distance_variance_penalty: 0.3  # 0.5  
    angle_variance_penalty: 0.3     # 1.0    

    fine_search_angle_offset: 0.00349     
    coarse_search_angle_offset: 0.349   
    coarse_angle_resolution: 0.0349 # 0.0349        
    minimum_angle_penalty: 0.9    # 0.9
    minimum_distance_penalty: 0.5 # 0.5
    use_response_expansion: true
```

SteveMacenski, 2021-07-14:
It sounds like your static TF (and/or normal TF) isn't publishing fast enough to keep up with the sensor data to transform into the odometry frame. You might want to look into that. If its not publishing at _at least_ 100hz for static and 50hz for normal TF, you have a problem there. 

Given all of my datasets are with the 40hz Hokuyo, I can pretty confidently say its not an issue with this software. Quite literally all of my tests on every distribution is with your proposed setup :-)

Tarzoozoo, 2021-07-15:
Thank you for your suggestion. So I've tested on simulation with turtlebot3. I modified lidar's scan rate 40Hz and the normal TF (odom_frame) update rate is 50 Hz as you suggest. I've test 3 times. The result is 3 times was fine, map and posegraph data were good, but the last test file (TB3_3.zip) I get the error like `[sync_slam_toolbox_node-2] [INFO] [1626349439.910556508] [slam_toolbox]: Message Filter dropping message: frame 'base_scan' at time 222.726 for reason 'Unknown'`. Could you explain why I still get the error while the map is good?  or increase the normal TF update rate?
 I've already attached the bag file below.

**TF frame:**
[frames.pdf](https://github.com/SteveMacenski/slam_toolbox/files/6822862/frames.pdf)

 **Bag file test**
[TB3_1.zip](https://github.com/SteveMacenski/slam_toolbox/files/6822874/TB3_1.zip)
[TB3_2.zip](https://github.com/SteveMacenski/slam_toolbox/files/6822877/TB3_2.zip)
[TB3_3.zip](https://github.com/SteveMacenski/slam_toolbox/files/6822880/TB3_3.zip)

SteveMacenski, 2021-07-15:
Hi,

I don't have cycles to run your bag files and debug your specific situation. I suggest you do some research and review some questions on ROS Answers. If you still can't find a solution, asking on ROS Answers itself. Since we've definitely cleared up that its TF related from your easy mostly fix, I think we should close this ticket since there's no a bug or feature requested for the software itself.

closed 2021-07-15 by SteveMacenski