SteveMacenski/slam_toolbox #423: rolling and galactic build failed

HosseinSheikhi, 2021-07-15:
**Required Info:**

- Operating System:
  -  Ubuntu 20.04 
- Installation type:
  - from source  
- ROS Version
  - ROS2, rolling
- Version or commit hash:
  - a5e1ad33b07bf9acddcdfc6953f348cb6b5b15ba

#### Steps to reproduce issue
colcon build --symlink-install
#### Expected behavior

#### Actual behavior
```
In file included from /home/hossein/slam_ws/src/slam_toolbox/include/slam_toolbox/slam_toolbox_common.hpp:40,
                 from /home/hossein/slam_ws/src/slam_toolbox/include/slam_toolbox/slam_toolbox_sync.hpp:25,
                 from /home/hossein/slam_ws/src/slam_toolbox/src/slam_toolbox_sync_node.cpp:21:
/home/hossein/ros2_rolling/install/tf2_sensor_msgs/include/tf2_sensor_msgs/tf2_sensor_msgs.h:32:2: warning: #warning This header is obsolete, please include tf2_sensor_msgs/tf2_sensor_msgs.hpp instead [-Wcpp]
   32 | #warning This header is obsolete, please include tf2_sensor_msgs/tf2_sensor_msgs.hpp instead
      |  ^~~~~~~
/usr/bin/ld: libtoolbox_common.so: undefined reference to `void tf2::fromMsg<geometry_msgs::msg::Quaternion_<std::allocator<void> >, tf2::Quaternion>(geometry_msgs::msg::Quaternion_<std::allocator<void> > const&, tf2::Quaternion&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/merge_maps_kinematic.dir/build.make:289: merge_maps_kinematic] Error 1
make[1]: *** [CMakeFiles/Makefile2:171: CMakeFiles/merge_maps_kinematic.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/hossein/slam_ws/src/slam_toolbox/src/slam_toolbox_async_node.cpp: In function ‘int main(int, char**)’:
/home/hossein/slam_ws/src/slam_toolbox/src/slam_toolbox_async_node.cpp:30:53: warning: ‘const rclcpp::ParameterValue& rclcpp::Node::declare_parameter(const string&)’ is deprecated: declare_parameter() with only a name is deprecated and will be deleted in the future.\nIf you want to declare a parameter that won't change type without a default value use:\n`node->declare_parameter<ParameterT>(name)`, where e.g. ParameterT=int64_t.\n\nIf you want to declare a parameter that can dynamically change type use:\n```\nrcl_interfaces::msg::ParameterDescriptor descriptor;\ndescriptor.dynamic_typing = true;\nnode->declare_parameter(name, rclcpp::ParameterValue{}, descriptor);\n``` [-Wdeprecated-declarations]
   30 |     temp_node->declare_parameter("stack_size_to_use");
      |                                                     ^
In file included from /home/hossein/ros2_rolling/install/rclcpp/include/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /home/hossein/ros2_rolling/install/rclcpp/include/rclcpp/executors.hpp:22,
                 from /home/hossein/ros2_rolling/install/rclcpp/include/rclcpp/rclcpp.hpp:156,
                 from /home/hossein/slam_ws/src/slam_toolbox/include/slam_toolbox/slam_toolbox_common.hpp:32,
                 from /home/hossein/slam_ws/src/slam_toolbox/include/slam_toolbox/slam_toolbox_async.hpp:24,
                 from /home/hossein/slam_ws/src/slam_toolbox/src/slam_toolbox_async_node.cpp:21:
/home/hossein/ros2_rolling/install/rclcpp/include/rclcpp/node.hpp:406:3: note: declared here
  406 |   declare_parameter(const std::string & name);
      |   ^~~~~~~~~~~~~~~~~
/home/hossein/slam_ws/src/slam_toolbox/src/slam_toolbox_sync_node.cpp: In function ‘int main(int, char**)’:
/home/hossein/slam_ws/src/slam_toolbox/src/slam_toolbox_sync_node.cpp:30:53: warning: ‘const rclcpp::ParameterValue& rclcpp::Node::declare_parameter(const string&)’ is deprecated: declare_parameter() with only a name is deprecated and will be deleted in the future.\nIf you want to declare a parameter that won't change type without a default value use:\n`node->declare_parameter<ParameterT>(name)`, where e.g. ParameterT=int64_t.\n\nIf you want to declare a parameter that can dynamically change type use:\n```\nrcl_interfaces::msg::ParameterDescriptor descriptor;\ndescriptor.dynamic_typing = true;\nnode->declare_parameter(name, rclcpp::ParameterValue{}, descriptor);\n``` [-Wdeprecated-declarations]
   30 |     temp_node->declare_parameter("stack_size_to_use");
      |                                                     ^
In file included from /home/hossein/ros2_rolling/install/rclcpp/include/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /home/hossein/ros2_rolling/install/rclcpp/include/rclcpp/executors.hpp:22,
                 from /home/hossein/ros2_rolling/install/rclcpp/include/rclcpp/rclcpp.hpp:156,
                 from /home/hossein/slam_ws/src/slam_toolbox/include/slam_toolbox/slam_toolbox_common.hpp:32,
                 from /home/hossein/slam_ws/src/slam_toolbox/include/slam_toolbox/slam_toolbox_sync.hpp:25,
                 from /home/hossein/slam_ws/src/slam_toolbox/src/slam_toolbox_sync_node.cpp:21:
/home/hossein/ros2_rolling/install/rclcpp/include/rclcpp/node.hpp:406:3: note: declared here
  406 |   declare_parameter(const std::string & name);
      |   ^~~~~~~~~~~~~~~~~
/home/hossein/slam_ws/src/slam_toolbox/src/experimental/slam_toolbox_lifelong_node.cpp: In function ‘int main(int, char**)’:
/home/hossein/slam_ws/src/slam_toolbox/src/experimental/slam_toolbox_lifelong_node.cpp:30:53: warning: ‘const rclcpp::ParameterValue& rclcpp::Node::declare_parameter(const string&)’ is deprecated: declare_parameter() with only a name is deprecated and will be deleted in the future.\nIf you want to declare a parameter that won't change type without a default value use:\n`node->declare_parameter<ParameterT>(name)`, where e.g. ParameterT=int64_t.\n\nIf you want to declare a parameter that can dynamically change type use:\n```\nrcl_interfaces::msg::ParameterDescriptor descriptor;\ndescriptor.dynamic_typing = true;\nnode->declare_parameter(name, rclcpp::ParameterValue{}, descriptor);\n``` [-Wdeprecated-declarations]
   30 |     temp_node->declare_parameter("stack_size_to_use");
      |                                                     ^
In file included from /home/hossein/ros2_rolling/install/rclcpp/include/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /home/hossein/ros2_rolling/install/rclcpp/include/rclcpp/executors.hpp:22,
                 from /home/hossein/ros2_rolling/install/rclcpp/include/rclcpp/rclcpp.hpp:156,
                 from /home/hossein/slam_ws/src/slam_toolbox/include/slam_toolbox/slam_toolbox_common.hpp:32,
                 from /home/hossein/slam_ws/src/slam_toolbox/include/slam_toolbox/experimental/slam_toolbox_lifelong.hpp:23,
                 from /home/hossein/slam_ws/src/slam_toolbox/src/experimental/slam_toolbox_lifelong_node.cpp:21:
/home/hossein/ros2_rolling/install/rclcpp/include/rclcpp/node.hpp:406:3: note: declared here
  406 |   declare_parameter(const std::string & name);
      |   ^~~~~~~~~~~~~~~~~
/usr/bin/ld: libtoolbox_common.so: undefined reference to `void tf2::fromMsg<geometry_msgs::msg::Quaternion_<std::allocator<void> >, tf2::Quaternion>(geometry_msgs::msg::Quaternion_<std::allocator<void> > const&, tf2::Quaternion&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/async_slam_toolbox_node.dir/build.make:290: async_slam_toolbox_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:765: CMakeFiles/async_slam_toolbox_node.dir/all] Error 2
/usr/bin/ld: libtoolbox_common.so: undefined reference to `void tf2::fromMsg<geometry_msgs::msg::Quaternion_<std::allocator<void> >, tf2::Quaternion>(geometry_msgs::msg::Quaternion_<std::allocator<void> > const&, tf2::Quaternion&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/sync_slam_toolbox_node.dir/build.make:290: sync_slam_toolbox_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:424: CMakeFiles/sync_slam_toolbox_node.dir/all] Error 2
/usr/bin/ld: libtoolbox_common.so: undefined reference to `void tf2::fromMsg<geometry_msgs::msg::Quaternion_<std::allocator<void> >, tf2::Quaternion>(geometry_msgs::msg::Quaternion_<std::allocator<void> > const&, tf2::Quaternion&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/lifelong_slam_toolbox_node.dir/build.make:290: lifelong_slam_toolbox_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:979: CMakeFiles/lifelong_slam_toolbox_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< slam_toolbox [1min 12s, exited with code 2]
                                      
Summary: 0 packages finished [1min 12s]
  1 package failed: slam_toolbox
  1 package had stderr output: slam_toolbox

```
#### Additional information
Same for Galactic. 
Am not quite sure is it a [ros2 problem](https://github.com/ros2/ros2/issues/1154#issue-927998366) or slam_toolbox issue.

SteveMacenski, 2021-07-15:
Are you using the right branches, sourcing your right distributions, and not moving between them to mess up caches?

The build warnings are real, and I'd welcome a PR to fix them, but the linking errors you have afterwards I could not reproduce (and are being built in the build farm OK, so no reason to suspect there's an issue). They do look vaguely familiar to me from issues we were seeing around the rolling-galactic branch off, so that's what makes me suspicious that you're mixing and matching.

HosseinSheikhi, 2021-07-16:
>Are you using the right branches, sourcing your right distributions, and not moving between them to mess up caches?

Should I clone the slam_toolbox/ros2 branch for ros2/rolling?
I have just tried to build on another PC after clearing all the previous dist and caches, the same issue again.  

>The build warnings are real, and I'd welcome a PR to fix them,

Here it is in #424.

SteveMacenski, 2021-07-16:
Lets focus on 1 distribution at a time. Do the following for me:
- Create a brand new workspace using a brand new terminal
- Clone this project (and maybe nav2 map server, or you can scrub some CMake to simplify your life) 
- Source opt/ros/galactic, ensure that you have NOT added that sourcing to your bashrc. If so, remove it, open a new terminal, and repeat from step 1
- Build

Do you still see that linking error? Those are my steps and the steps that are taken in the build farm and we have no linking issues https://build.ros2.org/job/Gdev__slam_toolbox__ubuntu_focal_amd64/ (albeit, the declaration warnings)

HosseinSheikhi, 2021-07-17:
That's fine, builds with no error. But the issue still remains for rolling.

HosseinSheikhi, 2021-07-17:
The problem was due to including header files in ros2:
```
diff --git a/tf2/include/tf2/impl/utils.h b/tf2/include/tf2/impl/utils.h
index a38eb45..bfeabf9 100644
--- a/tf2/include/tf2/impl/utils.h
+++ b/tf2/include/tf2/impl/utils.h
@@ -15,12 +15,9 @@
 #ifndef TF2__IMPL__UTILS_H_
 #define TF2__IMPL__UTILS_H_
 
-#include <tf2/convert.h>
+#include <tf2_geometry_msgs/tf2_geometry_msgs.h>
 #include <tf2/transform_datatypes.h>
 #include <tf2/LinearMath/Quaternion.h>
-#include <geometry_msgs/msg/quaternion.hpp>
-#include <geometry_msgs/msg/quaternion_stamped.hpp>
```

`#include <tf2_geometry_msgs/tf2_geometry_msgs.h>` is removed from `utils.h` in rolling branch. By including this  in utils.h one can build the slam_toolbox with no problem. However, am not quite sure it must be addressed in ros2, or here in slam_toolbox?

SteveMacenski, 2021-07-19:
What would be your suggested resolution in ROS2? It sounds like we just need to include that header in slam_toolbox and we're done. I'd just want to make sure it doesn't break galactic support -- if it does I'll need to branch off before we merge in that fix.

HosseinSheikhi, 2021-07-19:
I could not figure out why that is removed from ROS2.
On the other hand, I tried to solve the problem by editing the `slam_toolbox_common.hpp` but it did not work.
That's why I think it should be addressed in ROS2, however, am not sure.
Yes, those can be merged to `galactic` as well.

SteveMacenski, 2021-07-19:
I don't understand, does your solution work or not?

HosseinSheikhi, 2021-07-19:
Yes, it works, but by changing `utils.h` in ROS2. Not by changing header files in slam_toolbox.

flynneva, 2021-07-19:
jumping in here....have you tried switching to the new `.hpp` standard for ROS2 Rolling? I think this might be your issue...they may have migrated over to `.hpp` and dropped the `.h`.

```
#include "tf2_geometry_msgs/tf2_geometry_msgs.hpp"
```

not 100% sure though but worth a shot.

HosseinSheikhi, 2021-07-19:
Thanks, yes those are already in `.hpp` format.

maciej-3, 2021-09-04:
If this helps someone with ros2-foxy: I had compilation errors related to `#include "tf2_geometry_msgs/tf2_geometry_msgs.hpp"` when I wanted to install this repo from `ros2` or `foxy-devel` branches.

What helped me was switching to `foxy-devel-unfixed`  branch.

SteveMacenski, 2021-09-14:
Handling this myself shortly

SteveMacenski, 2021-09-14:
https://github.com/SteveMacenski/slam_toolbox/pull/436

closed 2021-09-14 by SteveMacenski