SteveMacenski/slam_toolbox #426: LaserRangeScan contains different readings

vinci-xiao, 2021-07-19:
- Operating System:
  - Ubuntu 20.04
- Installation type:
  - from source 
- ROS Version
  - ROS2 foxy
- Version or commit hash:
  - foxy-devel
- Laser unit:
  - Velodyne- VLP16
  
#### Steps to reproduce issue
Bring up the official velodyne_laserscan node and launch `online_sync_launch.py`

#### Expected behavior
map received

#### Actual behavior
`[sync_slam_toolbox_node-1] LaserRangeScan contains 897 range readings, expected 898` on termial
and no map received on rivz2

#### Additional information
I knew this case happened before (#63 and #141) and I fixed it by do a dirty hack in `Update ()` in karto.h:
```
    m_NumberOfRangeReadings = static_cast<kt_int32u>(math::Round((GetMaximumAngle() -
      GetMinimumAngle()) /
      GetAngularResolution() -0.5) + residual);
```
From the output of `laser_utils.cpp`  i'm sure that ` is_360_lidar = true` so the `residual` is set to **0** in `Update()`.
The problem happened here is because of the round function which makes 897.597 rounded to 898. 

I also checked the code in the `velodyne_laserscan.cpp` from https://github.com/ros-drivers/velodyne/tree/ros2
```
  if ((offset_x >= 0) && (offset_y >= 0) && (offset_r >= 0)) {
    const float kResolution = std::abs(resolution_);
    const size_t kSize = 2.0 * M_PI / kResolution;
    auto scan = std::make_unique<sensor_msgs::msg::LaserScan>();
    scan->header = msg->header;
    scan->angle_increment = kResolution;
    scan->angle_min = -M_PI;
    scan->angle_max = M_PI;
    scan->range_min = 0.0;
    scan->range_max = 200.0;
    scan->time_increment = 0.0;
    scan->ranges.resize(kSize, INFINITY);
```
The output of kSize is 897 since `size_t`  is an unsigned integer data type.（2.0 * M_PI / kResolution= 897.598）

Although I can received a map successfully, this modified code might not be compatible for other lidars. 
Any suggestion for better expandability?

SteveMacenski, 2021-07-19:
You linked to #141 which is of the same topic, which appears to be resolved in #145 . Rounding `897.597` to `898` seems incredibly reasonable to me and I see no error in that calculation. I think your real concern is with the velodyne drivers and improving it to publish more precise values such that the number of readings is correct. The angle increment `kResolution` could be refined to make that round much closer to the actual value of `897`.

Keep in mind this slam library needs the same number of readings per scan. It is OK if there are occasional scans where that is not met (errors happen, that's fine), this library will just throw it away by this error and then use the next laser scan with the "correct" number of readings. This only updates every N meters or M seconds or O radians moved so a single scan offset from when you go from 1.0 meters to 1.04 meters is trivial and unimportant.

vinci-xiao, 2021-07-21:
Understood. Thanks for explanation :)

closed 2021-07-21 by vinci-xiao