SteveMacenski/slam_toolbox #433: possible NULL deference karto::ScanMatcher::MatchScan

easylyou, 2021-09-07:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source
- ROS Version
  - foxy
- Version or commit hash:
  - latest
- Laser unit:
  - Gazebo simulator
  
#### Steps to reproduce issue
I'm doing some test and I could reproduce it. 
But it is hard to me to find the cause. 
I'm appreciated if any help.

```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==1301959==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000010 (pc 0x7f3618bb15eb bp 0x7f360993b580 sp 0x7f360993afa0 T23)
==1301959==The signal is caused by a READ memory access.
==1301959==Hint: address points to the zero page.
    #0 0x7f3618bb15eb in double karto::ScanMatcher::MatchScan<std::vector<karto::LocalizedRangeScan*, std::allocator<karto::LocalizedRangeScan*> > >(karto::LocalizedRangeScan*, std::vector<karto::LocalizedRangeScan*, std::allocator<karto::LocalizedRangeScan*> > const&, karto::Pose2&, karto::Matrix3&, bool, bool) /home/r1/ros2_nav_fuzz/src/slam_toolbox/lib/karto_sdk/src/Mapper.cpp:560:31
    #1 0x7f3618b9de68 in karto::Mapper::Process(karto::LocalizedRangeScan*) /home/r1/ros2_nav_fuzz/src/slam_toolbox/lib/karto_sdk/src/Mapper.cpp:2713:33
    #2 0x7f361da9373a in slam_toolbox::SlamToolbox::addScan(karto::LaserRangeFinder*, std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&, karto::Pose2&) /home/r1/ros2_nav_fuzz/src/slam_toolbox/src/slam_toolbox_common.cpp:542:40
    #3 0x7f361da93178 in slam_toolbox::SlamToolbox::addScan(karto::LaserRangeFinder*, toolbox_types::PosedScan&) /home/r1/ros2_nav_fuzz/src/slam_toolbox/src/slam_toolbox_common.cpp:523:10
    #4 0x7f361e1c8b4c in slam_toolbox::SynchronousSlamToolbox::run() /home/r1/ros2_nav_fuzz/src/slam_toolbox/src/slam_toolbox_sync.cpp:55:7
    #5 0x7f361e27e414 in boost::_mfi::mf0<void, slam_toolbox::SynchronousSlamToolbox>::operator()(slam_toolbox::SynchronousSlamToolbox*) const /usr/include/boost/bind/mem_fn_template.hpp:49:29
    #6 0x7f361e27e2ba in void boost::_bi::list1<boost::_bi::value<slam_toolbox::SynchronousSlamToolbox*> >::operator()<boost::_mfi::mf0<void, slam_toolbox::SynchronousSlamToolbox>, boost::_bi::list0>(boost::_bi::type<void>, boost::_mfi::mf0<void, slam_toolbox::SynchronousSlamToolbox>&, boost::_bi::list0&, int) /usr/include/boost/bind/bind.hpp:259:9
    #7 0x7f361e27e195 in boost::_bi::bind_t<void, boost::_mfi::mf0<void, slam_toolbox::SynchronousSlamToolbox>, boost::_bi::list1<boost::_bi::value<slam_toolbox::SynchronousSlamToolbox*> > >::operator()() /usr/include/boost/bind/bind.hpp:1294:16
    #8 0x7f361e27ae32 in boost::detail::thread_data<boost::_bi::bind_t<void, boost::_mfi::mf0<void, slam_toolbox::SynchronousSlamToolbox>, boost::_bi::list1<boost::_bi::value<slam_toolbox::SynchronousSlamToolbox*> > > >::run() /usr/include/boost/thread/detail/thread.hpp:120:17
    #9 0x7f36180ec43a  (/lib/x86_64-linux-gnu/libboost_thread.so.1.71.0+0x1143a)
    #10 0x7f3618419608 in start_thread (/lib/x86_64-linux-gnu/libpthread.so.0+0x9608)
    #11 0x7f3617c67292 in clone (/lib/x86_64-linux-gnu/libc.so.6+0x122292)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/r1/ros2_nav_fuzz/src/slam_toolbox/lib/karto_sdk/src/Mapper.cpp:560:31 in double karto::ScanMatcher::MatchScan<std::vector<karto::LocalizedRangeScan*, std::allocator<karto::LocalizedRangeScan*> > >(karto::LocalizedRangeScan*, std::vector<karto::LocalizedRangeScan*, std::allocator<karto::LocalizedRangeScan*> > const&, karto::Pose2&, karto::Matrix3&, bool, bool)
Thread T23 created by T0 here:
    #0 0x52f85a in pthread_create (/home/r1/ros2_nav_fuzz/build/slam_toolbox/sync_slam_toolbox_node+0x52f85a)
    #1 0x7f36180eb441 in boost::thread::start_thread_noexcept() (/lib/x86_64-linux-gnu/libboost_thread.so.1.71.0+0x10441)
    #2 0x7f361e27a1b5 in boost::thread::thread<boost::_bi::bind_t<void, boost::_mfi::mf0<void, slam_toolbox::SynchronousSlamToolbox>, boost::_bi::list1<boost::_bi::value<slam_toolbox::SynchronousSlamToolbox*> > > >(boost::_bi::bind_t<void, boost::_mfi::mf0<void, slam_toolbox::SynchronousSlamToolbox>, boost::_bi::list1<boost::_bi::value<slam_toolbox::SynchronousSlamToolbox*> > >&&) /usr/include/boost/thread/detail/thread.hpp:269:13
    #3 0x7f361e1cd113 in std::_MakeUniq<boost::thread>::__single_object std::make_unique<boost::thread, boost::_bi::bind_t<void, boost::_mfi::mf0<void, slam_toolbox::SynchronousSlamToolbox>, boost::_bi::list1<boost::_bi::value<slam_toolbox::SynchronousSlamToolbox*> > > >(boost::_bi::bind_t<void, boost::_mfi::mf0<void, slam_toolbox::SynchronousSlamToolbox>, boost::_bi::list1<boost::_bi::value<slam_toolbox::SynchronousSlamToolbox*> > >&&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/unique_ptr.h:857:34
    #4 0x7f361e1c71eb in slam_toolbox::SynchronousSlamToolbox::SynchronousSlamToolbox(rclcpp::NodeOptions) /home/r1/ros2_nav_fuzz/src/slam_toolbox/src/slam_toolbox_sync.cpp:35:22
    #5 0x60c0cf in void __gnu_cxx::new_allocator<slam_toolbox::SynchronousSlamToolbox>::construct<slam_toolbox::SynchronousSlamToolbox, rclcpp::NodeOptions&>(slam_toolbox::SynchronousSlamToolbox*, rclcpp::NodeOptions&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/ext/new_allocator.h:147:23
    #6 0x60bb50 in void std::allocator_traits<std::allocator<slam_toolbox::SynchronousSlamToolbox> >::construct<slam_toolbox::SynchronousSlamToolbox, rclcpp::NodeOptions&>(std::allocator<slam_toolbox::SynchronousSlamToolbox>&, slam_toolbox::SynchronousSlamToolbox*, rclcpp::NodeOptions&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/alloc_traits.h:484:8
    #7 0x60b4d6 in std::_Sp_counted_ptr_inplace<slam_toolbox::SynchronousSlamToolbox, std::allocator<slam_toolbox::SynchronousSlamToolbox>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<rclcpp::NodeOptions&>(std::allocator<slam_toolbox::SynchronousSlamToolbox>, rclcpp::NodeOptions&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr_base.h:548:4
    #8 0x60ae26 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<slam_toolbox::SynchronousSlamToolbox, std::allocator<slam_toolbox::SynchronousSlamToolbox>, rclcpp::NodeOptions&>(slam_toolbox::SynchronousSlamToolbox*&, std::_Sp_alloc_shared_tag<std::allocator<slam_toolbox::SynchronousSlamToolbox> >, rclcpp::NodeOptions&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr_base.h:680:6
    #9 0x60aaf7 in std::__shared_ptr<slam_toolbox::SynchronousSlamToolbox, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<slam_toolbox::SynchronousSlamToolbox>, rclcpp::NodeOptions&>(std::_Sp_alloc_shared_tag<std::allocator<slam_toolbox::SynchronousSlamToolbox> >, rclcpp::NodeOptions&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr_base.h:1344:14
    #10 0x60a8bc in std::shared_ptr<slam_toolbox::SynchronousSlamToolbox>::shared_ptr<std::allocator<slam_toolbox::SynchronousSlamToolbox>, rclcpp::NodeOptions&>(std::_Sp_alloc_shared_tag<std::allocator<slam_toolbox::SynchronousSlamToolbox> >, rclcpp::NodeOptions&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr.h:359:4
    #11 0x60a618 in std::shared_ptr<slam_toolbox::SynchronousSlamToolbox> std::allocate_shared<slam_toolbox::SynchronousSlamToolbox, std::allocator<slam_toolbox::SynchronousSlamToolbox>, rclcpp::NodeOptions&>(std::allocator<slam_toolbox::SynchronousSlamToolbox> const&, rclcpp::NodeOptions&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr.h:701:14
    #12 0x57b3a3 in std::shared_ptr<slam_toolbox::SynchronousSlamToolbox> std::make_shared<slam_toolbox::SynchronousSlamToolbox, rclcpp::NodeOptions&>(rclcpp::NodeOptions&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr.h:717:14
    #13 0x577fd6 in main /home/r1/ros2_nav_fuzz/src/slam_toolbox/src/slam_toolbox_sync_node.cpp:44:20
    #14 0x7f3617b6c0b2 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x270b2)

==1301959==ABORTING
```

SteveMacenski, 2021-09-07:
Please provide more context

zouyonghao, 2021-09-08:
@SteveMacenski 
There are several calls of `Mapper::Initialize`, through they are wrapped by the checking of `m_Initialized`, and I think this issue can still be reached if they can be called concurrently.

zouyonghao, 2021-09-08:
I have figured out the reason for this issue in our experiment.

We set some parameters to be negative which will cause `ScanMatcher::Create` returning `NULL`.

Though there is a `assert`, it is only functional when it is built in DEBUG mode.

https://github.com/SteveMacenski/slam_toolbox/blob/b2fc766c77e0a0d72c9a3e85191e5fcba33d7c00/lib/karto_sdk/src/Mapper.cpp#L2612-L2617

I have make a pull request for this by checking all the parameters and setting them to default when they are set to be negative.

Besides, I think there is another issue that all the `assert` in this repo should be replaced with another solution.

SteveMacenski, 2021-09-08:
Got it- closing ticket then -- follow up in PR

closed 2021-09-08 by SteveMacenski