SteveMacenski/slam_toolbox #435: map topic is silent

aidinism, 2021-09-09:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - from source
- ROS Version
  - ROS1 noetic
- Version or commit hash:
  - /commit/6fce7fa0cf64a21720e4a3a76c68fcb7f27604c1
- Laser unit:
  - VLP16 @ 600/1200 RPM
  
#### Steps to reproduce the issue
roscore
roslaunch velodyne_pointcloud VLP16_points.launch
roslaunch slam_toolbox online_sync.launch 
rosrun rviz rviz -f velodyne

#### Expected behavior
overtime rviz should show the generated map

#### Actual behavior
map topic is died silent, in the map->Status->Message shows "No map received" and no map is visible.

#### Additional information

LOG:

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.11
 * /slam_toolbox/angle_variance_penalty: 1.0
 * /slam_toolbox/base_frame: base_footprint
 * /slam_toolbox/ceres_dogleg_type: TRADITIONAL_DOGLEG
 * /slam_toolbox/ceres_linear_solver: SPARSE_NORMAL_CHO...
 * /slam_toolbox/ceres_loss_function: None
 * /slam_toolbox/ceres_preconditioner: SCHUR_JACOBI
 * /slam_toolbox/ceres_trust_strategy: LEVENBERG_MARQUARDT
 * /slam_toolbox/coarse_angle_resolution: 0.0349
 * /slam_toolbox/coarse_search_angle_offset: 0.349
 * /slam_toolbox/correlation_search_space_dimension: 0.5
 * /slam_toolbox/correlation_search_space_resolution: 0.01
 * /slam_toolbox/correlation_search_space_smear_deviation: 0.1
 * /slam_toolbox/debug_logging: True
 * /slam_toolbox/distance_variance_penalty: 0.5
 * /slam_toolbox/do_loop_closing: True
 * /slam_toolbox/enable_interactive_mode: False
 * /slam_toolbox/fine_search_angle_offset: 0.00349
 * /slam_toolbox/link_match_minimum_response_fine: 0.1
 * /slam_toolbox/link_scan_maximum_distance: 1.5
 * /slam_toolbox/loop_match_maximum_variance_coarse: 3.0
 * /slam_toolbox/loop_match_minimum_chain_size: 10
 * /slam_toolbox/loop_match_minimum_response_coarse: 0.35
 * /slam_toolbox/loop_match_minimum_response_fine: 0.45
 * /slam_toolbox/loop_search_maximum_distance: 3.0
 * /slam_toolbox/loop_search_space_dimension: 8.0
 * /slam_toolbox/loop_search_space_resolution: 0.05
 * /slam_toolbox/loop_search_space_smear_deviation: 0.03
 * /slam_toolbox/map_frame: map
 * /slam_toolbox/map_update_interval: 5.0
 * /slam_toolbox/max_laser_range: 20.0
 * /slam_toolbox/minimum_angle_penalty: 0.9
 * /slam_toolbox/minimum_distance_penalty: 0.5
 * /slam_toolbox/minimum_time_interval: 0.5
 * /slam_toolbox/minimum_travel_distance: 0.5
 * /slam_toolbox/minimum_travel_heading: 0.5
 * /slam_toolbox/mode: mapping
 * /slam_toolbox/odom_frame: odom
 * /slam_toolbox/resolution: 0.05
 * /slam_toolbox/scan_buffer_maximum_scan_distance: 10
 * /slam_toolbox/scan_buffer_size: 10
 * /slam_toolbox/scan_topic: /scan
 * /slam_toolbox/solver_plugin: solver_plugins::C...
 * /slam_toolbox/stack_size_to_use: 40000000
 * /slam_toolbox/tf_buffer_duration: 30.0
 * /slam_toolbox/throttle_scans: 1
 * /slam_toolbox/transform_publish_period: 0.02
 * /slam_toolbox/transform_timeout: 0.2
 * /slam_toolbox/use_response_expansion: True
 * /slam_toolbox/use_scan_barycenter: True
 * /slam_toolbox/use_scan_matching: True

NODES
  /
    slam_toolbox (slam_toolbox/sync_slam_toolbox_node)

ROS_MASTER_URI=http://localhost:11311

process[slam_toolbox-1]: started with pid [20025]
[ INFO] [1631146251.385417305]: Node using stack size 40000000
[DEBUG] [1631146251.427927883]: Attempting to create managed instance for class solver_plugins::CeresSolver.
[DEBUG] [1631146251.427959295]: Class solver_plugins::CeresSolver maps to library libceres_solver_plugin in classes_available_.
[DEBUG] [1631146251.428036453]: Iterating through all possible paths where libceres_solver_plugin could be located...
[DEBUG] [1631146251.428051943]: Checking path /home/aidin/ws_slam_vlp16/devel/lib/libceres_solver_plugin.so 
[DEBUG] [1631146251.428069738]: Library libceres_solver_plugin found at explicit path /home/aidin/ws_slam_vlp16/devel/lib/libceres_solver_plugin.so.
[DEBUG] [1631146251.430742760]: solver_plugins::CeresSolver maps to real class type solver_plugins::CeresSolver
[ INFO] [1631146251.432755492]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[DEBUG] [1631146251.432784729]: boost::shared_ptr to object of real type solver_plugins::CeresSolver created.
[ INFO] [1631146251.432794959]: Using plugin solver_plugins::CeresSolver
[DEBUG] [1631146251.649618808]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631146251.643, count now 1
[DEBUG] [1631146251.699681030]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631146251.693, count now 2
[DEBUG] [1631146251.749846132]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631146251.744, count now 3
[DEBUG] [1631146251.799658332]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631146251.794, count now 4
[DEBUG] [1631146251.850625797]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631146251.845, count now 5
[DEBUG] [1631146251.900609415]: MessageFilter [target=odom ]: Removed oldest message because buffer is full, count now 5 (frame_id=velodyne, stamp=1631146251.643045)
[DEBUG] [1631146251.900867818]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631146251.895, count now 5
[DEBUG] [1631146251.951484732]: MessageFilter [target=odom ]: Removed oldest message because buffer is full, count now 5 (frame_id=velodyne, stamp=1631146251.693458)
[DEBUG] [1631146251.951612234]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631146251.946, count now 5
[DEBUG] [1631146252.001119874]: MessageFilter [target=odom ]: Removed oldest message because buffer is full, count now 5 (frame_id=velodyne, stamp=1631146251.743912)
[DEBUG] [1631146252.001258508]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631146251.996, count now 5
[DEBUG] [1631146252.052355104]: MessageFilter [target=odom ]: Removed oldest message because buffer is full, count now 5 (frame_id=velodyne, stamp=1631146251.794319)
[DEBUG] [1631146252.052504000]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631146252.046, count now 5
[DEBUG] [1631146252.102126880]: MessageFilter [target=odom ]: Removed oldest message because buffer is full, count now 5 (frame_id=velodyne, stamp=1631146251.844755)
[DEBUG] [1631146252.102267079]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631146252.097, count now 5
[DEBUG] [1631146252.153140771]: MessageFilter [target=odom ]: Removed oldest message because buffer is full, count now 5 (frame_id=velodyne, stamp=1631146251.895209)
[DEBUG] [1631146252.153299207]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631146252.147, count now 5
[DEBUG] [1631146252.203473201]: MessageFilter [target=odom ]: Removed oldest message because buffer is full, count now 5 (frame_id=velodyne, stamp=1631146251.945554)
[DEBUG] [1631146252.203624462]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631146252.198, count now 5
[DEBUG] [1631146252.254081012]: MessageFilter [target=odom ]: Removed oldest message because buffer is full, count now 5 (frame_id=velodyne, stamp=1631146251.995988)
[DEBUG] [1631146252.254278535]: MessageFilter [target=odom ]: Added message in frame velodyne at time 1631146252.248, count now 5

SteveMacenski, 2021-09-09:
It looks like you don't have the TF tree needed to trigger SLAM to actually run if the TF message filter isn't giving messages to the SLAM system to use. See REP 105.

SteveMacenski, 2021-09-19:
Closing due to nonresponse.

closed 2021-09-19 by SteveMacenski

kokkalisko, 2021-11-19:
Facing the exact same issue for Noetic and the classic Turtlebot 3 waffle setup. I attach the TF tree. 
[image]

xwavex, 2022-03-23:
I have the same issue. Any tips how to solve this?