SteveMacenski/slam_toolbox #438: Map does not grows after its initialization

maciej-3, 2021-09-18:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  -  I installed this package both via `sudo apt install ros-foxy-slam-toolbox` as well as from the source (with proper sourcing of the environment). The same behavior in both cases.
- ROS Version
  -  ROS2, foxy
- Version or commit hash:
  - 48e97304618ef5b18aa38cf2ea12c41202ebc092, foxy-devel-unfixed
- Laser unit:
  - RP Lidar A1 and others downloaded from the this git repo's issues
  
#### Steps to reproduce issue

I provide the static transform between my lidar `base_scan` and odom frame 'odom' in `online_sync_launch.py`:
```
    odom_lidar_static_tf = Node(
        name='tf2_ros_odom_lidar',
        package='tf2_ros',
        executable='static_transform_publisher',
        output='screen',
        arguments=['0', '0', '0', '0.0', '0.0', '0.0', 'odom', 'base_scan'],
    )

    ...
   
    ld.add_action(odom_lidar_static_tf)

    ...
```
... and I modify `mapper_params_online_sync.yaml` like this:
```
    # ROS Parameters
    odom_frame: odom
    map_frame: map
    base_frame: base_scan
    scan_topic: /scan
```

and then I do `ros2 launch slam_toolbox online_sync_launch.py`

#### Expected behavior
The map should grow and build progressively as my lidar moves.

#### Actual behavior
The map stays the same after its initialization by the package like this:
[image]

and all the time after the initialization, the map stays as above even though the lidar moved and rotated a lot:
[image]

#### Additional information
After running `ros2 run tf2_tools view_frames.py` I get this TF tree:
[image]

and when I check the transform between `map` and `odom` is stays the same all the time:
```
ros2 run tf2_ros tf2_echo map odom
[INFO] [1631990416.761771843] [tf2_echo]: Waiting for transform map ->  odom: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
At time 1631990417.939988347
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
At time 1631990418.939988140
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
At time 1631990419.939991397
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
At time 1631990420.940001147
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
```

Any idea what I do wrong? Thanks!

maciej-3, 2021-09-21:
Ofc the map does not grow because I put a static transform between `odom` and `base_frame` which is wrong. There should be an odometry method that keeps computing the relative pose between these two frames.

closed 2021-09-21 by maciej-3

amjack0, 2021-12-07:
> Ofc the map does not grow because I put a static transform between `odom` and `base_frame` which is wrong. There should be an odometry method that keeps computing the relative pose between these two frames.

Hi @maciej-3 

I am having the similar issue. Did providing the `odometry` method between two `frames` worked?
How did you solve this?  

Regards,

maciej-3, 2022-01-03:
Hi @amjack0 

I wrote a simple scan-to-scan odometry node for my lidar and it works good enough for this slam package.

maxxbolotov, 2024-12-08:
> Hi @maciej-3
>
> I am having the similar issue. Did providing the `odometry` method between two `frames` worked? How did you solve this?
>
> Regards,

Did you understand what it takes to resolve this problem?
I faced the same issue but quiet don't understand what to do.

maxxbolotov, 2024-12-08:
> Hi @amjack0
> 
> I wrote a simple scan-to-scan odometry node for my lidar and it works good enough for this slam package.

Can you please provide step by step tutorial what I should do to resolve same issue?

TZECHIN6, 2025-03-18:
Just hit the same issue here... the slam toolbox working just fine in the simulated robot, but when i work with the real robot... the map doesnt growth at all.

You have mentioned that the transformation is static cause this issue. First I would like to ask how do I check for myself about this?

Besides, doesnt the odom is stick with the base_frame? or you mean the base_scan should be transformed in time against the odom?

Keegsy, 2025-05-18:
@mac137 Can you show how you wrote this node? I'm also stuck on trying to fix this issue

Keegsy, 2025-05-18:
> Hi [@amjack0](https://github.com/amjack0)
> 
> I wrote a simple scan-to-scan odometry node for my lidar and it works good enough for this slam package.

@mac137 Can you show how you wrote this node? I'm also stuck on trying to fix this issue