SteveMacenski/slam_toolbox #450: Failed to save a large map

rossihwang, 2021-11-06:
**Required Info:**

Indeed, it is not a kind of a BUG of slam_toolbox. however, I think modification make from this side is better

- Operating System:
Ubuntu 20.04
- Installation type:
source
- ROS Version
ROS2
- Version or commit hash:
foxy, df24dc045d5e64ce8655af3c32d9e02ab0af219b
- Laser unit:
with "2 MIT Killian Court" dataset
  
#### Steps to reproduce issue
- Build a large map, for example, the dataset I mention with 0.05 resolution and approximate 6000x6000 pix. Or you can use a test program to publish a large map.
- Call save_map service
- nav2_map_server gets Segmentation Fault

#### Expected behavior
Successfully save the map

#### Actual behavior
nav2_map_server crashs with Segmentation Fault

#### Additional information
After some debugs, I found the cause is nav2_map_server takes too much time to process the map in the process of saving the map. At the same time, slam_toolbox publishes a new message, which would flush the map data that the nav2_map_server is handling. Because the publisher is in setting(transient local, keep last 1).

I suggest stopping the map publishing if the robot is stopped to avoid the data racing.

SteveMacenski, 2021-11-08:
This isn't a slam toolbox issue, its a map server issue, please file the ticket in the appropriate location.

closed 2021-11-08 by SteveMacenski

SteveMacenski, 2021-11-08:
I looked into it quickly and it looks like the future<>promise architecture of that is causing the issue. That wouldn't be an issue if you called the map_server's map saver service which is now part of the default nav2 bringup and use that instead of the SLAM Toolbox utility calling the CLI (which was developed before map server had the availability of the run-time saver server). If you're using this in ROS 1, then its a hard `wontfix`

rossihwang, 2021-11-09:
Thanks, I will have a try