SteveMacenski/slam_toolbox #462: Two LiDAR setup seems to use wrong frames

JuliusSustarevas, 2021-12-18:
**Required Info:**

- Operating System:
ubuntu 18
- Installation type:
debian
- ROS Version
ros1
- Laser unit:
hokuyo 04lx
  
#### Steps to reproduce issue
Have 2 lidars, move around , run async mapping

#### Expected behavior
The maps builds as well as with 1 lidar

#### Actual behavior

The map builds much more poorly than using 1 lidar. From the weird  detections of occupancy in the middle of free space it suggests that  scan msg from one scan Is sometimes used as if with another frame id. 
Running with debug shows that both lidar frames are being read. 
And occupancy is definitely updated using both. And sometimes correctly. But sumtines it just puts scan readings from one side to another. 

#### Additional information
1Lidar results:
[image]
2 lidars:
[image]

SteveMacenski, 2021-12-19:
This project does not support multiple lasers and it is not in the roadmap to do so without contribution from community members

JuliusSustarevas, 2021-12-19:
thanks for a quick reply.
I'll take a look over xmas if it would be easy to make multi lidars work and make a pull request. I have not checked source yet how difficult this would be. 

Could you maybe recommend an approach or recommend what I should look at?  Or just what this would entail? Would it be as easy as adding some handler with appropriate tf lookups before the scan message is pushed deeper to scan matching/mapping etc ?

JuliusSustarevas, 2022-01-04:
Using ira_laser_tools laserscan_multi_merger solves the issue

closed 2022-01-04 by JuliusSustarevas