SteveMacenski/slam_toolbox #463: Need guidance as to why the SlamToolBoxPlugin is not loading

TCIII, 2021-12-26:
Hi Steve,

SBC: LattePanda Atom x86 
OS: Ubuntu 20.04.3 
ROS: Noetic 
Slam ToolBox version:(1.5.6-1focal.20210922.203113) 
rviz version 1.14.10

I am new to ROS/rviz/Slam Toolbox so please bare with me. I do not believe that this is a ROS issue as I have posted on ROS Answers and have not received any responses.

I received this error message when attempting to make a map using rviz in a navigation.rviz file" 

rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[rviz-1]: started with pid [5102]
[ERROR] [1640451452.432491635]: PluginlibFactory: The plugin for class 'slam_toolbox::SlamToolboxPlugin' failed to load.  Error: According to the loaded plugin descriptions the class slam_toolbox::SlamToolboxPlugin with base class type rviz::Panel does not exist. Declared types are  rviz_plugin_tutorials/Teleop
[rospack] Error: no package given
[librospack]: error while executing command
[rospack] Error: no package given
[librospack]: error while executing command
QXcbClipboard: Unable to receive an event from the clipboard manager in a reasonable time
[rviz-1] process has finished cleanly
log file: /home/thomas/.ros/log/0dcde9bc-65a3-11ec-bd19-0b651bddef40/rviz-1*.log
all processes on machine have died, roslaunch will exit

Rviz did open, but there was obviously no scan displayed on the grid.
 
Also, in the rviz "Panels" drop down there is no "Slam Toolbox Plugin" in the "Add New Panel" "Panel Type" display, so I get the feeling that the plugin is not available in my version of the Slam Toolbox.

I have attached the log file below. 

I have exhausted all of the resources I can think of on the IoT and would appreciate some direction as to how I should proceed.

[rviz-1-stdout.log](https://github.com/SteveMacenski/slam_toolbox/files/7777188/rviz-1-stdout.log)
.

SteveMacenski, 2021-12-26:
Did you source before launching rviz?

TCIII, 2021-12-26:
@SteveMacenski,

Thanks for the response, much appreciated.

rviz is launched by a rviz script called navigate.rviz. I run view_navigat.launch which does nothing more than the following:

<launch>
  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find arlobot_ros)/rviz/navigation.rviz"/>
</launch>

navigation.rviz can be found [here](https://github.com/chrisl8/ArloBot/blob/noetic/arlobot_ros/rviz/navigation.rviz).

Could you please provide an example of "source"? I assume that you are referring to updating /.bashrc?

TCIII, 2021-12-27:
@SteveMacenski,

Could this possibly be a bug in the installation of the Slam Toolbox module?

I used a [script](https://ekpyroticfrood.net/?p=537) that installed ROS Noetic, modules, and dependencies so there was no manual installation of the Slam Toolbox.

As I have posted above, is there a procedure that you can provide me that I can follow to install the class slam_toolbox::SlamToolboxPlugin in the rviz plugin descriptions xml file?

TCIII, 2021-12-27:
@SteveMacenski,

I found closed issue #234 which one poster stated he had the same issue that I am seeing with the SlamToolboxPlugin, but it never seemed to have been resolved even when the solution was applied.

I have looked at the solution and determined the following:

1) I have verified the "::" in slam_toolbox_default.rviz

2) However, "slam_toolbox_rviz_plugin.h" seems to be missing from /opt/ros/noetic/include/rviz

I believe that the Slam Toolbox was automatically installed by the Arlobot ROS Noetic install script using Cmake, but there is no independent slamtoolbox directory in the catkin_ws directory.

How should I proceed to add the missing "slam_toolbox_rviz_plugin.h"?

TCIII, 2021-12-27:
@SteveMacenski,

I have purged and then reinstalled rviz and then ran "source /opt/ros/noetic/setup.bash". This did not cure the problem.

I then purged and reinstalled slam toolbox using the follow command "sudo apt install ros-noetic-slam-toolbox", however this did not cure the problem.

I get the feeling that "sudo apt install ros-noetic-slam-toolbox" is not installing the noetic-devel version of the slam toolbox. 

I have attempted to build and install the noetic slam toolbox from source using "git clone -b noetic-devel https://github.com/SteveMacenski/slam_toolbox.git", but keep getting errors:

 -- Could NOT find cmake_modules (missing: cmake_modules_DIR)
-- Could not find the required component 'cmake_modules'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "cmake_modules"
  with any of the following names:

    cmake_modulesConfig.cmake
    cmake_modules-config.cmake

Suggestions on how I can proceed?

TCIII, 2021-12-28:
@SteveMacenski,

What is the difference between "[slam_toolbox_rviz](http://docs.ros.org/en/noetic/api/slam_toolbox_rviz/html/slam__toolbox__rviz__plugin_8h_source.html)" and "slam_toolbox_default_rviz" in /opt/ros/noetic/share/slam_toolbox/config/slam_toolbox_default.rviz?

closed 2021-12-28 by TCIII