SteveMacenski/slam_toolbox #464: slam toolbox rviz plugin not loading

TCIII, 2021-12-28:
**Required Info:**

- Operating System:
  - Ubuntu 20.04.3
- Installation type:
  - source
- ROS Version
  - ROS1 Noetic
- Version or commit hash:
  -rviz version 1.14.10
- Laser unit:
  - RPLidar A1M8
  
#### Steps to reproduce issue
Install ROS Noetic
Install Slam Toolbox using "apt install ros-noetic-slam-toolbox" or with the [Arlobot setup-rosnoetic.sh](https://github.com/chrisl8/ArloBot/blob/noetic/setup-noetic.sh) script the results are the same.
Run "roslaunch rplidar_ros view_rplidar.launch" or run "roslaunch arlobt_ros [view_navigation.launch](https://github.com/chrisl8/ArloBot/blob/noetic/arlobot_ros/launch/view_navigation.launch).

#### Expected behavior
For the Slam Toolbox Plugin to appear on the rviz desktop display.

#### Actual behavior
The Slam Toolbox Plugin appears in the rviz "Panel" dropdown, but does not appear on the rviz desktop display.

#### Additional information
This Slam Toolbox Plugin behavior, when running noetic rivz, was documented in closed issue [#234](https://github.com/SteveMacenski/slam_toolbox/pull/234)

"slam_toolbox_rvizplugin.h" is missing from /opt/ros/noetic/include/rviz 

Steve, I am really desperate here as I have four junior college students who can't move ahead with their robot project until we get this issue put to bed. I have documented my/our efforts in [issue #463](https://github.com/SteveMacenski/slam_toolbox/issues/463) with no resolution of this issue.

TCIII, 2021-12-29:
Hi Steve,

Observation:
I believe that the directory "slam_toolbox_rviz" is not being created in /opt/ros/noetic/share and populated with "rviz_plugins.xml" and "package.xml" at the very least during the build and installation of the Slam Toolbox for a start.

TCIII, 2021-12-29:
Hi Steve,

I have done the following to try and correct this issue with the Slam Toolbox Plugin:

1) Installed ROS Noetic, modules, dependencies, and Slam Toolbox using the setup-noetic.sh script at [Chris Lofland's Arlobot Github](https://github.com/chrisl8/ArloBot). Chris had some issues installing the Slam Toolbox as seen in closed issue [#291](https://github.com/SteveMacenski/slam_toolbox/issues/291).

2) Purged "ros-noetic-slam-toolbox" and "ros-noetic-slam-tool-box-msgs" and performed an "autoclean"

3) Used the following to attempt to install the noetic-devel slam toolbox:
    cd ~/catkin_ws/src/
    git clone -b noetic-devel https://github.com/SteveMacenski/slam_toolbox.git
    cd ..
    rosdep install -q -y -r --from-paths src --ignore-src
    catkin_make 

However catkin_make hung at the 84% point:
[ 84%] Building CXX object slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/src/laser_utils.cpp.o
^Cmake[2]: *** [slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/build.make:63: slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/src/slam_toolbox_common.cpp.o] Interrupt
make[2]: *** [slam_toolbox/slam_toolbox/CMakeFiles/ceres_solver_plugin.dir/build.make:63: slam_toolbox/slam_toolbox/CMakeFiles/ceres_solver_plugin.dir/solvers/ceres_solver.cpp.o] Interrupt
make[2]: *** [slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/build.make:89: slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/src/loop_closure_assistant.cpp.o] Interrupt
make[2]: *** [slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/build.make:102: slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/src/laser_utils.cpp.o] Interrupt
make[1]: *** [CMakeFiles/Makefile2:2358: slam_toolbox/slam_toolbox/CMakeFiles/ceres_solver_plugin.dir/all] Interrupt
make[1]: *** [CMakeFiles/Makefile2:2331: slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/all] Interrupt
make: *** [Makefile:141: all] Interrupt
Traceback (most recent call last):
  File "/opt/ros/noetic/bin/catkin_make", line 306, in <module>
    sys.exit(main())
  File "/opt/ros/noetic/bin/catkin_make", line 249, in main
    run_command(cmd, make_path)
  File "/opt/ros/noetic/lib/python3/dist-packages/catkin/builder.py", line 241, in run_command
    proc.wait()
  File "/usr/lib/python3.8/subprocess.py", line 1083, in wait
    return self._wait(timeout=timeout)
  File "/usr/lib/python3.8/subprocess.py", line 1806, in _wait
    (pid, sts) = self._try_wait(0)
  File "/usr/lib/python3.8/subprocess.py", line 1764, in _try_wait
    (pid, sts) = os.waitpid(self.pid, wait_flags)
KeyboardInterrupt

Also "apt install ros-noetic-slam-toolbox" installs without any issues, but still no Slam Toolbox Plugin in the rviz1 Panel dropdown.

TCIII, 2021-12-29:
Hi Steve,

I also tried the following:
mkdir -p ~/catkin_ws_slam_toolbox/src
cd ~/catkin_ws_slam_toolbox/src
git clone -b noetic-devel https://github.com/SteveMacenski/slam_toolbox.git
cd ..
rosdep install -q -y -r --from-paths src --ignore-src
catkin_make_isolated

This method hangs at the 27% point while attempting to "build CXX object slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/src/laser_utils.cpp.o" where Chris' build method above hangs at the 84% point.

Comments?

TCIII, 2021-12-29:
@SteveMacenski,

As a sanity check I did the following:

1) Did a fresh install of Ubuntu 20.04.3 on a Rpi 4B/4GB
2) In stalled ROS Noetic desktop so the install included rqt and rviz
3) Installed ros_rplidar to ensure that rviz was working and displayed the output of the RPLidar A1M8.
4) Attempted to install Slam Toolbox as follows:
 cd ~/catkin_ws/src/
git clone -b noetic-devel https://github.com/SteveMacenski/slam_toolbox.git
cd ..
rosdep install -q -y -r --from-paths src --ignore-src
catkin_make

The build fails in the same place as above: "Building CXX object slam_toolbox/slam_toolbox/CMakeFiles/toolbox_common.dir/src/laser_utils.cpp.o"

Comments?

TCIII, 2021-12-29:
@SteveMacenski,

After the catkin_make build failed, I figured what the heck and ran "sudo apt install ros-noetic-slam-toolbox" which as usual installed without issue.

Then I ran "roslaunch rplidar_ros view_rplidar.launch". 
The rplidar launched, followed by startup data and then rviz opened.
When rviz opened, I looked at the "Panels" dropdown menu and the Slam Toolbox Plugin was not there.
So I selected "Add New Panel" and when the "Panel Type" display opened, there was the "Slam Toolbox Plugin" under the "slam_toolbox_rviz folder.
So I selected the "Slam Toolbox Plugin" and it was added to the "Panels" dropdown menu and appeared on the left side of the rviz display.
I can turn the ""Slam Toolbox Plugin" on and off using the "Panels" dropdown menu.

I would say at this point that there is an issue with the ROS Noetic installation/packages/dependencies in the Arlobot "setup-noetic.sh script possibly?

Comments?

SteveMacenski, 2022-01-03:
Did you install the rviz package? `ros-noetic-slam-toolbox-rviz`

I just did a clean install of noetic on my Ubuntu 20.04 machine, installed slam toolbox and all of the related packages and opened rviz and works fine.

closed 2022-01-03 by SteveMacenski

TCIII, 2022-01-03:
@SteveMacenski,

Thanks for the response, much appreciated.

Interesting to say the least.

Chris Lofland over at Arlobot has seen the same issue as me when installing Slam Toolbox on Ubuntu 20.04 running ROS Noetic.

However I think that the issue has been that we both have been using "apt install ros-noetic-slam-toolbox" and not "apt install ros-noetic-slam-toolbox-rviz" as you did.

Comments?