SteveMacenski/slam_toolbox #466: Assert pClosestScan != NULL fail when loading from file

JuliusSustarevas, 2022-01-07:
**Required Info:**

- Operating System:
Host ubuntu 20. Docker with ubuntu 18 and ros melodic
- Installation type:
binaries
- ROS Version
ROS1 melodic
- Version or commit hash:
 current melodic deb
- Laser unit:
2x hokuyo urg-04lx-ug01  merged with iris merger into a single lidar
  
#### Steps to reproduce issue
1. use mapping yaml to map an environment -> works
2.  save/serialse map to file
3. relaunch systems using localisation yaml. set file name / starting pose
4. get error

#### Expected behavior
that the map loads from file
#### Actual behavior

process[slam_toolbox-11]: started with pid [959]
[ INFO] [1641574995.032614014]: Node using stack size 40000000
[ INFO] [1641574995.073004728]: Starting calibration. This will take a few seconds.
[ WARN] [1641574995.073034597]: Time calibration is still experimental.
[DEBUG] [1641574995.089224198]: Attempting to create managed instance for class solver_plugins::CeresSolver.
[DEBUG] [1641574995.089299769]: Class solver_plugins::CeresSolver maps to library libceres_solver_plugin in classes_available_.
[DEBUG] [1641574995.089539644]: Iterating through all possible paths where libceres_solver_plugin could be located...
[DEBUG] [1641574995.089582943]: Checking path /home/armstone/ws/devel/lib/libceres_solver_plugin.so 
[DEBUG] [1641574995.089611872]: Checking path /home/armstone/ws/devel/lib/libceres_solver_plugin.so 
[DEBUG] [1641574995.089635548]: Checking path /opt/ros/melodic/lib/libceres_solver_plugin.so 
[DEBUG] [1641574995.089664880]: Library libceres_solver_plugin found at explicit path /opt/ros/melodic/lib/libceres_solver_plugin.so.
[DEBUG] [1641574995.092260505]: solver_plugins::CeresSolver maps to real class type solver_plugins::CeresSolver
[ INFO] [1641574995.094039167]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[DEBUG] [1641574995.094110035]: boost::shared_ptr to object of real type solver_plugins::CeresSolver created.
[ INFO] [1641574995.094140768]: Using plugin solver_plugins::CeresSolver
Load From File /home/armstone/jan7_2.posegraph 
Mapper <- Module
Mapper <- m_pSequentialScanMatcher
Mapper <- m_pGraph
MapperGraph <- Graph; Graph <- m_Edges; Graph <- m_Vertices
MapperGraph <- m_pMapper; MapperGraph <- m_pLoopScanMatcher; MapperGraph <- m_pTraversal
Mapper <- m_pMapperSensorManager
MapperSensorManager <- m_ScanManagers; MapperSensorManager <- m_Scans
Mapper <- m_Listeners
**Finished serializing Mapper**
Load From File
**Serializing Dataset**
Dataset <- m_SensorNameLookup
Dataset <- m_Data
Dataset <- m_Lasers
Dataset <- m_pDatasetInfo
**Finished serializing Dataset**
[DEBUG] [1641574995.244474255]: DeserializePoseGraph: Successfully read file.
Registering sensor: [Custom Described Lidar]
[ INFO] [1641574995.250753989]: Waiting for incoming scan to get metadata...
[ INFO] [1641574995.451421642]: Starting calibration. This will take a few seconds.
[ WARN] [1641574995.451602210]: Time calibration is still experimental.
[ INFO] [1641574995.657105685]: Subscribing to topics   2
[ INFO] [1641574996.253442602]: Waiting for incoming scan to get metadata...
[ INFO] [1641574996.298933723]: Odometry frame_id set to odom
[ INFO] [1641574996.303446489]: Base frame_id : base_link_footprint
[ INFO] [1641574997.260709250]: Waiting for incoming scan to get metadata...
[ INFO] [1641574998.263332446]: Waiting for incoming scan to get metadata...
[ INFO] [1641574999.172078549]: Calibration finished. Latency is: -0.0771.
[ INFO] [1641574999.270033862]: Waiting for incoming scan to get metadata...
[ INFO] [1641574999.372182464]: Streaming data.
[ INFO] [1641574999.565655978]: Calibration finished. Latency is: -0.0804.
[ INFO] [1641574999.766368138]: Streaming data.
[ INFO] [1641574999.974763346]: Got scan!
[DEBUG] [1641574999.975277522]: laser base_link_footprint's pose wrt base: 0.000 0.000 0.000 0.000
Info: clipped range threshold to be within minimum and maximum range!
[ERROR] [1641574999.977614917]: CeresSolver: Ceres was called when there are no nodes. This shouldn't happen.
[DEBUG] [1641575000.051204895]: Trying to publish message of type [nav_msgs/OccupancyGrid/3381f2d731d4076ec5c71b0759edbe4e] on a publisher with type [nav_msgs/OccupancyGrid/3381f2d731d4076ec5c71b0759edbe4e]
[DEBUG] [1641575000.051392352]: Trying to publish message of type [nav_msgs/MapMetaData/10cfc8a2818024d3248802c00c95f11b] on a publisher with type [nav_msgs/MapMetaData/10cfc8a2818024d3248802c00c95f11b]
[DEBUG] [1641575000.052451052]: MessageFilter [target=odom ]: Added message in frame base_link_footprint at time 1641574999.808, count now 0
AddEdge: At least one vertex is invalid.
localization_slam_toolbox_node: /tmp/binarydeb/ros-melodic-slam-toolbox-1.1.6/lib/karto_sdk/src/Mapper.cpp:1644: void karto::MapperGraph::LinkChainToScan(const LocalizedRangeScanVector&, karto::LocalizedRangeScan*, const karto::Pose2&, const karto::Matrix3&): Assertion `pClosestScan != NULL' failed.
No handlers could be found for logger "roslaunch"
[slam_toolbox-11] process has died [pid 959, exit code -6, cmd /opt/ros/melodic/lib/slam_toolbox/localization_slam_toolbox_node __name:=slam_toolbox __log:=/home/armstone/.ros/log/b0785ee4-6fdb-11ec-8e9a-54b20314f881/slam_toolbox-11.log].
log file: /home/armstone/.ros/log/b0785ee4-6fdb-11ec-8e9a-54b20314f881/slam_toolbox-11*.log

#### Additional information

SteveMacenski, 2022-01-10:
>  [ERROR] [1641574999.977614917]: CeresSolver: Ceres was called when there are no nodes. This shouldn't happen.

That error message says it all - you tried to serialize with no information in the SLAM session

SteveMacenski, 2022-01-11:
Did your deserialized graph your loaded contain any information?

JuliusSustarevas, 2022-01-11:
The pose graph files are that I'm trying are 4.2mb and 11mb. 4.2 one was created with real hardware. 11mb with sim. The 11mb actually loads  so maybe its the lidar merger than I'm using  having some weird interaction. The .pgm saves perfectly in both. I'll close and add comment later if i resolve it.

closed 2022-01-11 by JuliusSustarevas

SteveMacenski, 2022-01-11:
OK