SteveMacenski/slam_toolbox #469: TBB does not have `tbb_stddef.h` anymore

goekce, 2022-01-20:
**Required Info:**

- Operating System:
  - Arch Linux
- Installation type:
  - from source
- ROS Version
  - ROS2 galactic
- Version or commit hash:
  - 2.5.1
- Laser unit:
  - N/A
  
#### Steps to reproduce issue

Compilation

#### Expected behavior

Compilation succeeds

#### Actual behavior

I get the following error during compilation
```
CMake Error at CMake/FindTBB.cmake:400 (file):                                                                                                                                                                                               
  file STRINGS file "/usr/include/tbb/tbb_stddef.h" cannot be read.                                                                                                                                                                          
Call Stack (most recent call first):                                                                                                                                                                                                         
  lib/karto_sdk/CMakeLists.txt:15 (find_package)
```

#### Additional information

https://github.com/SteveMacenski/slam_toolbox/blob/8b251f1a4c8bb84b9dc3376cd8c3511826dd4e9b/CMake/FindTBB.cmake#L399-L403

Intel TBB does not seem to have `tbb_stddef.h` anymore since version 2021.1 [[1]](https://github.com/opencv/opencv/issues/19358). I have the version 2021.5.0.

SteveMacenski, 2022-01-20:
We don’t specifically support Arch. Happy to have a PR that would enable it, but that isn’t a target platform for ROS2 (yet)

goekce, 2022-01-21:
Thanks Steve, I also do not expect that ROS supports Archlinux :) I thought Ubuntu will probably also update `libtbb-dev` in the next months so it is better to include the patch.
Then it is better to wait for Ubuntu to update the package.

closed 2022-01-21 by goekce

m2-farzan, 2022-01-22:
For future reference:

1. The new version of TBB (>= 2021.1) comes with cmake discovery included. So we no longer need `FindTBB.cmake` file. We just have to use `find_package(TBB REQUIRED)` and `target_link_libraries(... TBB::tbb)`
2. TBB (>= 2021.1) has [retired some of its functions](https://www.intel.com/content/dam/develop/external/us/en/documents/tbbrevamp.pdf) but there are close equivalents for all of our use cases.

Based on these points I have put together [a patch](https://github.com/m2-farzan/ros2-supplementary-pkgbuilds/blob/3f8a7f09701b0778a73aede42ba9134092471979/ros2-galactic-common/slam_toolbox.patch) that makes slam_toolbox compile against the new version of tbb but I don't know tbb so I'm not sure about correctness of my patch. Plus, ubuntu still has the old version ([even 22.04 comes with 2018.12](https://packages.ubuntu.com/jammy/libbtbb-dev)) so I guess there's no point in making a PR now...

SteveMacenski, 2022-01-22:
Got it! Thanks for the detail, I imagine we’ll be needing it shortly for the 22.04 update when Humble testing starts in April. I’ve pinned this to my calendar around then!

Thanks for the debugging time and it will definitely be made use of in the near future

goekce, 2022-01-22:
>... ([even 22.04 comes with 2018.12](https://packages.ubuntu.com/jammy/libbtbb-dev))

You have a tiny typo, `libbtbb` is Bluetooth lib :)

[Ubuntu 22.04 currently has TBB 2020.03](https://packages.ubuntu.com/jammy/libtbb-dev).

samuk, 2022-07-28:
Trying to use this script https://raw.githubusercontent.com/milq/milq/master/scripts/bash/install-opencv.sh

To compile OpenCV on 22.04 / Humble I get

```
/home/ubuntu/OpenCV/modules/core/src/parallel.cpp:104:14: fatal error: tbb/tbb_stddef.h: No such file or directory
  104 |     #include "tbb/tbb_stddef.h"
```

afrixs, 2022-09-06:
[Ubuntu 22.04 has currently moved to TBB 2021.5](https://packages.ubuntu.com/jammy/libtbb-dev) so this should be reopened. Approving that @m2-farzan 's [patch](https://github.com/m2-farzan/ros2-supplementary-pkgbuilds/blob/3f8a7f09701b0778a73aede42ba9134092471979/ros2-galactic-common/slam_toolbox.patch) builds successfully and mapping seems to work like before.

SteveMacenski, 2022-09-06:
Sure! Can someone submit the patch? I'll pull in updates and test that it still turns over fine