SteveMacenski/slam_toolbox #473: Incorrect residual scaling.

dimasad, 2022-01-28:
**Required Info:**

- Operating System:
  - Any, but I'm using Ubuntu 20.04
- Installation type:
  - Any, but I'm building from source.
- ROS Version
  - ROS
- Version or commit hash:
  - fab61e1225d6c1ee5d634a4851b0a387c11c2c39 (galactic)
- Laser unit:
  - Any
  
#### Steps to reproduce issue

There is an error in the scaling of the residual error during creation of `PoseGraph2dErrorTerm` in `CeresSolver::AddConstraint`. This is a logic or math error. Basically, using some LaTeX, the cost associated to each Ceres block is $1/2 e^T e$, where e is the normalized residual vector and ^{T} indicates the transpose. However, the function we are trying to minimize, associated to the log-density of a multivariate Gaussian distribution, is $1/2 r^T Q^{-1} r$, where r is the unscaled residual, Q is the covariance matrix and its inverse Q^{-1} is the information or precision matrix. Now, take the Cholesky decomposition $Q^{-1} = LL^T = U^T U$, using the notation of [Eigen](https://eigen.tuxfamily.org/dox/classEigen_1_1LLT.html), where L is lower triangular and $L^T = U$ is upper triangular. If we define $e=Ur=L^T r$ then $1/2 e^T e  = 1/2 r^T U^T U r = 1/2 r^T Q^{-1} r$.

#### Expected behavior
The log-density associated to the posegraph is a full Bayesian representation of the degree of belief about the robot's path, even if getting moments such as mean and covariance is intractable. However, if techniques to approximate the covariance such as the [ESGVI](http://doi.org/10.1177/0278364920937608) are used, they should yield reasonable uncertainty bounds. 

#### Actual behavior
I used slam_toolbox as a front-end to test my experimental SLAM backend that is equivalent to the [ESGVI](http://doi.org/10.1177/0278364920937608) and found unrealistically small (submillimeter) uncertainty bounds, using the Intel research labs or ACES3 [datasets](https://www.ipb.uni-bonn.de/datasets/).

#### Additional information
I also noticed that the `ceres_solver.cpp` is very similar to the [Ceres 2d example](https://ceres-solver.googlesource.com/ceres-solver/+/refs/heads/master/examples/slam/pose_graph_2d/pose_graph_2d.cc), which uses the Cholesky decomposition, albeit incorrectly as well. They took L and not U.

SteveMacenski, 2022-01-28:
Yeah, my ceres plugin is derivative of the example you mention. If you submit a PR with the updated calculations, I can review and test the next time I'm doing a big slam toolbox update push.

dimasad, 2022-01-28:
I actually fixed and tested a bit before submitting the bug report. The solution is very simple, although it is hard to know how much it will change (positively or negatively) existing examples and applications. I do believe it is the _correct_ way to do it, though.

SteveMacenski, 2022-03-21:
Forward ported to `ros2` https://github.com/SteveMacenski/slam_toolbox/commit/646ef41b14e2b5f72e721963fafa490afa3d64fc