SteveMacenski/slam_toolbox #476: add a api to feed a extern localization pose to slam_toolbox localization mode to adapt dynamic environment

ZOUJIASHUAI, 2022-02-22:
I wrote a reflector localization algorithim can output a good pose result when the lidar detect the reflector,since I mapped reflector in to the slam_toolbox map, so I can get almost the same pose result from slam_toolbox and reflector localization algorithim if the environment has not changed a lot,but some factory environment is not static and unable to put a lot of reflector,  so  Im wonder  if I can continue(lidar has detect reflector) input a correct localization pose to slam_toolbox localization mode,then I can let the localization more Robustness.
      I know I can use initpose api ,but can I keep send pose to this api?since the robot doesnt know when or where can detect reflector and when or where the environment will change.I found out that most of the time slam_toolbox local mode can output a good pose result even out of map,and some time it will get a little drift.

ZOUJIASHUAI, 2022-02-22:
I wrote a reflector localization algorithim can output a good pose result when the lidar detect the reflector,since I mapped reflector in to the slam_toolbox map, so I can get almost the same pose result from slam_toolbox and reflector localization algorithim if the environment has not changed a lot,but some factory environment is not static and unable to put a lot of reflector,  so  Im wonder  if I can continue(lidar has detect reflector) input a correct localization pose to slam_toolbox localization mode,then I can let the localization more Robustness.
  I know I can use initpose api ,but can I keep send pose to this api?since the robot doesnt know when or where can detect reflector and when or where the environment will change.I found out that most of the time slam_toolbox local mode can output a good pose result even out of map,and some time it will get a little drift.

SteveMacenski, 2022-02-22:
Yes, you can keep sending initial poses over time as needed

closed 2022-02-22 by SteveMacenski

ZOUJIASHUAI, 2022-02-23:
> Yes, you can keep sending initial poses over time as needed

will the result gose wrong if I send the inital pose at 30hz?

SteveMacenski, 2022-02-23:
If you have accurate reflector positioning at 30hz, you should use that instead.