SteveMacenski/slam_toolbox #480: snap image for arm64 machines

austin-InDro, 2022-02-24:
#### Feature description
snap image for arm64 processors (NVIDIA Jetsons)

#### Implementation considerations
Would allow for an optimized slam-toolbox in more machines and robots. It's quite common to use an arm computers in smaller robots.

SteveMacenski, 2022-02-25:
I don't disagree, though you can always build from source with release flags, it will be very similar. Honestly the snap ecosystem has passed me by since I released that snap about 4 years ago and I'm not sure (1) if I can actually build a ARM snap on my x86 computer and (2) if any of my tools I built for snaps even work anymore with all the ABI changes. 

If you can find a way to generate the snap locally with changes to work on ARM, I'd be happy to run your commands and push it to snapcraft so that you can use it on all of your robots without manually building.

austin-InDro, 2022-02-25:
The main benefit I thought may come of having the snap would be the optimizations you mentioned in the README. If those wouldn't be applied, it won't be as attractive to me. Having said that, the build for your toolbox takes quite a while, so I can try to build an image when I have some time. I'll recomment here if I have any questions regarding your package specifically.

Thanks,

closed 2022-02-25 by austin-InDro