SteveMacenski/slam_toolbox #481: Fusing SLAM into Odom Topic output for EKF

austin-InDro, 2022-02-25:
#### Feature description
Instead of the SLAM toolbox outputting to the TF node directly. Could there be an option to allow for a topic to publish to?

#### Implementation considerations
This would allow for intergration with the robot_localization package so that the calculated Odom position from the SLAM could be fused with other sensors.

SteveMacenski, 2022-02-25:
You could add a parameter to disable it, sure

austin-InDro, 2022-02-25:
Where in your code would I make that modification? I've been poking around `slam_toolbox_common.cpp`

Would I publish the `read_pose` variable?

austin-InDro, 2022-02-25:
Hi @SteveMacenski,

I haven't found a good place to implement this. Would you be able to help with some guidence on your code?

Thanks,

SteveMacenski, 2022-02-27:
Add a parameter that would be put a boolean over this https://github.com/SteveMacenski/slam_toolbox/blob/ros2/src/slam_toolbox_common.cpp#L76-L78 to create the thread necessary to publish the transform.

Also, I think you can disable it now if you just set the `transform_publish_period` to `0`, no?

austin-InDro, 2022-03-01:
Ah I see, thanks. I did a much more hack method by placing some publishing code for a PoseWithCovariance in `slam_toolbox_common.cpp`. So I kept the thread, but commented out the publish transform message. I can go into more detail if you'd like, however I'm just currently away from my robot now. I can publish my modifications here next week.

I could attempt to implement it they way you mentioned above, then set up a PR request if you'd like?

SteveMacenski, 2022-03-01:
Sure, sounds good! I'm always a fan of less hacky methods where possible :-)

SteveMacenski, 2022-07-11:
#508 implements

closed 2022-07-11 by SteveMacenski