SteveMacenski/slam_toolbox #485: fatal error: tf2_geometry_msgs/tf2_geometry_msgs.hpp: No such file or directory

XiongHenning, 2022-03-23:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source
- ROS Version
  -  ROS2 
- Version or commit hash:
  on branch ros2
- Laser unit:

  
When I run colcon build --cmake-args -DCMAKE_BUILD_TYPE=Debug
then ｉgot errors
/home/code/ros2_ws/src/slam_toolbox/include/slam_toolbox/slam_mapper.hpp:25:10: fatal error: tf2_geometry_msgs/tf2_geometry_msgs.hpp: No such file or directory
   25 | #include "tf2_geometry_msgs/tf2_geometry_msgs.hpp"

even after i run apt-get install ros-foxy-tf2-geometry-msgs

hidmic, 2022-03-23:
@XiongHenning there have been changes in upstream `tf2_geometry_msgs` going from Foxy to Rolling.

Either switch to the `foxy-devel` branch or build against Rolling.

closed 2022-03-23 by hidmic

XiongHenning, 2022-03-23:
thanks for your answer．　
yes, when ｉcheckout to foxy-devel, then i can build it now

ShuiYunXi, 2022-05-13:
ubuntu:20.04
ros:foxy
when i colcon bulild and report some error as follow:
note:I had used the command   with sudo apt-get install  ros-foxy-tf2-geometry-msgs .
```
In file included from /mnt/hgfs/ShareUbuntu/AGV/ros2_prj/AGV_SIM_WS_test/src/slam_toolbox/src/loop_closure_assistant.cpp:22:
/mnt/hgfs/ShareUbuntu/AGV/ros2_prj/AGV_SIM_WS_test/src/slam_toolbox/include/slam_toolbox/loop_closure_assistant.hpp:29:10: fatal error: tf2_geometry_msgs/tf2_geometry_msgs.hpp: No such file or directory
   29 | #include "tf2_geometry_msgs/tf2_geometry_msgs.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
and:
In file included from /mnt/hgfs/ShareUbuntu/AGV/ros2_prj/AGV_SIM_WS_test/src/slam_toolbox/src/slam_toolbox_common.cpp:23:
/mnt/hgfs/ShareUbuntu/zxkj/AGV/ros2_prj/AGV_SIM_WS_test/src/slam_toolbox/include/slam_toolbox/slam_toolbox_common.hpp:40:10: fatal error: tf2_sensor_msgs/tf2_sensor_msgs.hpp: No such file or directory
   40 | #include "tf2_sensor_msgs/tf2_sensor_msgs.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
when I change include "tf2_geometry_msgs/tf2_geometry_msgs.h and include "tf2_sensor_msgs/tf2_sensor_msgs.hpp and reprort error as as follows：
`
/opt/ros/foxy/include/rclcpp/node_impl.hpp:169:23: error: no matching function for call to ‘rclcpp::ParameterValue::get<rclcpp::ParameterType>() const’
  169 |     ).get<ParameterT>();
/opt/ros/foxy/include/rclcpp/parameter_value.hpp:254:3: error: no type named ‘type’ in ‘struct std::enable_if<false, const bool&>’
/opt/ros/foxy/include/rclcpp/parameter_value.hpp:263:3: error: no type named ‘type’ in ‘struct std::enable_if<false, const long int&
/opt/ros/foxy/include/rclcpp/parameter_value.hpp:271:3: error: no type named ‘type’ in ‘struct std::enable_if<false, const double&>’
/opt/ros/foxy/include/rclcpp/parameter_value.hpp:279:3: error: no type named ‘type’ in ‘struct std::enable_if<false, const std::__cxx11::basic_string<char>&>’
/opt/ros/foxy/include/rclcpp/parameter_value.hpp:289:3: error: no type named ‘type’ in ‘struct std::enable_if<false, const std::vector<unsigned char, std::allocator<unsigned char> >&>’
/opt/ros/foxy/include/rclcpp/parameter_value.hpp:299:3: error: no type named ‘type’ in ‘struct std::enable_if<false, const std::vector<bool>&>’
/opt/ros/foxy/include/rclcpp/parameter_value.hpp:309:3: error: no type named ‘type’ in ‘struct std::enable_if<false, const std::vector<long int, std::allocator<long int> >&
/opt/ros/foxy/include/rclcpp/parameter_value.hpp:319:3: error: no type named ‘type’ in ‘struct std::enable_if<false, const std::vector<double, std::allocator<double> >&>’
/opt/ros/foxy/include/rclcpp/parameter_value.hpp:329:3: error: no type named ‘type’ in ‘struct std::enable_if<false, const std::vector<std::__cxx11::basic_string<char> >&>’
/src/slam_toolbox/src/slam_toolbox_common.cpp:309:8: error: ‘void map_start_pose’ has incomplete type
  309 |   auto map_start_pose = this->declare_parameter("map_start_pose",rclcpp::ParameterType::PARAMETER_DOUBLE_ARRAY);
      |        ^~~~~~~~~~~~~~
src/slam_toolbox_common.cpp:310:8: error: ‘void map_start_at_dock’ has incomplete type
  310 |   auto map_start_at_dock = this->declare_parameter("map_start_at_dock",rclcpp::ParameterType::PARAMETER_BOOL)
/src/slam_toolbox_common.cpp:315:56: error: expected primary-expression before ‘>’ token
  315 |       read_pose = map_start_pose.get<std::vector<double>>();
/src/slam_toolbox_common.cpp:315:59: error: expected primary-expression before ‘)’ token
  315 |       read_pose = map_start_pose.get<std::vector<double>>();
/src/slam_toolbox/src/slam_toolbox_common.cpp:330:45: error: expected primary-expression before ‘bool’
  330 |       start_at_dock = map_start_at_dock.get<bool>();

`